open-mmlab/OpenPCDet #93: How to inference a pcd file

zhiyuan57, 2020-06-23:
Hi Sshaoshuai,

I want to use pcdet for inference in my project, but I don’t know how to use it, can you tell me  how to inference a 360 degree point cloud data ?

Thanks!

sshaoshuai, 2020-06-24:
You need first to decode the pcd file to regular points Nx(3+C). For training & inference with 360 degree point cloud data, you need to write your own dataset and configs (refer to our provided [kitti configs](https://github.com/open-mmlab/OpenLidarPerceptron/blob/master/tools/cfgs/dataset_configs/kitti_dataset.yaml)).

closed 2020-07-01 by sshaoshuai

xxp912, 2021-06-02:
> Hi Sshaoshuai,
> 
> I want to use pcdet for inference in my project, but I don’t know how to use it, can you tell me how to inference a 360 degree point cloud data ?
> 
> Thanks!

Hello, do you have saved this problem? I meet the same problem with you and I don't know how convert the .pad file to .npy file. Looking forward to your reply.

Oh-JongJin, 2024-07-05:
```python
import numpy as np
import open3d as o3d

def flatten(lst):
    result = []
    for item in lst:
        result.extend(item)
    return result

pcd_file = 'test.pcd'   # Your pcd file
pcd = o3d.io.read_point_cloud(pcd_file)
points = np.asarray(pcd.points, dtype=np.float32)

pcd_to_npy = []
for idx, i in enumerate(points):
    values = i.tolist()
    values.append(0.)
    pcd_to_npy.append(values)

pcd_to_npy = np.array(flatten(pcd_to_npy), dtype=np.float32)

fname = os.path.splitext(pcd_file)[0]
np.save(fname, pcd_to_npy)
sys.exit()
```

A `.npy` file with the same name will be created in the same directory as the input `.pcd` file.

Oh-JongJin, 2024-07-05:
> > Hi Sshaoshuai,
> > I want to use pcdet for inference in my project, but I don’t know how to use it, can you tell me how to inference a 360 degree point cloud data ?
> > Thanks!
> 
> Hello, do you have saved this problem? I meet the same problem with you and I don't know how convert the .pcd file to .npy file. Looking forward to your reply.

```python
import numpy as np
import open3d as o3d

def flatten(lst):
    result = []
    for item in lst:
        result.extend(item)
    return result

pcd_file = 'test.pcd'   # Your pcd file
pcd = o3d.io.read_point_cloud(pcd_file)
points = np.asarray(pcd.points, dtype=np.float32)

pcd_to_npy = []
for idx, i in enumerate(points):
    values = i.tolist()
    values.append(0.)
    pcd_to_npy.append(values)

pcd_to_npy = np.array(flatten(pcd_to_npy), dtype=np.float32)

fname = os.path.splitext(pcd_file)[0]
np.save(fname, pcd_to_npy)
sys.exit()
```

A `.npy` file with the same name will be created in the same directory as the input `.pcd` file.