open-mmlab/OpenPCDet #96: NotImplementedError

LittleYuanzi, 2020-06-24:
when i run the new version, i got this error,any ideas?
`  File "/home/jty/pcdet/pvrcnn/OpenLidarPerceptron/tools/train.py", line 166, in <module>
    main()
  File "/home/jty/pcdet/pvrcnn/OpenLidarPerceptron/tools/train.py", line 99, in main
    training=True
  File "/home/jty/pcdet/pvrcnn/OpenLidarPerceptron/pcdet/datasets/__init__.py", line 58, in build_dataloader
    drop_last=False, sampler=sampler, timeout=0
  File "/home/jty/anaconda3/envs/pointpillars/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 213, in __init__
    sampler = RandomSampler(dataset)
  File "/home/jty/anaconda3/envs/pointpillars/lib/python3.7/site-packages/torch/utils/data/sampler.py", line 92, in __init__
    if not isinstance(self.num_samples, int) or self.num_samples <= 0:
  File "/home/jty/anaconda3/envs/pointpillars/lib/python3.7/site-packages/torch/utils/data/sampler.py", line 100, in num_samples
    return len(self.data_source)
  File "/home/jty/pcdet/pvrcnn/OpenLidarPerceptron/pcdet/datasets/dataset.py", line 43, in __len__
    raise NotImplementedError
NotImplementedError`

sshaoshuai, 2020-06-24:
Could you please paste the command you run? I have double checked to run the test.py and didn't see such errors.

LittleYuanzi, 2020-06-24:
my mistake.sorry

closed 2020-06-24 by sshaoshuai