open-mmlab/OpenPCDet #98: multi GPUS training error.

LittleYuanzi, 2020-06-25:
I can train with single GPU,but when I use multi-gpus, I got this error.

`
Traceback (most recent call last):
  File "train.py", line 168, in <module>
    main()
  File "train.py", line 160, in main
    merge_all_iters_to_one_epoch=args.merge_all_iters_to_one_epoch
  File "/home/jty/pcdet/pvrcnn/OpenLidarPerceptron/tools/train_utils/train_utils.py", line 75, in train_model
    dataloader_iter = iter(train_loader)
  File "/home/jty/anaconda3/envs/pointpillars/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 278, in __iter__
    return _MultiProcessingDataLoaderIter(self)
  File "/home/jty/anaconda3/envs/pointpillars/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 682, in __init__
    w.start()
  File "/home/jty/anaconda3/envs/pointpillars/lib/python3.7/multiprocessing/process.py", line 112, in start
    self._popen = self._Popen(self)
  File "/home/jty/anaconda3/envs/pointpillars/lib/python3.7/multiprocessing/context.py", line 223, in _Popen
    return _default_context.get_context().Process._Popen(process_obj)
  File "/home/jty/anaconda3/envs/pointpillars/lib/python3.7/multiprocessing/context.py", line 284, in _Popen
    return Popen(process_obj)
  File "/home/jty/anaconda3/envs/pointpillars/lib/python3.7/multiprocessing/popen_spawn_posix.py", line 32, in __init__
    super().__init__(process_obj)
  File "/home/jty/anaconda3/envs/pointpillars/lib/python3.7/multiprocessing/popen_fork.py", line 20, in __init__
    self._launch(process_obj)
  File "/home/jty/anaconda3/envs/pointpillars/lib/python3.7/multiprocessing/popen_spawn_posix.py", line 47, in _launch
    reduction.dump(process_obj, fp)
  File "/home/jty/anaconda3/envs/pointpillars/lib/python3.7/multiprocessing/reduction.py", line 60, in dump
    ForkingPickler(file, protocol).dump(obj)
_pickle.PicklingError: Can't pickle <function DataAugmentor.random_world_flip at 0x7f81d52ecc20>: it's not the same object as datasets.augmentor.data_augmentor.DataAugmentor.random_world_flip
`

Gus-Guo, 2020-06-25:
Could you please post your command here ?

LittleYuanzi, 2020-06-25:
> Could you please post your command here ?
thanks for your reply. I use this command to run multi-gpus:
`CUDA_VISIBLE_DEVICES=6,7 python -m torch.distributed.launch --nproc_per_node=2 train.py --launcher pytorch --cfg_file ./cfgs/kitti_models/pv_rcnn.yaml --batch_size 2 --extra_tag deepvrcnn`

Gus-Guo, 2020-06-25:
Sorry for the delay. Our codes are tested with python 3.6.5 and pytorch 1.1. I am changing my python version to 3.7 and reproducing the errors you encountered. I will fix that if there is really a bug there. For now you can try this solution and see whether it can help you.
1. see the pull request at https://github.com/open-mmlab/OpenLidarPerceptron/pull/100/commits/38c52139af7ced8cb88ae002330bdb9747f754f0 
and change the codes like it.
2. set the input parameter of build_dataloader function in tools/train.py as None. That means
build_dataloader(
        dataset_cfg=cfg.DATA_CONFIG,
        class_names=cfg.CLASS_NAMES,
        batch_size=args.batch_size,
        dist=dist_train, workers=args.workers,
        logger=None,
        training=True
    )

Gus-Guo, 2020-06-25:
> > Could you please post your command here ?
> > thanks for your reply. I use this command to run multi-gpus:
> > `CUDA_VISIBLE_DEVICES=6,7 python -m torch.distributed.launch --nproc_per_node=2 train.py --launcher pytorch --cfg_file ./cfgs/kitti_models/pv_rcnn.yaml --batch_size 2 --extra_tag deepvrcnn`

Sorry for the delay. Our codes are tested with python 3.6.5 and pytorch 1.1. I am changing my python version to 3.7 and reproducing the errors you encountered. I will fix that if there is really a bug there. For now you can try this solution and see whether it can help you.
1. see the pull request at https://github.com/open-mmlab/OpenLidarPerceptron/pull/100/commits/38c52139af7ced8cb88ae002330bdb9747f754f0 
and change the codes like it.
2. set the input parameter of build_dataloader function in tools/train.py as None. That means
build_dataloader(
        dataset_cfg=cfg.DATA_CONFIG,
        class_names=cfg.CLASS_NAMES,
        batch_size=args.batch_size,
        dist=dist_train, workers=args.workers,
        logger=None,
        training=True
    )

LittleYuanzi, 2020-06-25:
> > > Could you please post your command here ?
> > > thanks for your reply. I use this command to run multi-gpus:
> > > `CUDA_VISIBLE_DEVICES=6,7 python -m torch.distributed.launch --nproc_per_node=2 train.py --launcher pytorch --cfg_file ./cfgs/kitti_models/pv_rcnn.yaml --batch_size 2 --extra_tag deepvrcnn`
> 
> Sorry for the delay. Our codes are tested with python 3.6.5 and pytorch 1.1. I am changing my python version to 3.7 and reproducing the errors you encountered. I will fix that if there is really a bug there. For now you can try this solution and see whether it can help you.
> 
> 1. see the pull request at [38c5213](https://github.com/open-mmlab/OpenLidarPerceptron/commit/38c52139af7ced8cb88ae002330bdb9747f754f0)
>    and change the codes like it.
> 2. set the input parameter of build_dataloader function in tools/train.py as None. That means
>    build_dataloader(
>    dataset_cfg=cfg.DATA_CONFIG,
>    class_names=cfg.CLASS_NAMES,
>    batch_size=args.batch_size,
>    dist=dist_train, workers=args.workers,
>    logger=None,
>    training=True
>    )

Thanks for your help, I am trying now

closed 2020-06-26 by LittleYuanzi

liyang0522, 2020-07-17:
have you solved this issue? how did you do that? I have the same problem! Thank you for explaining it !

liyang0522, 2020-07-17:
have you solved this issue? how did you do that? I have the similar problem! Thank you for explaining it !
`Traceback (most recent call last):
  File "train.py", line 172, in <module>
    main()
  File "train.py", line 164, in main
    merge_all_iters_to_one_epoch=args.merge_all_iters_to_one_epoch
  File "/data/amax/b510/ly/OpenPCDet/tools/train_utils/train_utils.py", line 75, in train_model
    dataloader_iter = iter(train_loader)
  File "/data/amax/b510/ly/.conda/envs/new_pvrcnn/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 279, in __iter__
    return _MultiProcessingDataLoaderIter(self)
  File "/data/amax/b510/ly/.conda/envs/new_pvrcnn/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 719, in __init__
    w.start()
  File "/data/amax/b510/ly/.conda/envs/new_pvrcnn/lib/python3.6/multiprocessing/process.py", line 105, in start
    self._popen = self._Popen(self)
  File "/data/amax/b510/ly/.conda/envs/new_pvrcnn/lib/python3.6/multiprocessing/context.py", line 223, in _Popen
    return _default_context.get_context().Process._Popen(process_obj)
  File "/data/amax/b510/ly/.conda/envs/new_pvrcnn/lib/python3.6/multiprocessing/context.py", line 284, in _Popen
    return Popen(process_obj)
  File "/data/amax/b510/ly/.conda/envs/new_pvrcnn/lib/python3.6/multiprocessing/popen_spawn_posix.py", line 32, in __init__
    super().__init__(process_obj)
  File "/data/amax/b510/ly/.conda/envs/new_pvrcnn/lib/python3.6/multiprocessing/popen_fork.py", line 19, in __init__
    self._launch(process_obj)
  File "/data/amax/b510/ly/.conda/envs/new_pvrcnn/lib/python3.6/multiprocessing/popen_spawn_posix.py", line 47, in _launch
    reduction.dump(process_obj, fp)
  File "/data/amax/b510/ly/.conda/envs/new_pvrcnn/lib/python3.6/multiprocessing/reduction.py", line 60, in dump
    ForkingPickler(file, protocol).dump(obj)
TypeError: can't pickle _thread.RLock objects
Traceback (most recent call last):
  File "/data/amax/b510/ly/.conda/envs/new_pvrcnn/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/data/amax/b510/ly/.conda/envs/new_pvrcnn/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/data/amax/b510/ly/.conda/envs/new_pvrcnn/lib/python3.6/site-packages/torch/distributed/launch.py", line 263, in <module>
    main()
  File "/data/amax/b510/ly/.conda/envs/new_pvrcnn/lib/python3.6/site-packages/torch/distributed/launch.py", line 259, in main
    cmd=cmd)
subprocess.CalledProcessError: Command '['/data/amax/b510/ly/.conda/envs/new_pvrcnn/bin/python', '-u', 'train.py', '--local_rank=1', '--launcher', 'pytorch', '--cfg_file', '/data/amax/b510/ly/OpenPCDet/tools/cfgs/kitti_models/pv_rcnn.yaml', '--batch_size', '2', '--epochs', '80']' returned non-zero exit status 1.
`

Gus-Guo, 2020-07-17:
You can try to pull the newest codes. And set the input logger of build_dataloader function in tools/train.py as None. That means
build_dataloader(
dataset_cfg=cfg.DATA_CONFIG,
class_names=cfg.CLASS_NAMES,
batch_size=args.batch_size,
dist=dist_train, workers=args.workers,
logger=None,
training=True
)