open-mmlab/OpenPCDet #101: Got "TypeError: can't pickle _thread.RLock objects" when training with multiple GPUs

tn00364361, 2020-06-26:
I was trying to train with 4 GPUs using the following command:
```
# ./scripts/dist_train.sh 4 --cfg_file cfgs/kitti_models/PartA2.yaml --batch_size 16 --epoch 80
````
but got

```
2020-06-26 21:33:39,800   INFO  **********************Start training kitti_models/PartA2(default)**********************
epochs:   0%|                                                                                                                                                                                           | 0/80 [00:00<?, ?it/s]
Traceback (most recent call last):
  File "train.py", line 166, in <module>
    main()
  File "train.py", line 158, in main
    merge_all_iters_to_one_epoch=args.merge_all_iters_to_one_epoch
  File "/olp/tools/train_utils/train_utils.py", line 75, in train_model
    dataloader_iter = iter(train_loader)
  File "/opt/conda/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 193, in __iter__
    return _DataLoaderIter(self)
  File "/opt/conda/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 469, in __init__
Traceback (most recent call last):
  File "train.py", line 166, in <module>
Traceback (most recent call last):
  File "train.py", line 166, in <module>
    w.start()
  File "/opt/conda/lib/python3.6/multiprocessing/process.py", line 105, in start
epochs:   0%|                                                                                                                                                                                           | 0/80 [00:00<?, ?it/s$
    self._popen = self._Popen(self)
  File "/opt/conda/lib/python3.6/multiprocessing/context.py", line 223, in _Popen
    return _default_context.get_context().Process._Popen(process_obj)
  File "/opt/conda/lib/python3.6/multiprocessing/context.py", line 284, in _Popen
    return Popen(process_obj)
  File "/opt/conda/lib/python3.6/multiprocessing/popen_spawn_posix.py", line 32, in __init__
    super().__init__(process_obj)
  File "/opt/conda/lib/python3.6/multiprocessing/popen_fork.py", line 19, in __init__
    self._launch(process_obj)
  File "/opt/conda/lib/python3.6/multiprocessing/popen_spawn_posix.py", line 47, in _launch
    reduction.dump(process_obj, fp)
  File "/opt/conda/lib/python3.6/multiprocessing/reduction.py", line 60, in dump
    ForkingPickler(file, protocol).dump(obj)
TypeError: can't pickle _thread.RLock objects
    main()
  File "train.py", line 158, in main
    main()
  File "train.py", line 158, in main
    merge_all_iters_to_one_epoch=args.merge_all_iters_to_one_epoch
  File "/olp/tools/train_utils/train_utils.py", line 75, in train_model
    dataloader_iter = iter(train_loader)
  File "/opt/conda/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 193, in __iter__
Traceback (most recent call last):
  File "train.py", line 166, in <module>
    merge_all_iters_to_one_epoch=args.merge_all_iters_to_one_epoch
  File "/olp/tools/train_utils/train_utils.py", line 75, in train_model
    return _DataLoaderIter(self)
  File "/opt/conda/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 469, in __init__
    dataloader_iter = iter(train_loader)
  File "/opt/conda/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 193, in __iter__
    w.start()
  File "/opt/conda/lib/python3.6/multiprocessing/process.py", line 105, in start
    return _DataLoaderIter(self)
  File "/opt/conda/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 469, in __init__
    self._popen = self._Popen(self)
  File "/opt/conda/lib/python3.6/multiprocessing/context.py", line 223, in _Popen
    return _default_context.get_context().Process._Popen(process_obj)
  File "/opt/conda/lib/python3.6/multiprocessing/context.py", line 284, in _Popen
    return Popen(process_obj)
    w.start()
  File "/opt/conda/lib/python3.6/multiprocessing/popen_spawn_posix.py", line 32, in __init__
  File "/opt/conda/lib/python3.6/multiprocessing/process.py", line 105, in start
    super().__init__(process_obj)
  File "/opt/conda/lib/python3.6/multiprocessing/popen_fork.py", line 19, in __init__
    self._launch(process_obj)
  File "/opt/conda/lib/python3.6/multiprocessing/popen_spawn_posix.py", line 47, in _launch
    self._popen = self._Popen(self)
  File "/opt/conda/lib/python3.6/multiprocessing/context.py", line 223, in _Popen
    reduction.dump(process_obj, fp)
  File "/opt/conda/lib/python3.6/multiprocessing/reduction.py", line 60, in dump
    ForkingPickler(file, protocol).dump(obj)
TypeError: can't pickle _thread.RLock objects
    return _default_context.get_context().Process._Popen(process_obj)
  File "/opt/conda/lib/python3.6/multiprocessing/context.py", line 284, in _Popen
    main()
  File "train.py", line 158, in main
    merge_all_iters_to_one_epoch=args.merge_all_iters_to_one_epoch
    return Popen(process_obj)
  File "/olp/tools/train_utils/train_utils.py", line 75, in train_model
  File "/opt/conda/lib/python3.6/multiprocessing/popen_spawn_posix.py", line 32, in __init__
    dataloader_iter = iter(train_loader)
  File "/opt/conda/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 193, in __iter__
    return _DataLoaderIter(self)
  File "/opt/conda/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 469, in __init__
    super().__init__(process_obj)
  File "/opt/conda/lib/python3.6/multiprocessing/popen_fork.py", line 19, in __init__
    w.start()
  File "/opt/conda/lib/python3.6/multiprocessing/process.py", line 105, in start
    self._launch(process_obj)
  File "/opt/conda/lib/python3.6/multiprocessing/popen_spawn_posix.py", line 47, in _launch
    self._popen = self._Popen(self)
  File "/opt/conda/lib/python3.6/multiprocessing/context.py", line 223, in _Popen
    return _default_context.get_context().Process._Popen(process_obj)
  File "/opt/conda/lib/python3.6/multiprocessing/context.py", line 284, in _Popen
    reduction.dump(process_obj, fp)
  File "/opt/conda/lib/python3.6/multiprocessing/reduction.py", line 60, in dump
    return Popen(process_obj)
  File "/opt/conda/lib/python3.6/multiprocessing/popen_spawn_posix.py", line 32, in __init__
    super().__init__(process_obj)
  File "/opt/conda/lib/python3.6/multiprocessing/popen_fork.py", line 19, in __init__
    self._launch(process_obj)
  File "/opt/conda/lib/python3.6/multiprocessing/popen_spawn_posix.py", line 47, in _launch
    reduction.dump(process_obj, fp)
  File "/opt/conda/lib/python3.6/multiprocessing/reduction.py", line 60, in dump
    ForkingPickler(file, protocol).dump(obj)
    ForkingPickler(file, protocol).dump(obj)
TypeError: can't pickle _thread.RLock objects
TypeError: can't pickle _thread.RLock objects
Traceback (most recent call last):
  File "/opt/conda/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/opt/conda/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/opt/conda/lib/python3.6/site-packages/torch/distributed/launch.py", line 235, in <module>
    main()
  File "/opt/conda/lib/python3.6/site-packages/torch/distributed/launch.py", line 231, in main
    cmd=process.args)
subprocess.CalledProcessError: Command '['/opt/conda/bin/python', '-u', 'train.py', '--local_rank=0', '--launcher', 'pytorch', '--cfg_file', 'cfgs/kitti_models/PartA2.yaml', '--batch_size', '16', '--epoch', '80']' returned
non-zero exit status 1.
```

I am using a docker image based on `pytorch/pytorch:1.1.0-cuda10.0-cudnn7.5-devel` and the Python version is 3.6.8.

Training on one GPU with a batch size of 4 works. I don't know why it won't work with multiple GPUs. Please advise.

ChengyaoLi, 2020-06-27:
I got same error with torch 1.3 and Python 3.6.8.

sshaoshuai, 2020-06-27:
Have you tried to pull the latest codes? We have tried to fix it as in issue https://github.com/open-mmlab/OpenLidarPerceptron/issues/98.

tn00364361, 2020-06-27:
I can confirm that using commit 12b8c131bb69c514d00c62834dde869b69c0e5f5 resolved this issue. Thanks!

closed 2020-06-30 by sshaoshuai