open-mmlab/OpenPCDet #102: Running the model on the whole point cloud

MohamedRamzy1, 2020-06-28:
Hi
I want to run a model trained on kitti on the whole point cloud not just the points in the FOV. I tried to set FOV_POINTS_ONLY with False but it still predicts boxes for just objects inside FOV

sshaoshuai, 2020-06-30:
Note that KITTI only has ground-truth annotations in the front view within the image area, so you could not train KITTI with the whole point cloud since it doesn't have labels in the surrounding areas except the front view. `FOV_POINTS_ONLY` is only used to decide whether to clip the points within the image area, and it is useless in this case.

closed 2020-07-01 by sshaoshuai

josy43, 2020-09-01:
That doesn´t seem correct to me, because the demo predicts also objects on the left and right of the origin. But if I run test.py it doenst show me the labels for the objects on the left and right.