open-mmlab/OpenPCDet #104: Where to get kitti_dbinfos_train.pkl file?

Number1JT, 2020-06-28:
When I run 
`python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos tools/cfgs/dataset_configs/kitti_dataset.yaml
`
always raise Error below, I wonder what kitti_dbinfos_train.pkl actually is and how to generate/download this file, THX.
`Traceback (most recent call last):
  File "/home/jiangtao.jiangtao/envoriment/local/anaconda3/envs/LidarDet/lib/python3.6/site-packages/ipdb/__main__.py", line 276, in main
    pdb._runscript(mainpyfile)
  File "/home/jiangtao.jiangtao/envoriment/local/anaconda3/envs/LidarDet/lib/python3.6/pdb.py", line 1548, in _runscript
    self.run(statement)
  File "/home/jiangtao.jiangtao/envoriment/local/anaconda3/envs/LidarDet/lib/python3.6/bdb.py", line 434, in run
    exec(cmd, globals, locals)
  File "<string>", line 1, in <module>
  File "/home/jiangtao.jiangtao/code/opensource/OpenLidarPerceptron/tmp.py", line 1, in <module>
    from pcdet.datasets.kitti.kitti_dataset import create_kitti_infos
  File "/home/jiangtao.jiangtao/code/opensource/OpenLidarPerceptron/pcdet/datasets/kitti/kitti_dataset.py", line 379, in create_kitti_infos
    dataset = KittiDataset(dataset_cfg=dataset_cfg, class_names=class_names, root_path=data_path)
  File "/home/jiangtao.jiangtao/code/opensource/OpenLidarPerceptron/pcdet/datasets/kitti/kitti_dataset.py", line 22, in __init__
    dataset_cfg=dataset_cfg, class_names=class_names, training=training, root_path=root_path, logger=logger
  File "/home/jiangtao.jiangtao/code/opensource/OpenLidarPerceptron/pcdet/datasets/dataset.py", line 30, in __init__
    ) if self.training else None
  File "/home/jiangtao.jiangtao/code/opensource/OpenLidarPerceptron/pcdet/datasets/augmentor/data_augmentor.py", line 24, in __init__
    cur_augmentor = getattr(self, cur_cfg.NAME)(config=cur_cfg)
  File "/home/jiangtao.jiangtao/code/opensource/OpenLidarPerceptron/pcdet/datasets/augmentor/data_augmentor.py", line 32, in gt_sampling
    logger=self.logger
  File "/home/jiangtao.jiangtao/code/opensource/OpenLidarPerceptron/pcdet/datasets/augmentor/database_sampler.py", line 19, in __init__
    with open(str(db_info_path), 'rb') as f:
FileNotFoundError: [Errno 2] No such file or directory: 'data/kitti/kitti_dbinfos_train.pkl'
Uncaught exception. Entering post mortem debugging`

sshaoshuai, 2020-06-30:
Have you tried to pull the latest codes? 
If not, please pull the latest codes and run it again. We fix a similar error in https://github.com/open-mmlab/OpenLidarPerceptron/commit/c6302949d86d83d59e562416d1fa046879fea908.

Number1JT, 2020-06-30:
> Have you tried to pull the latest codes?
> If not, please pull the latest codes and run it again. We fix a similar error in [c630294](https://github.com/open-mmlab/OpenLidarPerceptron/commit/c6302949d86d83d59e562416d1fa046879fea908).

Thanks for reply, we already solve this problem

closed 2020-06-30 by Number1JT

cuge1995, 2020-10-13:
> > Have you tried to pull the latest codes?
> > If not, please pull the latest codes and run it again. We fix a similar error in [c630294](https://github.com/open-mmlab/OpenLidarPerceptron/commit/c6302949d86d83d59e562416d1fa046879fea908).
> 
> Thanks for reply, we already solve this problem

Can you share how you solved this problem ?

GeneralJing, 2022-05-12:
request this file too.

DurgaTejaGunupudi, 2025-04-12:
> > Have you tried to pull the latest codes?
> > If not, please pull the latest codes and run it again. We fix a similar error in [c630294](https://github.com/open-mmlab/OpenLidarPerceptron/commit/c6302949d86d83d59e562416d1fa046879fea908).
> 
> Thanks for reply, we already solve this problem

Did you generated this file or downloaded? How did you do this?