open-mmlab/OpenPCDet #106: AttributeError: 'EasyDict' object has no attribute 'DATA_SPLIT'

MRRRKING, 2020-06-29:
Thanks for your codes.
In the process of training and testing, I encountered some problems. what should I do？Thanks a lot.
PS: I used the model you provided.

1. test
root@da5e3c517fb2:/code/OpenLidarPerceptron/tools# **python test.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml --batch_size 8 --ckpt ../pretrained_models/pv_rcnn_8369.pth** 
/code/OpenLidarPerceptron/pcdet/config.py:73: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  new_config = yaml.load(f)
/code/OpenLidarPerceptron/pcdet/config.py:54: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  yaml_config = yaml.load(f)
Traceback (most recent call last):
  File "test.py", line 189, in <module>
    main()
  File "test.py", line 148, in main
    eval_output_dir = eval_output_dir / ('epoch_%s' % epoch_id) / cfg.DATA_CONFIG.DATA_SPLIT['test']
**AttributeError: 'EasyDict' object has no attribute 'DATA_SPLIT'**

2. train
root@da5e3c517fb2:/code/OpenLidarPerceptron/tools# **python train.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml --batch_size 8 --epoch 50**
Traceback (most recent call last):
  File "train.py", line 166, in <module>
    main()
  File "train.py", line 99, in main
    training=True
  File "/code/OpenLidarPerceptron/pcdet/datasets/__init__.py", line 40, in build_dataloader
    dataset = __all__[dataset_cfg.DATASET](
**AttributeError: 'EasyDict' object has no attribute 'DATASET'**

MRRRKING, 2020-06-29:
The reason has been found. The version of EasyDict is too low, and it will be fine after the update.

sshaoshuai, 2020-06-30:
Since it is solved now, I will close this issue. Feel free to re-open it if it is needed.

closed 2020-06-30 by sshaoshuai

Cknex, 2021-03-14:
I have a similar problem:
AttributeError: 'EasyDict' object has no attribute 'DATA_CONFIG'
but my easydict version is 1.9.

What's your version?

sychou11, 2022-01-13:
Hi, I have update easydict to 1.9, but it still turned out to be this error. Could please tell me what version of easydict have you used?

Baymax1520301, 2024-05-20:
> 嗨，我已将 easydict 更新到 1.9，但结果仍然是这个错误。你能告诉我你用过什么版本的easydict吗？

May I ask if you have solved this problem? I have also encountered it and currently do not know how to solve it