open-mmlab/OpenPCDet #107: A question about Dataset Preparation

Blinjuan, 2020-06-30:
hi，thanks for your excellent work，however，when i try to prepare my dataset by command 
“python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos tools/cfgs/dataset_configs/kitti_dataset.yaml”
it always occur the error as follow：
“gt_database sample: 1/3712
Segmentation fault (core dumped)”
[image]
i don't know the reason ，i try the command many times  and did't fix i

Gus-Guo, 2020-07-01:
Hi, I see that you have already generated the kitti infos pickle file successfully and the problem comes out when it wanted to generate the groundtruth database. You can try to comment the codes for generating kitti infos,  only run the codes for generating database and see whether it can work.

Blinjuan, 2020-07-01:
> 
> 
> Hi, I see that you have already generated the kitti infos pickle file successfully and the problem comes out when it wanted to generate the groundtruth database. You can try to comment the codes for generating kitti infos, only run the codes for generating database and see whether it can work.

Thanks for your reply，I try to  comment the codes for generating kitti infos。
[image]
however,it does't work
[image]

sshaoshuai, 2020-07-01:
I guess the error occurs at `roiaware_pool3d_utils.points_in_boxes_cpu`. 
Maybe there are some problems in your compilation when you run `python setup.py develop`.  So could you post the compilation logs to see if there are some problems?

Blinjuan, 2020-07-01:
> 
> 
> I guess the error occurs at `roiaware_pool3d_utils.points_in_boxes_cpu`.
> Maybe there are some problems in your compilation when you run `python setup.py develop`. So could you post the compilation logs to see if there are some problems?
After run command “python setup.py develop” i get this
[image]
[image]
[image]

Blinjuan, 2020-07-01:
> 
> 
> I guess the error occurs at `roiaware_pool3d_utils.points_in_boxes_cpu`.
> Maybe there are some problems in your compilation when you run `python setup.py develop`. So could you post the compilation logs to see if there are some problems?
after  run the command “python setup.py develop” 
[image]
[image]

my environment as follow：
[image]

when i try to debug，i find  in the roiaware_pool3d_utils.py around 20 line "roiaware_pool3d_cuda.points_in_boxes_cpu(boxes.float().contiguous(), points.float().contiguous(), point_indices)" cause this problem

sshaoshuai, 2020-07-01:
You should remove the `build` directory first in your root path of PCDet and then re-run `python setup.py develop`, since your current log doesn't have useful information for debugging. 
The error position shows that there are some problems in your compilation process, and you need to pay more attention to the errors or warnings of your compilation log.

Blinjuan, 2020-07-01:
> You should remove the `build` directory first in your root path of PCDet and then re-run `python setup.py develop`, since your current log doesn't have useful information for debugging.
> The error [...]
ouch~~~ Thank you very much. I understand the reason now，I change gcc4.8.5 to gcc5.4.0. It works!  it seems that we should use gcc5.4.0.
Thank you.

closed 2020-07-01 by Blinjuan