open-mmlab/OpenPCDet #108: Number of Epochs to train for

Divadi, 2020-06-30:
Hello, to reproduce results for PV-RCNN (and other methods) on v0.2, how many epochs should we train for?
I recall in v0.1, it was 80 epochs (as in the paper), but it looks like the LR_DECAY has changed, and I was wondering if the recommended max epoch changed as well.

Thank you for your work!

Gus-Guo, 2020-07-01:
Hi, you can train the models for 80 epochs as in the paper.

sshaoshuai, 2020-07-01:
The `LR_DECAY` is useless if you trained with the default one-cycle strategy.

Divadi, 2020-07-02:
I had trained on the old generated info files accidentally.
However, even when I toggled the "DATABASE_WITH_FAKELIDAR" option, the results did not change very much. There was still a 2 - 3 point gap.
When I regenerated the info files, I was able to get good results for car & cyclist.

However, for pedestrian, do you have any suggestions for changes to get better results?
I see that I can add another upsampling layer here
https://github.com/open-mmlab/OpenPCDet/blob/be9ea5670339f283c292cce56289aaf1b890e842/tools/cfgs/kitti_models/pv_rcnn.yaml#L49
that affects
https://github.com/open-https://github.com/open-mmlab/OpenPCDet/blob/be9ea5670339f283c292cce56289aaf1b890e842/pcdet/models/backbones_2d/base_bev_backbone.py#L51-L56

Would you recommend adding an upsampling here or adjusting the backbone so that it's 4x instead of 8x?

sshaoshuai, 2020-07-05:
For pedestrian detection, generally a higher resolution will achieve better results, and we also saw this affects in KITTI dataset, but it may still not be stable, you could have a try.

Divadi, 2020-07-05:
I tried with higher resolution and got good results for pedestrian.
It seems like good results on Cyclist are achievable with the multi-class training optimized for Car.

Thank you for your help!

closed 2020-07-05 by Divadi

ys920826, 2021-04-12:
> I tried with higher resolution and got good results for pedestrian.
> It seems like good results on Cyclist are achievable with the multi-class training optimized for Car.
> 
> Thank you for your help!

@Divadi  Hi, you mentioned 'higher resolution' and 'adjusting the backbone so that it's 4x instead of 8x', can you explain in detail how to set the parameters, I also care about pedestrian data, thank you very much