open-mmlab/OpenPCDet #110: About using the old generated data.

ltphy, 2020-07-01:
The option 'DATABASE_WITH_FAKE_LIDAR' is only applied when augmenting boxes from the list of sampled boxes. The old version of test and train pickle files cannot be used (different in the boxes center and the angle of ground truth boxes). I think the only way is to used the new version to generate new train, test, val pickle files? Thank you.

sshaoshuai, 2020-07-01:
The old train/test pickle infos shoud be ok. You could refer to here https://github.com/open-mmlab/OpenPCDet/blob/master/pcdet/datasets/kitti/kitti_dataset.py#L361, where you could see that we use `location`, `dimensions`, `rotation_y` (they are in the KITTI camera coordinate) to generate the boxes. 

So if you would like to use old version pickle files, please set `DATABASE_WITH_FAKE_LIDAR=True`. But I suggest to generate new pickle files to avoid potential misunderstanding.

ltphy, 2020-07-01:
Thank you very much. I understand the situation now. I think I should generate new pickle files.

Let my verify, either use the pickle file from previous version or use the new version one, boxes in camera coordinate have to be converted into lidar coordinate with the new lidar boxes' configuration when feeding into the model.

sshaoshuai, 2020-07-01:
Yes, you are right.

closed 2020-07-01 by ltphy