open-mmlab/OpenPCDet #113: error: command 'x86_64-linux-gnu-gcc' failed with exit status 1

sakura-iv, 2020-07-02:
Hello, when I ran the ' python setup.py develop' , I met an error 'error: command 'x86_64-linux-gnu-gcc' failed with exit status 1'. I have searched the Internet for some time, but I did not find a suitable way to deal with it. I just wanna ask why this error happens and what I could do with it? Many thanks!!!

`/usr/local/lib/python3.8/dist-packages/setuptools/dist.py:454: UserWarning: Normalizing '0.2.0+0000000' to '0.2.0+0'
  warnings.warn(tmpl.format(**locals()))
running develop
running egg_info
writing pcdet.egg-info/PKG-INFO
writing dependency_links to pcdet.egg-info/dependency_links.txt
writing requirements to pcdet.egg-info/requires.txt
writing top-level names to pcdet.egg-info/top_level.txt
/usr/local/lib/python3.8/dist-packages/torch/utils/cpp_extension.py:305: UserWarning: Attempted to use ninja as the BuildExtension backend but we could not find ninja.. Falling back to using the slow distutils backend.
  warnings.warn(msg.format('we could not find ninja.'))
reading manifest file 'pcdet.egg-info/SOURCES.txt'
writing manifest file 'pcdet.egg-info/SOURCES.txt'
running build_ext
building 'pcdet.ops.iou3d_nms.iou3d_nms_cuda' extension
creating build/temp.linux-x86_64-3.8
creating build/temp.linux-x86_64-3.8/pcdet
creating build/temp.linux-x86_64-3.8/pcdet/ops
creating build/temp.linux-x86_64-3.8/pcdet/ops/iou3d_nms
creating build/temp.linux-x86_64-3.8/pcdet/ops/iou3d_nms/src
x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/local/lib/python3.8/dist-packages/torch/include -I/usr/local/lib/python3.8/dist-packages/torch/include/torch/csrc/api/include -I/usr/local/lib/python3.8/dist-packages/torch/include/TH -I/usr/local/lib/python3.8/dist-packages/torch/include/THC -I/usr/local/cuda/include -I/usr/include/python3.8 -c pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp -o build/temp.linux-x86_64-3.8/pcdet/ops/iou3d_nms/src/iou3d_cpu.o -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=iou3d_nms_cuda -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++14
pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp: In function ‘int boxes_iou_bev_cpu(at::Tensor, at::Tensor, at::Tensor)’:
pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp:16:29: error: ‘AT_CHECK’ was not declared in this scope
 #define CHECK_CONTIGUOUS(x) AT_CHECK(x.is_contiguous(), #x, " must be contiguous ")
                             ^
pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp:16:29: note: in definition of macro ‘CHECK_CONTIGUOUS’
 #define CHECK_CONTIGUOUS(x) AT_CHECK(x.is_contiguous(), #x, " must be contiguous ")
                             ^~~~~~~~
pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp:16:29: note: suggested alternative: ‘DCHECK’
 #define CHECK_CONTIGUOUS(x) AT_CHECK(x.is_contiguous(), #x, " must be contiguous ")
                             ^
pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp:16:29: note: in definition of macro ‘CHECK_CONTIGUOUS’
 #define CHECK_CONTIGUOUS(x) AT_CHECK(x.is_contiguous(), #x, " must be contiguous ")
                             ^~~~~~~~
pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp:230:55: warning: ‘T* at::Tensor::data() const [with T = float]’ is deprecated: Tensor.data<T>() is deprecated. Please use Tensor.data_ptr<T>() instead. [-Wdeprecated-declarations]
     const float *boxes_a = boxes_a_tensor.data<float>();
                                                       ^
In file included from /usr/local/lib/python3.8/dist-packages/torch/include/ATen/Tensor.h:11:0,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/ATen/Context.h:4,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/ATen/ATen.h:5,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/torch/csrc/api/include/torch/types.h:3,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/torch/csrc/api/include/torch/serialize/input-archive.h:6,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/torch/csrc/api/include/torch/serialize/archive.h:3,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/torch/csrc/api/include/torch/serialize/tensor.h:3,
                 from pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp:9:
/usr/local/lib/python3.8/dist-packages/torch/include/ATen/core/TensorBody.h:341:7: note: declared here
   T * data() const {
       ^~~~
pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp:231:55: warning: ‘T* at::Tensor::data() const [with T = float]’ is deprecated: Tensor.data<T>() is deprecated. Please use Tensor.data_ptr<T>() instead. [-Wdeprecated-declarations]
     const float *boxes_b = boxes_b_tensor.data<float>();
                                                       ^
In file included from /usr/local/lib/python3.8/dist-packages/torch/include/ATen/Tensor.h:11:0,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/ATen/Context.h:4,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/ATen/ATen.h:5,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/torch/csrc/api/include/torch/types.h:3,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/torch/csrc/api/include/torch/serialize/input-archive.h:6,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/torch/csrc/api/include/torch/serialize/archive.h:3,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/torch/csrc/api/include/torch/serialize/tensor.h:3,
                 from pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp:9:
/usr/local/lib/python3.8/dist-packages/torch/include/ATen/core/TensorBody.h:341:7: note: declared here
   T * data() const {
       ^~~~
pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp:232:49: warning: ‘T* at::Tensor::data() const [with T = float]’ is deprecated: Tensor.data<T>() is deprecated. Please use Tensor.data_ptr<T>() instead. [-Wdeprecated-declarations]
     float *ans_iou = ans_iou_tensor.data<float>();
                                                 ^
In file included from /usr/local/lib/python3.8/dist-packages/torch/include/ATen/Tensor.h:11:0,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/ATen/Context.h:4,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/ATen/ATen.h:5,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/torch/csrc/api/include/torch/types.h:3,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/torch/csrc/api/include/torch/serialize/input-archive.h:6,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/torch/csrc/api/include/torch/serialize/archive.h:3,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/torch/csrc/api/include/torch/serialize/tensor.h:3,
                 from pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp:9:
/usr/local/lib/python3.8/dist-packages/torch/include/ATen/core/TensorBody.h:341:7: note: declared here
   T * data() const {
       ^~~~
error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
`

sshaoshuai, 2020-07-02:
What's your pytorch version, GCC version, and CUDA version?

sakura-iv, 2020-07-02:
pytorch 1.5.1
cuda 10.1
GCC 7.4.0
Do I have some problems with these tools' versions? Is there some conflict between them? thanks !

sshaoshuai, 2020-07-02:
Currently this repo could only be used with pytorch 1.1/1.2, since there are some writing changes of the custom c++/cuda operations after pytorch 1.3. You could try to create a new environment with pytorch 1.1/1.2 for running with PCDet.

sakura-iv, 2020-07-02:
OK, I will install a lower version of pytorch. I just saw that 'pytorch 1.1 or higher' in the requirements  so I thought the latest version is ok. Thank you a lot !!

closed 2020-07-02 by sakura-iv