open-mmlab/OpenPCDet #123: config for pedestrian and cyclist

KPeng9510, 2020-07-03:
Hi, could you provide a template of the config file just for pedestrain and cyclist (pointpillars)? The test result for pedestrian which is trained together with car and cyclist is a little worse. 
Thank you very much!

sshaoshuai, 2020-07-05:
You could simply remove the car related config for a pure detection of pedestrian and/or cyclist. The results of pedestrian/cyclist of KITTI dataset are not stable and need to be further tuned for a better performance. 
The configs we provided focus on a higher performance of car, we simply add the predictions of three classes for a full detection results, and do not tune the ped/cyc KITTI results with this new repo yet.

closed 2020-07-06 by sshaoshuai

josy43, 2020-08-18:
Is there any pretrained file which considers also the other classes of KITTI, for example Tram and Truck ?