open-mmlab/OpenPCDet #125: how to test on the  KITTI test split

suqi10, 2020-07-05:
running  on the version 0.1
I change the "TEST.INFO_PATH.kitti_infos_val.pkl" to "kitti_infos_test.pkl"
but when testing using "python test.py --cfg_file cfgs/PartA2_car.yaml --batch_size 4 --eval_all ",after finishing one epoch
The error appears as follows
eval: 100%|██████████| 1880/1880 [06:45<00:00,  4.64it/s, recall_0.5=(0, 0) / 0]
2020-07-05 19:49:32,040   INFO  *************** Performance of EPOCH 66 *****************
2020-07-05 19:49:32,041   INFO  Generate label finished(sec_per_example: 0.0539 second).
2020-07-05 19:49:32,041   INFO  recall_roi_0.5: 0.000000
2020-07-05 19:49:32,041   INFO  recall_rcnn_0.5: 0.000000
2020-07-05 19:49:32,041   INFO  recall_roi_0.7: 0.000000
2020-07-05 19:49:32,041   INFO  recall_rcnn_0.7: 0.000000
2020-07-05 19:49:32,045   INFO  Average predicted number of objects(7518 samples): 6.361
Traceback (most recent call last):
  File "test.py", line 182, in <module>
    main()
  File "test.py", line 176, in main
    repeat_eval_ckpt(model, test_loader, args, eval_output_dir, logger, ckpt_dir)
  File "test.py", line 114, in repeat_eval_ckpt
    model, test_loader, cur_epoch_id, logger, result_dir=cur_result_dir, save_to_file=args.save_to_file
  File "/home/wenhao/program/PCDet_0/tools/eval_utils/eval_utils.py", line 82, in eval_one_epoch
    result_str, result_dict = dataset.evaluation(det_annos, class_names, eval_metric=cfg.MODEL.TEST.EVAL_METRIC)
  File "/home/wenhao/program/PCDet_0/pcdet/datasets/kitti/kitti_dataset.py", line 329, in evaluation
    assert 'annos' in self.kitti_infos[0].keys()
AssertionError

sshaoshuai, 2020-07-05:
Please pull the latest codes and re-run it, since I have fixed this problem in https://github.com/open-mmlab/OpenPCDet/commit/b32fbddbe06183507bad433ed99b407cbc2175c2. 
Please also add `--save_to_file` config when you run `python test.py xxxx` for saving the predictions.

suqi10, 2020-07-06:
@sshaoshuai  thanks, it works

closed 2020-07-06 by sshaoshuai

jialeli1, 2020-08-18:
Hey, I have run test.py with --save_to_file, but it still tests the result on the val set. 
And I couldn't find where I can set DATA_SPLIT['test'] = 'test', as you mentioned in #89.
Thank you in advance for your help. @sshaoshuai @suqi10

jihanyang, 2020-08-18:
@jialeli1 Please set INFO_PATH to `kitti_infos_test.pkl`

jialeli1, 2020-08-18:
> @jialeli1 Please set INFO_PATH to `kitti_infos_test.pkl`

Thanks a lot. 
I still can't find where I can set INFO_PATH, after I searched the entire directory.
Do you mean it is in *.yaml or in which *.py file?

jihanyang, 2020-08-18:
try to check *.yaml in dataset_configs

jialeli1, 2020-08-18:
> try to check *.yaml in dataset_configs

Oh, I get it. 
There are two kinds of configs (*.yaml). INFO_PATH is under dataset_configs/.
It works now.
Thank you very much!

MengAaron, 2020-08-24:
I run on version 0.3. And I want to test my model on test split. Following method above, It finished with no error, but also no output.
2020-08-24 06:31:01,696   INFO  ==> Loading parameters from checkpoint ../output/kitti_models/second/default/ckpt/checkpoint_epoch_80.pth to GPU
2020-08-24 06:31:01,746   INFO  ==> Checkpoint trained from version: pcdet+0.3.0+9e460da
2020-08-24 06:31:02,102   INFO  ==> Done (loaded 163/163)
2020-08-24 06:31:02,149   INFO  *************** EPOCH 80 EVALUATION *****************
eval: 100%|##############################| 1880/1880 [07:15<00:00,  4.32it/s, recall_0.3=(0, 0) / 0]
2020-08-24 06:38:17,607   INFO  *************** Performance of EPOCH 80 *****************
2020-08-24 06:38:17,607   INFO  Generate label finished(sec_per_example: 0.0579 second).
2020-08-24 06:38:17,607   INFO  recall_roi_0.3: 0.000000
2020-08-24 06:38:17,607   INFO  recall_rcnn_0.3: 0.000000
2020-08-24 06:38:17,607   INFO  recall_roi_0.5: 0.000000
2020-08-24 06:38:17,608   INFO  recall_rcnn_0.5: 0.000000
2020-08-24 06:38:17,608   INFO  recall_roi_0.7: 0.000000
2020-08-24 06:38:17,608   INFO  recall_rcnn_0.7: 0.000000
2020-08-24 06:38:17,623   INFO  Average predicted number of objects(7518 samples): 15.337
2020-08-24 06:38:18,507   INFO  None
2020-08-24 06:38:18,507   INFO  Result is save to /remote-home/my/OpenPCDet/output/kitti_models/second/default/eval/epoch_80/test/default
2020-08-24 06:38:18,508   INFO  ****************Evaluation done.*****************

jialeli1, 2020-08-24:
> I run on version 0.3. And I want to test my model on test split. Following method above, It finished with no error, but also no output.
> 2020-08-24 06:31:01,696 INFO ==> Loading parameters from [...]

Please also add --save_to_file config when you run python test.py xxxx for saving the predictions (xxxxxx.txt).

suqi10, 2020-08-31:
@jialeli1  after adding --save_to_file, but also no output 
2020-08-31 14:33:17,978   INFO  Generate label finished(sec_per_example: 0.0207 second).
2020-08-31 14:33:17,978   INFO  recall_roi_0.3: 0.000000
2020-08-31 14:33:17,978   INFO  recall_rcnn_0.3: 0.000000
2020-08-31 14:33:17,978   INFO  recall_roi_0.5: 0.000000
2020-08-31 14:33:17,978   INFO  recall_rcnn_0.5: 0.000000
2020-08-31 14:33:17,978   INFO  recall_roi_0.7: 0.000000
2020-08-31 14:33:17,978   INFO  recall_rcnn_0.7: 0.000000
2020-08-31 14:33:17,981   INFO  Average predicted number of objects(7518 samples): 16.609
2020-08-31 14:33:18,289   INFO  None
2020-08-31 14:33:18,289   INFO  Result is save to /home/hwh/program/OpenPCDet_0/OpenPCDet_0/output/kitti_models/pointpillar/default/eval/epoch_7728/test/default
2020-08-31 14:33:18,289   INFO  ****************Evaluation done.****************

arxidinakbar, 2020-11-27:
Hi, I'm trying to evaluate KITTI testing set with trained model, and follow the instructions from your discussions, I had the results below. There is two question want to discuss:
1. I've notice the recall in every evaluation was 0.0000(same as yours above), not sure if it is normal, because it did generate final_results.
2. Could I generate the AP_40(AP_11) results from the txt files in final_result?
`2020-11-26 21:13:52,581   INFO  *************** EPOCH 79 EVALUATION *****************                                                                                                        2020-11-26 21:23:27,695   INFO  *************** Performance of EPOCH 79 *****************                                                                                                    2020-11-26 21:23:27,695   INFO  Generate label finished(sec_per_example: 0.0765 second).                                                                                                     2020-11-26 21:23:27,695   INFO  recall_roi_0.3: 0.000000                                                                                                                                     2020-11-26 21:23:27,695   INFO  recall_rcnn_0.3: 0.000000                                                                                                                                    2020-11-26 21:23:27,695   INFO  recall_roi_0.5: 0.000000                                                                                                                                     2020-11-26 21:23:27,695   INFO  recall_rcnn_0.5: 0.000000                                                                                                                                    2020-11-26 21:23:27,695   INFO  recall_roi_0.7: 0.000000                                                                                                                                     2020-11-26 21:23:27,695   INFO  recall_rcnn_0.7: 0.000000                                                                                                                                    2020-11-26 21:23:27,701   INFO  Average predicted number of objects(7518 samples): 9.974                                                                                                     2020-11-26 21:23:28,051   INFO  None                                                                                                                                                         2020-11-26 21:23:28,051   INFO  Result is save to /home/omnisky/aixiding/OpenPCDet/output/kitti_models/pv_rcnn/default/eval/eval_all_default/default/epoch_79/test                           2020-11-26 21:23:28,051   INFO  ****************Evaluation done.*****************                                                                                                            2020-11-26 21:23:28,086   INFO  Epoch 79 has been evaluated                                                                                                                                  2020-11-26 21:23:28,087   INFO  ==> Loading parameters from checkpoint /home/omnisky/aixiding/OpenPCDet/output/cfgs/kitti_models/pv_rcnn/default/ckpt/checkpoint_epoch_80.pth to GPU         2020-11-26 21:23:28,589   INFO  ==> Checkpoint trained from version: pcdet+0.3.0+f982b5b                                                                                                     2020-11-26 21:23:29,086   INFO  ==> Done (loaded 367/367)                                                                                                                                    2020-11-26 21:23:29,088   INFO  *************** EPOCH 80 EVALUATION *****************                                                                                                        2020-11-26 21:33:07,023   INFO  *************** Performance of EPOCH 80 *****************                                                                                                    2020-11-26 21:33:07,023   INFO  Generate label finished(sec_per_example: 0.0769 second).                                                                                                     2020-11-26 21:33:07,023   INFO  recall_roi_0.3: 0.000000                                                                                                                                     2020-11-26 21:33:07,023   INFO  recall_rcnn_0.3: 0.000000                                                                                                                                    2020-11-26 21:33:07,023   INFO  recall_roi_0.5: 0.000000                                                                                                                                     2020-11-26 21:33:07,023   INFO  recall_rcnn_0.5: 0.000000                                                                                                                                    2020-11-26 21:33:07,024   INFO  recall_roi_0.7: 0.000000                                                                                                                                     2020-11-26 21:33:07,024   INFO  recall_rcnn_0.7: 0.000000                                                                                                                                    2020-11-26 21:33:07,030   INFO  Average predicted number of objects(7518 samples): 9.262                                                                                             [... cut ...]                                                              2020-11-26 21:33:07,439   INFO  Result is save to /home/omnisky/aixiding/OpenPCDet/output/kitti_models/pv_rcnn/default/eval/eval_all_default/default/epoch_80/test                           2020-11-26 21:33:07,439   INFO  ****************Evaluation done.*****************                                                                                                            2020-11-26 21:33:07,484   INFO  Epoch 80 has been evaluated `

huangzhicong3, 2022-01-05:
I think the reason is that KITTI do  provide the ground truth labels of testing datasets.

DeclK, 2022-01-25:
I don't think it's possible to get the evaluation results without test split labels...

sgrrlll, 2023-10-13:
> @jialeli1 after adding --save_to_file, but also no output 2020-08-31 14:33:17,978 INFO Generate label finished(sec_per_example: 0.0207 second). 2020-08-31 14:33:17,978 INFO recall_roi_0.3: 0.000000 [...]

I also encountered the same problem, and there was no output in the generated txt file，Do you know how to solve it?