open-mmlab/OpenPCDet #128: Why shift the center upward by half the height when create_kitti_infos

zyms5244, 2020-07-06:
https://github.com/open-mmlab/OpenPCDet/blob/9fdb4435ac5375adba530b8813ef608dffb681c0/pcdet/datasets/kitti/kitti_dataset.py#L163

sshaoshuai, 2020-07-06:
It aims to transform the boxes_3d from KITTI format ((x,y,z) is the bottom center of object bounding box) to our unified normative coordinate ((x, y, z) is the 3D geometry center of object bounding box).

zyms5244, 2020-07-07:
> It aims to transform the boxes_3d from KITTI format ((x,y,z) is the bottom center of object bounding box) to our unified normative coordinate ((x, y, z) is the 3D geometry center of object bounding box).

Thanks. btw, yaw in unified normative coordinate means counterclockwise rotation angle from x axis to y axis, right?

sshaoshuai, 2020-07-07:
Yes, you are right.

closed 2020-07-07 by sshaoshuai

pk1996, 2022-06-30:
As far as my understanding goes, the yaw angle is typically measured in the CCW direction, starting from the x-axis. So the definition of yaw angle as per the unified normative coordinate is the CW angle measured wrt negative y-axis as the following figure indicates - 

[image]

Kindly correct me if I am wrong! 
Also, it would be great if some documentation could be added if not done already! 

Thanks!