open-mmlab/OpenPCDet #129: confused with  Car AP and Car AP_R40

liyang0522, 2020-07-07:
when i get th AP result, there is a Car AP@0.70, 0.70, 0.70  and a Car AP_R40@0.70, 0.70, 0.70,i am confused with the difference between AP and AP_R40 ,can any one help? Thank you!

MartinHahner88, 2020-07-07:
"**Note 2:** On 08.10.2019, we have followed the suggestions of the Mapillary team in their paper [Disentangling Monocular 3D Object Detection](https://research.mapillary.com/img/publications/MonoDIS.pdf) and use 40 recall positions instead of the 11 recall positions proposed in the original Pascal VOC benchmark. This results in a more fair comparison of the results, please check their paper." [[KITTI website](http://www.cvlibs.net/datasets/kitti/eval_object.php?obj_benchmark=3d)]

so _AP_ is over 11 recall positions 
and _AP_R40_ is over 40 recall positions

best practice nowadays is to report and compare using _AP_R40_ and not _AP_ anymore

liyang0522, 2020-07-07:
I  get it !Thank you!

closed 2020-07-10 by liyang0522