open-mmlab/OpenPCDet #130: When I run "python test.py --xxx", I got CUDA problem.

Gojay001, 2020-07-07:
Hi! Thanks for your excellent work!
When I run "python test.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml --batch_size 4 --ckpt pv_rcnn_8369.pth", I got CUDA problem, details as follows.

my first enviroments are:
- cuda=10.0
- pytorch=1.1 (I have tried to change to 1.2 or ohther versions, but I got same problem)
- spconv=1.0

then I got problem: **RuntimeError: CUDA error: invalid argument**, as #69 
```
INFO  *************** EPOCH 8369 EVALUATION *****************
eval:   0%|                                             | 0/943 [00:00<?, ?it/s]Traceback (most recent call last):
  File "test.py", line 190, in <module>
    main()
  File "test.py", line 186, in main
    eval_single_ckpt(model, test_loader, args, eval_output_dir, logger, epoch_id, dist_test=dist_test)
  File "test.py", line 61, in eval_single_ckpt
    result_dir=eval_output_dir, save_to_file=args.save_to_file
  File "/home/ease/Documents/OpenLidarPerceptron/tools/eval_utils/eval_utils.py", line 57, in eval_one_epoch
    pred_dicts, ret_dict = model(batch_dict)
  File "/home/ease/anaconda3/envs/siammask/lib/python3.7/site-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/ease/Documents/OpenLidarPerceptron/pcdet/models/detectors/pv_rcnn.py", line 11, in forward
    batch_dict = cur_module(batch_dict)
  File "/home/ease/anaconda3/envs/siammask/lib/python3.7/site-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/ease/Documents/OpenLidarPerceptron/pcdet/models/backbones_3d/spconv_backbone.py", line 99, in forward
    x = self.conv_input(input_sp_tensor)
  File "/home/ease/anaconda3/envs/siammask/lib/python3.7/site-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/ease/anaconda3/envs/siammask/lib/python3.7/site-packages/spconv/modules.py", line 127, in forward
    input.features = module(input.features)
  File "/home/ease/anaconda3/envs/siammask/lib/python3.7/site-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/ease/anaconda3/envs/siammask/lib/python3.7/site-packages/torch/nn/modules/activation.py", line 99, in forward
    return F.relu(input, inplace=self.inplace)
  File "/home/ease/anaconda3/envs/siammask/lib/python3.7/site-packages/torch/nn/functional.py", line 943, in relu
    result = torch.relu(input)
RuntimeError: CUDA error: invalid argument
eval:   0%|                                             | 0/943 [00:01<?, ?it/s]
```

Then I read other ideas about the `spconv version`, futher changed my enviroments:
- cuda=10.0
- pytorch=1.4
- spconv=1.1

and I got problem **assert block_filtering is False AssertionError**, as #21 
```
INFO  *************** EPOCH 7940 EVALUATION *****************
eval:   0%|                                             | 0/943 [00:00<?, ?it/s]Traceback (most recent call last):
  File "test.py", line 190, in <module>
    main()
  File "test.py", line 186, in main
    eval_single_ckpt(model, test_loader, args, eval_output_dir, logger, epoch_id, dist_test=dist_test)
  File "test.py", line 61, in eval_single_ckpt
    result_dir=eval_output_dir, save_to_file=args.save_to_file
  File "/home/ease/Documents/OpenLidarPerceptron/tools/eval_utils/eval_utils.py", line 54, in eval_one_epoch
    for i, batch_dict in enumerate(dataloader):
  File "/home/ease/anaconda3/envs/siammask/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 345, in __next__
    data = self._next_data()
  File "/home/ease/anaconda3/envs/siammask/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 856, in _next_data
    return self._process_data(data)
  File "/home/ease/anaconda3/envs/siammask/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 881, in _process_data
    data.reraise()
  File "/home/ease/anaconda3/envs/siammask/lib/python3.7/site-packages/torch/_utils.py", line 394, in reraise
    raise self.exc_type(msg)
AssertionError: Caught AssertionError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/home/ease/anaconda3/envs/siammask/lib/python3.7/site-packages/torch/utils/data/_utils/worker.py", line 178, in _worker_loop
    data = fetcher.fetch(index)
  File "/home/ease/anaconda3/envs/siammask/lib/python3.7/site-packages/torch/utils/data/_utils/fetch.py", line 44, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/ease/anaconda3/envs/siammask/lib/python3.7/site-packages/torch/utils/data/_utils/fetch.py", line 44, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/ease/Documents/OpenLidarPerceptron/pcdet/datasets/kitti/kitti_dataset.py", line 372, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)
  File "/home/ease/Documents/OpenLidarPerceptron/pcdet/datasets/dataset.py", line 131, in prepare_data
    data_dict=data_dict
  File "/home/ease/Documents/OpenLidarPerceptron/pcdet/datasets/processor/data_processor.py", line 78, in forward
    data_dict = cur_processor(data_dict=data_dict)
  File "/home/ease/Documents/OpenLidarPerceptron/pcdet/datasets/processor/data_processor.py", line 56, in transform_points_to_voxels
    voxels, coordinates, num_points = voxel_generator.generate(points)
  File "/home/ease/anaconda3/envs/siammask/lib/python3.7/site-packages/spconv/utils/__init__.py", line 164, in generate
    or self._max_voxels, self._full_mean)
  File "/home/ease/anaconda3/envs/siammask/lib/python3.7/site-packages/spconv/utils/__init__.py", line 65, in points_to_voxel
    assert block_filtering is False
AssertionError
eval:   0%|                                             | 0/943 [00:00<?, ?it/s]
```

Now, I am confused with the configuration, can you give me some suggestions? Thanks!

sshaoshuai, 2020-07-07:
This repo only support spconv 1.0, and currently this repo doesn't support spconv 1.1 since it costs more GPU memory than spconv 1.0. 
The PyTorch version should be 1.1 or 1.2. 
I guess perhaps there is something wrong in your installation of spconv. 
Maybe you could step by step to check the status of tensor in `x = self.conv_input(input_sp_tensor)` in `spconv_backbone.py`

MartinHahner88, 2020-07-08:
maybe [this thread](https://github.com/open-mmlab/OpenPCDet/issues/72#issuecomment-639615724) can help

Gojay001, 2020-07-09:
Thanks for your both help, **finally solved** the problem.
I wonder it remains error cache in my previous enviroment, then I just create a new conda enviroment followed with #72 step by step, finally fixed it.
Thanks again!

closed 2020-07-09 by Gojay001