open-mmlab/OpenPCDet #131: Saving the model as torch script and load to C++.

ltphy, 2020-07-08:
Hi. Is it possible to save the model as torch script and load to C++ code.
I try to save the model by using the following code:
```
if('batch_size' in batch_dict):
   batch_dict.pop('batch_size', none)
trace = torch.jit.trace(model, batch_dict)
trace.save('pvrcn.pt')
```
As well as remove the key 'batch_size' so that all the value of dict is tensor values and change the batch_size inside all the code that preferred to 'batch_size' as 1.
However it produces the error from spconv code:
```
RuntimeError: Found an unsupported argument type c10::List<at::Tensor> in the JIT tracer.
```
This seems to be the problem with input arguments which do not support type 'list'.
Have you ever tried to load the model to C++ code?
Thank you very much.

sshaoshuai, 2020-07-08:
Sorry, I didn't try to load to C++ code. Maybe you could ask it on the repo of `spconv`.

closed 2020-07-08 by ltphy

ltphy, 2020-07-09:
Thank you. But as I viewed from the issues of repo `spconv`. It seems to be not supported also.