open-mmlab/OpenPCDet #134: Algorithm reproduction

alsun-oven, 2020-07-08:
When i get the car's AP result of pv_rcnn, the values is 89.4918, 79.2140,  78.3794,  but  your values is 89.34, 83.69, 78.70。Do your have some tricks.

sshaoshuai, 2020-07-08:
What's the batch size and training epochs? Maybe you could try to re-train the models with 80 epochs.

alsun-oven, 2020-07-08:
batch size 24 , 80 epochs, LR_WARMUP: False

sshaoshuai, 2020-07-08:
Generally I trained with batch size 16 and 80 epochs with 8 gpus.

AutoVision-cloud, 2020-07-08:
I tried with the above configuration and was able to get the AP on the validation set as 89.5295, 83.3945, 78.8758.

INFO  Car AP@0.70, 0.70, 0.70:
bbox AP:95.6460, 89.5244, 89.1397
bev  AP:90.2876, 88.1063, 87.4073
3d   AP:89.5295, 83.3945, 78.8758
aos  AP:95.60, 89.42, 88.97

Pedestrian AP@0.50, 0.50, 0.50:
bbox AP:73.7535, 70.1916, 66.8279
bev  AP:68.5050, 62.8781, 58.7984
3d   AP:65.9928, 59.4248, 55.4530
aos  AP:70.86, 66.59, 62.77

Cyclist AP@0.50, 0.50, 0.50:
bbox AP:95.7840, 82.3173, 77.4679
bev  AP:87.9788, 74.6423, 72.3375
3d   AP:86.7078, 73.0689, 69.3516
aos  AP:95.62, 81.97, 77.05
Cyclist AP_R40@0.50, 0.50, 0.50:
bbox AP:96.9239, 83.5845, 80.4162
bev  AP:93.0078, 77.0466, 72.5361
3d   AP:91.5720, 74.0862, 69.5900
aos  AP:96.79, 83.25, 79.90
Cyclist AP@0.50, 0.25, 0.25:
bbox AP:95.7840, 82.3173, 77.4679
bev  AP:94.1489, 78.3904, 73.6979                                     

Also, do you know when you can release the code for the Waymo OD?

sshaoshuai, 2020-07-08:
Congratulations! This AP is already normal for PV-RCNN. 
I could not give a specific time for the waymo-related codes. But I will support some other datasets (like NuScenes) soon.

alsun-oven, 2020-07-09:
The loss curve has large fluctuations. Is this normal?

sshaoshuai, 2020-07-09:
I didn't notice much about the loss fluctuations. You could find the mAP curve instead to see if it is normal.

alsun-oven, 2020-07-09:
Will it be better to use a syncBatchNorm?

alsun-oven, 2020-07-09:
I tried with the above configuration and was not able to get the AP on the validation set as 89.5295, 83.3945, 78.8758. I did not use ROAD_PLANE.

sshaoshuai, 2020-07-10:
Maybe you could download the road_planes and try to train with the road planes. I don't know how much the road planes could help in the KITTI datasets. 
Also, the official sync-bn is useless, please do not use it.

alsun-oven, 2020-07-10:
I tried with the configuration of pvrcnn, and noticed that using road plane has some points of improvement on the AP_r11.Do you recommend a good ground estimation method.

sshaoshuai, 2020-07-10:
The KITTI road planes were borrowed from other repos long long ago, and I didn't have a ground estimation method. 
For other datasets (like waymo / NuScenes), I could also achieve good results without road planes. So I think the road planes may not be the key issue for performance. Maybe you could try to train without road planes for several times to get a stable conclusions.

closed 2020-07-17 by sshaoshuai

LittleYuanzi, 2020-08-03:
> The loss curve has large fluctuations. Is this normal?

hi~, do you find where is wrong in the paramenter setting, I also can not get a good AP with PVRCNN

MartinHahner88, 2020-08-04:
my PVRCNN training looks like this:
[image]
For me, out of all models provided in this repo, it achieves the highest IoU (84.42%) on moderate cars in KITTI.

davidwang200099, 2020-08-24:
Excuse me, should I download some other files before enabling USE_ROAD_PLANE and training with kitti dataset?