open-mmlab/OpenPCDet #135: ImportError: cannot import name 'roiaware_pool3d_cuda' from 'pcdet.ops.roiaware_pool3d' (unknown location)

hughlee815, 2020-07-08:
Traceback (most recent call last):
  File "train.py", line 9, in <module>
    from pcdet.datasets import build_dataloader
  File "/home/nio/pointdet/OpenPCDet/pcdet/datasets/__init__.py", line 3, in <module>
    from .dataset import DatasetTemplate
  File "/home/nio/pointdet/OpenPCDet/pcdet/datasets/dataset.py", line 5, in <module>
    from .augmentor.data_augmentor import DataAugmentor
  File "/home/nio/pointdet/OpenPCDet/pcdet/datasets/augmentor/data_augmentor.py", line 3, in <module>
    from . import augmentor_utils, database_sampler
  File "/home/nio/pointdet/OpenPCDet/pcdet/datasets/augmentor/database_sampler.py", line 3, in <module>
    from ...utils import box_utils
  File "/home/nio/pointdet/OpenPCDet/pcdet/utils/box_utils.py", line 4, in <module>
    from ..ops.roiaware_pool3d import roiaware_pool3d_utils
  File "/home/nio/pointdet/OpenPCDet/pcdet/ops/roiaware_pool3d/roiaware_pool3d_utils.py", line 5, in <module>
    from . import roiaware_pool3d_cuda
ImportError: cannot import name 'roiaware_pool3d_cuda' from 'pcdet.ops.roiaware_pool3d' (unknown location)

does anyone also encounter this problem?
how can i fix this bug? 
please help, i have been confused by this for a long long time

hughlee815, 2020-07-08:
oh, this error happens when i run "train.py".

AutoVision-cloud, 2020-07-08:
During setup, did `python setup.py develop` throw any errors for roiaware_pool3d? 
It seems like this wasn't installed correctly.

hughlee815, 2020-07-09:
yes, it says `gpu may not support`. then i change to another machine, it is solved!
thanks so much.
but a new problem comes:
when i run "python tools/train.py --cfg_file /home/adas/congcong.li/OpenPCDet/tools/cfgs/kitti_models/pointpillar.yaml --batch_size 32 --epochs 50"
it throws a error: FileNotFoundError: [Errno 2] No such file or directory: 'cfgs/dataset_configs/kitti_dataset.yaml'
which is a little strange...

hughlee815, 2020-07-09:
yes, it says gpu may not support. then i change to another machine, it is solved!
thanks so much.

closed 2020-07-09 by hughlee815