open-mmlab/OpenPCDet #138: KeyError: 'road_plane'

hughlee815, 2020-07-09:
when i run "train.py", it throws `KeyError: 'road_plane'` like below. i know there is no road in my /data directory, but i think road is not necessary for pcdet. i try to figure out where i can close the requirement of road, but i fail.
does anyone have any idea?
thank you in advance.

Traceback (most recent call last):                                                                                                       
  File "train.py", line 185, in <module>
    main()
  File "train.py", line 157, in main
    merge_all_iters_to_one_epoch=args.merge_all_iters_to_one_epoch
  File "/home/adas/congcong.li/OpenPCDet/tools/train_utils/train_utils.py", line 92, in train_model
    dataloader_iter=dataloader_iter
  File "/home/adas/congcong.li/OpenPCDet/tools/train_utils/train_utils.py", line 18, in train_one_epoch
    batch = next(dataloader_iter)
  File "/home/adas/anaconda3/envs/pcdet/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 345, in __next__
    data = self._next_data()
  File "/home/adas/anaconda3/envs/pcdet/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 856, in _next_data
    return self._process_data(data)
  File "/home/adas/anaconda3/envs/pcdet/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 881, in _process_data
    data.reraise()
  File "/home/adas/anaconda3/envs/pcdet/lib/python3.7/site-packages/torch/_utils.py", line 394, in reraise
    raise self.exc_type(msg)
KeyError: Caught KeyError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/home/adas/anaconda3/envs/pcdet/lib/python3.7/site-packages/torch/utils/data/_utils/worker.py", line 178, in _worker_loop
    data = fetcher.fetch(index)
  File "/home/adas/anaconda3/envs/pcdet/lib/python3.7/site-packages/torch/utils/data/_utils/fetch.py", line 44, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/adas/anaconda3/envs/pcdet/lib/python3.7/site-packages/torch/utils/data/_utils/fetch.py", line 44, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/adas/congcong.li/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 378, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)
  File "/home/adas/congcong.li/OpenPCDet/pcdet/datasets/dataset.py", line 122, in prepare_data
    'gt_boxes_mask': gt_boxes_mask
  File "/home/adas/congcong.li/OpenPCDet/pcdet/datasets/augmentor/data_augmentor.py", line 91, in forward
    data_dict = cur_augmentor(data_dict=data_dict)
  File "/home/adas/congcong.li/OpenPCDet/pcdet/datasets/augmentor/database_sampler.py", line 195, in __call__
    data_dict = self.add_sampled_boxes_to_scene(data_dict, sampled_gt_boxes, total_valid_sampled_dict)
  File "/home/adas/congcong.li/OpenPCDet/pcdet/datasets/augmentor/database_sampler.py", line 123, in add_sampled_boxes_to_scene
    sampled_gt_boxes, data_dict['road_plane'], data_dict['calib']
KeyError: 'road_plane'

closed 2020-07-09 by hughlee815

Physu, 2020-07-16:
So, could you please tell me how fix this problem?
Thanks a lot!

Physu, 2020-07-16:
find tools/cfgs/*.yaml, then change "USE_ROAD_PLANE: True" to "USE_ROAD_PLANE: False"