open-mmlab/OpenPCDet #143: RuntimeError occurs when running PartA2 on torch 1.2

gardenbaby, 2020-07-09:
Hello,

I found that the following error will occur when running PartA2 on torch 1.2.

> RuntimeError: range.second - range.first == t.size() INTERNAL ASSERT FAILED at /pytorch/torch/csrc/autograd/generated/Functions.cpp:55, please report a bug to PyTorch. inconsistent range for TensorList output

Since the requirements in [INSTALL.md](https://github.com/open-mmlab/OpenPCDet/blob/master/docs/INSTALL.md) says that the code was tested on `PyTorch 1.1`, I changed the PyTorch version and it worked.

The issue [here](https://discuss.pytorch.org/t/torch-nonzero-leads-to-crash-in-pytorch-1-2/55729) is quite similar.  Is function `torch.nonzero()` the reason?

Thank you.

sshaoshuai, 2020-07-12:
Sorry I didn't check it with pytorch 1.2. 
`torch.nonzero()` may be used here https://github.com/open-mmlab/OpenPCDet/blob/master/pcdet/models/roi_heads/target_assigner/proposal_target_layer.py#L121 in PartA2. 
Maybe you could check here.

gujiaqivadin, 2020-07-14:
@gardenbaby @sshaoshuai 
Hello, I have the same problem as you. And I choose to change my torch version from 1.2.0 to 1.5.1, but after I change my version, new error occurs:
```
Traceback (most recent call last):
  File "train.py", line 7, in <module>
    from pcdet.datasets import build_dataloader
  File "/home/vadingu/OpenPCDet/pcdet/datasets/__init__.py", line 3, in <module>
    from .dataset import DatasetTemplate
  File "/home/vadingu/OpenPCDet/pcdet/datasets/dataset.py", line 5, in <module>
    from .augmentor.data_augmentor import DataAugmentor
  File "/home/vadingu/OpenPCDet/pcdet/datasets/augmentor/data_augmentor.py", line 3, in <module>
    from . import augmentor_utils, database_sampler
  File "/home/vadingu/OpenPCDet/pcdet/datasets/augmentor/database_sampler.py", line 3, in <module>
    from ...utils import box_utils
  File "/home/vadingu/OpenPCDet/pcdet/utils/box_utils.py", line 4, in <module>
    from ..ops.roiaware_pool3d import roiaware_pool3d_utils
  File "/home/vadingu/OpenPCDet/pcdet/ops/roiaware_pool3d/roiaware_pool3d_utils.py", line 5, in <module>
    from . import roiaware_pool3d_cuda
ImportError: /home/vadingu/OpenPCDet/pcdet/ops/roiaware_pool3d/roiaware_pool3d_cuda.cpython-36m-x86_64-linux-gnu.so: undefined symbol: _ZN6caffe26detail37_typeMetaDataInstance_preallocated_32E
```
It seems problem of "python setup.py develop" when torch version changes, and I need to rerun setup. But after I run it, nothing changes as all files has been installed. It seems that I need to delete old "setup" files, but I don't know delete what files.

```
(pcdet) vadingu@ubuntu:~/OpenPCDet$ python setup.py develop
No CUDA runtime is found, using CUDA_HOME='/usr/local/cuda-10.1/'
running develop
running egg_info
writing pcdet.egg-info/PKG-INFO
writing dependency_links to pcdet.egg-info/dependency_links.txt
writing requirements to pcdet.egg-info/requires.txt
writing top-level names to pcdet.egg-info/top_level.txt
/home/vadingu/anaconda3/envs/pcdet/lib/python3.6/site-packages/torch/utils/cpp_extension.py:305: UserWarning: Attempted to use ninja as the BuildExtension backend but we could not find ninja.. Falling back to using the slow distutils backend.
  warnings.warn(msg.format('we could not find ninja.'))
reading manifest file 'pcdet.egg-info/SOURCES.txt'
writing manifest file 'pcdet.egg-info/SOURCES.txt'
running build_ext
copying build/lib.linux-x86_64-3.6/pcdet/ops/iou3d_nms/iou3d_nms_cuda.cpython-36m-x86_64-linux-gnu.so -> pcdet/ops/iou3d_nms
copying build/lib.linux-x86_64-3.6/pcdet/ops/roiaware_pool3d/roiaware_pool3d_cuda.cpython-36m-x86_64-linux-gnu.so -> pcdet/ops/roiaware_pool3d
copying build/lib.linux-x86_64-3.6/pcdet/ops/pointnet2/pointnet2_stack/pointnet2_stack_cuda.cpython-36m-x86_64-linux-gnu.so -> pcdet/ops/pointnet2/pointnet2_stack
Creating /home/vadingu/anaconda3/envs/pcdet/lib/python3.6/site-packages/pcdet.egg-link (link to .)
pcdet 0.2.0+b7553c8 is already the active version in easy-install.pth

Installed /home/vadingu/OpenPCDet
Processing dependencies for pcdet==0.2.0+b7553c8
Searching for PyYAML==5.3.1
Best match: PyYAML 5.3.1
Adding PyYAML 5.3.1 to easy-install.pth file

Using /home/vadingu/anaconda3/envs/pcdet/lib/python3.6/site-packages
Searching for easydict==1.9
Best match: easydict 1.9
Adding easydict 1.9 to easy-install.pth file

Using /home/vadingu/anaconda3/envs/pcdet/lib/python3.6/site-packages
Searching for tensorboardX==2.0
Best match: tensorboardX 2.0
Adding tensorboardX 2.0 to easy-install.pth file

Using /home/vadingu/anaconda3/envs/pcdet/lib/python3.6/site-packages
Searching for numba==0.48.0
Best match: numba 0.48.0
Adding numba 0.48.0 to easy-install.pth file

Using /home/vadingu/anaconda3/envs/pcdet/lib/python3.6/site-packages
Searching for spconv==1.0
Best match: spconv 1.0
Adding spconv 1.0 to easy-install.pth file

Using /home/vadingu/anaconda3/envs/pcdet/lib/python3.6/site-packages
Searching for torch==1.5.1
Best match: torch 1.5.1
Adding torch 1.5.1 to easy-install.pth file
Installing convert-caffe2-to-onnx script to /home/vadingu/anaconda3/envs/pcdet/bin
Installing convert-onnx-to-caffe2 script to /home/vadingu/anaconda3/envs/pcdet/bin

Using /home/vadingu/anaconda3/envs/pcdet/lib/python3.6/site-packages
Searching for numpy==1.18.2
Best match: numpy 1.18.2
Adding numpy 1.18.2 to easy-install.pth file
Installing f2py script to /home/vadingu/anaconda3/envs/pcdet/bin
Installing f2py3 script to /home/vadingu/anaconda3/envs/pcdet/bin
Installing f2py3.6 script to /home/vadingu/anaconda3/envs/pcdet/bin

Using /home/vadingu/anaconda3/envs/pcdet/lib/python3.6/site-packages
Searching for six==1.14.0
Best match: six 1.14.0
Adding six 1.14.0 to easy-install.pth file

Using /home/vadingu/anaconda3/envs/pcdet/lib/python3.6/site-packages
Searching for protobuf==3.11.3
Best match: protobuf 3.11.3
Adding protobuf 3.11.3 to easy-install.pth file

Using /home/vadingu/anaconda3/envs/pcdet/lib/python3.6/site-packages
Searching for setuptools==46.1.1.post20200322
Best match: setuptools 46.1.1.post20200322
Adding setuptools 46.1.1.post20200322 to easy-install.pth file
Installing easy_install script to /home/vadingu/anaconda3/envs/pcdet/bin

Using /home/vadingu/anaconda3/envs/pcdet/lib/python3.6/site-packages
Searching for llvmlite==0.31.0
Best match: llvmlite 0.31.0
Adding llvmlite 0.31.0 to easy-install.pth file

Using /home/vadingu/anaconda3/envs/pcdet/lib/python3.6/site-packages
Searching for future==0.18.2
Best match: future 0.18.2
Adding future 0.18.2 to easy-install.pth file
Installing futurize script to /home/vadingu/anaconda3/envs/pcdet/bin
Installing pasteurize script to /home/vadingu/anaconda3/envs/pcdet/bin

Using /home/vadingu/anaconda3/envs/pcdet/lib/python3.6/site-packages
Finished processing dependencies for pcdet==0.2.0+b7553c8
```

sshaoshuai, 2020-07-14:
Currently you need to run `PCDet` with PyTorch 1.1 or 1.2. It doesn't support 1.5 yet since some CPP/CUDA interfaces need to be modified. 
I have tested it with PyTorch 1.1, maybe you could remove the build files and re-build it with PyTorch 1.1.

gujiaqivadin, 2020-07-14:
Thanks, and where to find build files.

sshaoshuai, 2020-07-14:
Under the root directory of OpenPCDet
1. `rm -r build`
2. `python setup.py clean`

ltphy, 2020-07-21:
Hi @sshaoshuai. 
I also face the same issue when trying to train PartA2 in v0.2.0, while version 0.1.1 still works for me.
I am currently using Pytorch v1.2 and PVRCNN stilll works well but not PartA2.
The `fg_inds` have an empty tensor array. The result iou in `max_overlaps` is very small which make `max_overlaps >= fg_thresh` cannot find any.

ltphy, 2020-07-22:
So, should I downgrade PyTorch version from 1.2 to 1.1 to train PartA2? It is weird because old version of PCDet still can train PartA2 model with Pytorch 1.2.

sshaoshuai, 2020-07-22:
Sorry, I do not understand what your problem is. Could you achieve a similar performance with PCDet v0.2? 
I think the current version should run successfully with PyTorch 1.2.

ltphy, 2020-07-22:
Yeah. It still works when I test PartA2 model but not when training, while PVRCNN works in both situations.

gardenbaby, 2020-07-22:
> Yeah. It still works when I test PartA2 model but not when training, while PVRCNN works in both situations.
> Here is the problem.
> 
> ```
> RuntimeError: range.second - range.first == t.size() INTERNAL ASSERT FAILED at /opt/conda/conda-
> bld/pytorch_1565272271120/work/torch/csrc/autograd/generated/Functions.cpp:55, 
> please report a bug to PyTorch. inconsistent range for TensorList output
> ```

Same here. The error only occurs when training PartA2 in PCDet v0.2. I changed the PyTorch version from 1.2 to 1.1 and it worked.

closed 2022-03-08 by github-actions[bot]