open-mmlab/OpenPCDet #145: how to make a simple inference

Tartisan, 2020-07-11:
how to make a simple inference like [second.pytorch](https://github.com/traveller59/second.pytorch/blob/master/second/simple-inference.ipynb) does?

sshaoshuai, 2020-07-12:
What do you mean for a simple inference? 
We have provided several pretrained models as shown in the README.md, you could use the following command to inference with a pretrained model: 
```
python test.py --cfg_file ${CONFIG_FILE} --batch_size ${BATCH_SIZE} --ckpt ${CKPT}
```
It will generate the evaluation results by default.

jacoblambert, 2020-07-17:
What he means is it would be nice to have simple inference.py function which:
- loads model and config file, --cfg_file --ckpt
- loads pointcloud file --pointcloud, csv file
- runs pointcloud through model
- returns bounding boxes

This "minimal inference function" would give users a starting point to easily build their custom modules around your code. It would probably be easy for you to put this together, so it would be greatly appreciated. Taking apart test.py to run on custom data is not easy for someone not familiar with your code. Thank you!

Tartisan, 2020-07-17:
> What he means is it would be nice to have simple inference.py function which:
>
> * loads model and config file, --cfg_file --ckpt
> * loads pointcloud file --pointcloud, csv file
> * runs pointcloud [...]

Well, this is exactly what i mean :)

sshaoshuai, 2020-07-17:
Hi all, I have just added the quick demo to test a pretrained model on your custom data. Please refer [DEMO.md](https://github.com/open-mmlab/OpenPCDet/blob/master/docs/DEMO.md) for details

closed 2020-07-17 by sshaoshuai