open-mmlab/OpenPCDet #146: Anchor bottom height for pedestrians & cyclists

Divadi, 2020-07-13:
I noticed here:
https://github.com/open-mmlab/OpenPCDet/blob/b7553c879d0ba36477931efe07a55adbc39823b9/tools/cfgs/kitti_models/pv_rcnn.yaml#L78
that the anchor_bottom_height for pedestrians and cyclists is actually set to -0.6. From other works, I saw that this value is actually the center of the pedestrian & cyclist object, not the bottom. Should it instead be shifted down by h/2 of these two objects? 

Also, I was wondering if you could shed some light on how exactly the bottom height for car was calculated. I know that papers generally use -1.0 for the center, and lowering that by half the height of the car yields -1.78. However, when I just took the average of the z bottom height for cars, I ended up with -1.51. Why would there be a difference?

Thank you for your great work!

sshaoshuai, 2020-07-13:
Hi, actually I have tried several `anchor_bottom_heights` with small variance, and the results are similar. 
I guess -1.51 and -1.78 could achieve similar results. 
For the KITTI dataset, I remembered that -1.78 is copied from original second codebase. 
For a new dataset, you could statistics the box height from training set.

Divadi, 2020-07-14:
I see - I guess the regressor will learn whatever bias there is
I do think the bottom heights for ped & cyc should be lower, but I'll see if it makes a difference

closed 2020-07-21 by sshaoshuai