open-mmlab/OpenPCDet #147: prediction on the custom data

Shawn-Zhuang, 2020-07-13:
I have successfully run the test.py on the KITTI dataset.
I am a student new to the 3D detection research. I want to ask for help, with this wonderful toolkit OpenPCDet, how can I predict using a pre-trained model on my own data provided by the teacher. Can anybody give me some help or blog which I can refer to.
I want to check the result on the data my radar collected.
Thanks for the kind help!

here is the bag information on my own data:
```
path:        2018-01-06-15-14-01.bag
version:     2.0
duration:    14:51s (891s)
start:       Jan 06 2018 15:14:01.43 (1515222841.43)
end:         Jan 06 2018 15:28:52.53 (1515223732.53)
size:        8.5 GB
messages:    45378
compression: none [8842/8842 chunks]
types:       nav_msgs/Odometry       [cd5e73d190d741a2f92e81eda573aca7]
             sensor_msgs/Imu         [6a62c6daae103f4ff57a132d6f95cec2]
             sensor_msgs/NavSatFix   [2d3a8cd499b9b4a0249fb98fd05cfa48]
             sensor_msgs/PointCloud2 [1158d486dd51d683ce2f1be655c3c181]
topics:      /fix              892 msgs    : sensor_msgs/NavSatFix  
             /imu_raw        17823 msgs    : sensor_msgs/Imu        
             /odom_encoder   17822 msgs    : nav_msgs/Odometry      
             /points_raw      8841 msgs    : sensor_msgs/PointCloud2

```

sshaoshuai, 2020-07-14:
1. Load your custom data, and get the raw point cloud (N, 3) with x, y, z. (The provided pretrained-model used (N, 4) with x, y, z, intensity, if you doesn't have the intensity, then you need to train a new model without intensity. )
2. You need to feed the points (N, 3 or 4) to the model, just refer to the `_getitem_` function of `kitti_dataset.py` to repace the kitti points with your own points. 

Also remember that the coordinate of your points (N, 3) should in the unified normative coordinate of PCDet as shown in the README.md. 
And the radar points are much sparser than the KITTI LiDAR points, so I guess you may get a bad prediction results if you only use the radar points.

Shawn-Zhuang, 2020-07-15:
@sshaoshuai Thanks for your kind help and it is my mistake, it is lidar data, not radar.

From other issues I did the following things.
1. Replace the files in ./data/kitti/testing/velodyne with my PC.bin files
2.  Copy one file in the /testing/calib as much as the number of my PC.bin files and rename them after the index, because I don't 
     have that kind of file and just want to check the result on lidar rather than images.
3. Do the second step to the data in the image_2 directory.  
4.  run 
`python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos tools/cfgs/dataset_configs/kitti_dataset.yaml`
to generate info.pkl, which is successful
5. according to other contents in issues, I change the DATA_SPLIT['test']=test and INFO_PATH['test']=kitti_infos_test.pkl in kitti_dataset.yaml, and then run
`python test.py --cfg_file ./cfgs/kitti_models/pv_rcnn.yaml --batch_size 2 --ckpt ./cfgs/kitti_models/pv_rcnn_8369.pth`

the following errors appear. Is there anything wrong with my steps?
I used to test successfully on the official kitti test data.
Could you please kindly have a check on my steps and the errors?

`
(base) shawn@shawn-HP:~/OpenPCDet/tools$ python test.py --cfg_file ./cfgs/kitti_models/pv_rcnn.yaml --batch_size 2 --ckpt ./cfgs/kitti_models/pv_rcnn_8369.pth
2020-07-15 17:34:52,322   INFO  **********************Start logging**********************
2020-07-15 17:34:52,323   INFO  CUDA_VISIBLE_DEVICES=ALL
2020-07-15 17:34:52,323   INFO  cfg_file         ./cfgs/kitti_models/pv_rcnn.yaml
2020-07-15 17:34:52,323   INFO  batch_size       2
2020-07-15 17:34:52,323   INFO  epochs           80
2020-07-15 17:34:52,323   INFO  workers          4
2020-07-15 17:34:52,323   INFO  extra_tag        default
2020-07-15 17:34:52,323   INFO  ckpt             ./cfgs/kitti_models/pv_rcnn_8369.pth
2020-07-15 17:34:52,323   INFO  mgpus            False
2020-07-15 17:34:52,323   INFO  launcher         none
2020-07-15 17:34:52,323   INFO  tcp_port         18888
2020-07-15 17:34:52,323   INFO  local_rank       0
2020-07-15 17:34:52,323   INFO  set_cfgs         None
2020-07-15 17:34:52,323   INFO  max_waiting_mins 30
2020-07-15 17:34:52,323   INFO  start_epoch      0
2020-07-15 17:34:52,323   INFO  eval_tag         default
2020-07-15 17:34:52,323   INFO  eval_all         False
2020-07-15 17:34:52,323   INFO  ckpt_dir         None
2020-07-15 17:34:52,323   INFO  save_to_file     False
2020-07-15 17:34:52,323   INFO  cfg.ROOT_DIR: /home/shawn/OpenPCDet
2020-07-15 17:34:52,323   INFO  cfg.LOCAL_RANK: 0
2020-07-15 17:34:52,324   INFO  cfg.CLASS_NAMES: ['Car', 'Pedestrian', 'Cyclist']
2020-07-15 17:34:52,324   INFO  
cfg.DATA_CONFIG = edict()
2020-07-15 17:34:52,324   INFO  cfg.DATA_CONFIG.DATASET: KittiDataset
2020-07-15 17:34:52,324   INFO  cfg.DATA_CONFIG.DATA_PATH: ../data/kitti
2020-07-15 17:34:52,324   INFO  cfg.DATA_CONFIG.POINT_CLOUD_RANGE: [0, -40, -3, 70.4, 40, 1]
2020-07-15 17:34:52,324   INFO  
cfg.DATA_CONFIG.DATA_SPLIT = edict()
2020-07-15 17:34:52,324   INFO  cfg.DATA_CONFIG.DATA_SPLIT.train: train
2020-07-15 17:34:52,324   INFO  cfg.DATA_CONFIG.DATA_SPLIT.test: test
2020-07-15 17:34:52,324   INFO  
cfg.DATA_CONFIG.INFO_PATH = edict()
2020-07-15 17:34:52,324   INFO  cfg.DATA_CONFIG.INFO_PATH.train: ['kitti_infos_train.pkl']
2020-07-15 17:34:52,324   INFO  cfg.DATA_CONFIG.INFO_PATH.test: ['kitti_infos_test.pkl']
2020-07-15 17:34:52,324   INFO  cfg.DATA_CONFIG.FOV_POINTS_ONLY: True
2020-07-15 17:34:52,324   INFO  
cfg.DATA_CONFIG.DATA_AUGMENTOR = edict()
2020-07-15 17:34:52,324   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.DISABLE_AUG_LIST: ['placeholder']
2020-07-15 17:34:52,324   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.AUG_CONFIG_LIST: [{'NAME': 'gt_sampling', 'USE_ROAD_PLANE': True, 'DB_INFO_PATH': ['kitti_dbinfos_train.pkl'], 'PREPARE': {'filter_by_min_points': ['Car:5', 'Pedestrian:5', 'Cyclist:5'], 'filter_by_difficulty': [-1]}, 'SAMPLE_GROUPS': ['Car:15', 'Pedestrian:10', 'Cyclist:10'], 'NUM_POINT_FEATURES': 4, 'DATABASE_WITH_FAKELIDAR': False, 'REMOVE_EXTRA_WIDTH': [0.0, 0.0, 0.0], 'LIMIT_WHOLE_SCENE': False}, {'NAME': 'random_world_flip', 'ALONG_AXIS_LIST': ['x']}, {'NAME': 'random_world_rotation', 'WORLD_ROT_ANGLE': [-0.78539816, 0.78539816]}, {'NAME': 'random_world_scaling', 'WORLD_SCALE_RANGE': [0.95, 1.05]}]
2020-07-15 17:34:52,324   INFO  
cfg.DATA_CONFIG.POINT_FEATURE_ENCODING = edict()
2020-07-15 17:34:52,324   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.encoding_type: absolute_coordinates_encoding
2020-07-15 17:34:52,324   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.used_feature_list: ['x', 'y', 'z', 'intensity']
2020-07-15 17:34:52,324   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.src_feature_list: ['x', 'y', 'z', 'intensity']
2020-07-15 17:34:52,324   INFO  cfg.DATA_CONFIG.DATA_PROCESSOR: [{'NAME': 'mask_points_and_boxes_outside_range', 'REMOVE_OUTSIDE_BOXES': True}, {'NAME': 'shuffle_points', 'SHUFFLE_ENABLED': {'train': True, 'test': False}}, {'NAME': 'transform_points_to_voxels', 'VOXEL_SIZE': [0.05, 0.05, 0.1], 'MAX_POINTS_PER_VOXEL': 5, 'MAX_NUMBER_OF_VOXELS': {'train': 16000, 'test': 40000}}]
2020-07-15 17:34:52,324   INFO  cfg.DATA_CONFIG._BASE_CONFIG_: cfgs/dataset_configs/kitti_dataset.yaml
2020-07-15 17:34:52,324   INFO  
cfg.MODEL = edict()
2020-07-15 17:34:52,324   INFO  cfg.MODEL.NAME: PVRCNN
2020-07-15 17:34:52,324   INFO  
[... 190 lines cut ...]
  pts_valid_flag = np.logical_and(val_flag_merge, pts_rect_depth >= 0)
[... cut ...]
    result = self.forward(*input, **kwargs)
  File "/home/shawn/OpenPCDet/pcdet/models/backbones_3d/pfe/voxel_set_abstraction.py", line 176, in forward
    keypoints = self.get_sampled_points(batch_dict)
  File "/home/shawn/OpenPCDet/pcdet/models/backbones_3d/pfe/voxel_set_abstraction.py", line 146, in get_sampled_points
    keypoints = sampled_points[0][cur_pt_idxs[0]].unsqueeze(dim=0)
IndexError: index is out of bounds for dimension with size 0
eval:  17%|██          | 154/886 [00:54<04:18,  2.83it/s, recall_0.3=(0, 0) / 0]
`

Shawn-Zhuang, 2020-07-16:
The position of some pc equals 0, which causes such errors.
I would close the issue.

closed 2020-07-16 by Shawn-Zhuang

yangnvzi, 2020-09-02:
@sshaoshuai @Shawn-Zhuang 
@Shawn-Zhuang i refer to your guide but i get this error:
Traceback (most recent call last):
  File "test.py", line 199, in <module>
    main()
  File "test.py", line 195, in main
    eval_single_ckpt(model, test_loader, args, eval_output_dir, logger, epoch_id, dist_test=dist_test)
  File "test.py", line 63, in eval_single_ckpt
    result_dir=eval_output_dir, save_to_file=args.save_to_file
  File "/home/sata2/wangying/OpenPCDet-master/tools/eval_utils/eval_utils.py", line 54, in eval_one_epoch
    for i, batch_dict in enumerate(dataloader):
  File "/opt/conda/envs/openpcdet/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 345, in __next__
    data = self._next_data()
  File "/opt/conda/envs/openpcdet/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 856, in _next_data
    return self._process_data(data)
  File "/opt/conda/envs/openpcdet/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 881, in _process_data
    data.reraise()
  File "/opt/conda/envs/openpcdet/lib/python3.6/site-packages/torch/_utils.py", line 394, in reraise
    raise self.exc_type(msg)
AssertionError: Caught AssertionError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/opt/conda/envs/openpcdet/lib/python3.6/site-packages/torch/utils/data/_utils/worker.py", line 178, in _worker_loop
    data = fetcher.fetch(index)
  File "/opt/conda/envs/openpcdet/lib/python3.6/site-packages/torch/utils/data/_utils/fetch.py", line 44, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/opt/conda/envs/openpcdet/lib/python3.6/site-packages/torch/utils/data/_utils/fetch.py", line 44, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/sata2/wangying/OpenPCDet-master/pcdet/datasets/kitti/kitti_dataset.py", line 344, in __getitem__
    points = self.get_lidar(sample_idx)
  File "/home/sata2/wangying/OpenPCDet-master/pcdet/datasets/kitti/kitti_dataset.py", line 58, in get_lidar
    assert lidar_file.exists()
AssertionError

eval:   0%|                                                                                                           
i am sure kitti_infos.test.pkl is TRUE and .bin file is N*4 narray

zzqjh, 2020-09-22:
hello,i also want to use my own dataset ,but i wonder how i can get the .bin file about the lidar data.could you please give me some suggestions about getting the .bin file from the rosbag file？

Shawn-Zhuang, 2020-09-24:
@zzqjh 
please refer to this project 
https://github.com/leofansq/Tools_RosBag2KITTI

zzqjh, 2020-09-24:
thank you for giving me the great advice！

alixiansen91, 2020-11-02:
Hello, on the basis of using only 3D point cloud data, how to use the reference coordinate system under KITTI? There is another question, how do you know what your coordinate system is like?

zjx99, 2020-11-22:
> Hello, on the basis of using only 3D point cloud data, how to use the reference coordinate system under KITTI? There is another question, how do you know what your coordinate system is like?

Have the same question.

clytze0216, 2021-07-26:
@ zjx99 
hi, I have the same question. Did you solve it?

zjx99, 2021-07-26:
Hi, clytze0216, 

Sorry, not yet, still in the middle of it.

Ysnnnn, 2024-03-20:
> The position of some pc equals 0, which causes such errors. I would close the issue.

why the position of some pc equals 0 will cause error? Would you please tell me how do you fix this problem, thanks!

zhixiongzh, 2024-03-20:
> > The position of some pc equals 0, which causes such errors. I would close the issue.
> 
> why the position of some pc equals 0 will cause error? Would you please tell me how do you fix this problem, thanks!

@Ysnnnn I cannot recall what was happening, it is long time ago, but `index is out of bounds for dimension with size 0` usually means the `index` is larger than the amount of data. So you can set a debug point there and check your data. I guess in my case there is no data( i guess).

Ysnnnn, 2024-03-20:
> @Ysnnnn I cannot recall what was happening, it is long time ago, but `index is out of bounds for dimension with size 0` usually means the `index` is larger than the amount of data. So you can set a [...]

I will try, thank you

adri1cc, 2024-04-11:
> @sshaoshuai Thanks for your kind help and it is my mistake, it is lidar data, not radar.
>
> From other issues I did the following things.
>
> 1. Replace the files in ./data/kitti/testing/velodyne with my [...]

@zhixiongzh Hi, I'm new to 3D object detection and I'm trying to learn how to use this library, and how to use my own dataset. Was this the right method to start ?

Petros626, 2024-10-01:
@zhixiongzh following the custom dataset readme, read all issues about this process.