open-mmlab/OpenPCDet #151: can DATA_AUGMENTOR be closed?

hughlee815, 2020-07-15:
if i want quickly to train my model with a new dataset, i may not be glad to spend time for data_augmentor at first try. so is there any simple interface to close it? i didnt find it.
thanks for your reply so much.

sshaoshuai, 2020-07-15:
Of course, we do provide this feature. 
To disable a single or multiple augmentations, you could put the name of augmentation into the list of `DATA_AUGMENTOR.DISABLE_AUG_LIST` as follows:
```
DATA_AUGMENTOR:
    DISABLE_AUG_LIST: ['gt_sampling', 'random_world_flip', 'random_world_rotation', 'random_world_scaling']
```

closed 2020-07-17 by sshaoshuai

Petros626, 2024-10-08:
@sshaoshuai I tried this option with the original `kitti_dataset.yaml`, but gt samples is produced everytime i run

```python 
python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos tools/cfgs/dataset_configs/kitti_dataset.yaml
```