open-mmlab/OpenPCDet #153: stucked using single-machine multi-gpus training

Number1JT, 2020-07-15:
command：

> CUDA_VISIBLE_DEVICES=4,5,6,7 OMP_NUM_THREADS=1 python -m torch.distributed.launch --nproc_per_node=4 train.py --launcher pytorch --cfg_file  cfgs/kitti_models/pv_rcnn.yaml --batch_size 8 --epochs 50 --workers 8

closed 2020-07-15 by Number1JT

LCJHust, 2020-08-11:
> training using command in tutorial，the utility rate stay at 0 for a very long time，It looks like the training is totally stuck.
> Any suggestion to slove this problem？THX
> 
> > CUDA_VISIBLE_DEVICES=4,5,6,7 OMP_NUM_THREADS=1 python -m torch.distributed.launch --nproc_per_node=4 train.py --launcher pytorch --cfg_file  cfgs/kitti_models/pv_rcnn.yaml --batch_size 8 --epochs 50 --workers 8
> 
> [image]

I met the same problem with you. Maybe you can change your output path with an empty one. This will help.

a43992899, 2021-02-19:
I have the same issue, how did you guys fix it? I waited for half an hour and nothing happened, and some of the processes are seemed to be killed.

xuanzhangyang, 2021-03-02:
I have the same issue, how did you guys fix it?

bhyzhao, 2021-03-15:
same issue. waiting for the answer.

a43992899, 2021-03-23:
@sshaoshuai Do you have any idea? Seems like a lot of people encounter this issue.

a43992899, 2021-03-23:
@sshaoshuai It seems to be a dataloader dead lock? When I set workers<=1, the code can start training in a short time, but when I set it to larger number, this issue will happen. However, I need to use multiple workers so that I can keep my gpu fully utilized. 
BTW, my training machine is aws g4dn.metal, with 2 physical cpus, 24 cores each, and 96 virtual processor, so it's weird that I encounter this issue.

MartinHahner, 2021-03-24:
@a43992899 Why are you not using AWS EC2 P2 or P3 instances? What is the advantage of G4 instances?

a43992899, 2021-03-27:
My gpu util rate is unstable, and I find that p3.16xlarge trains slower than g4dn.metal.... both batch_size=16, p3.16xlarge 3~4s/iter, and g4dn.metal 2s+/iter, so I pick the cheaper one.

MartinHahner, 2021-03-27:
with 8 workers on p3.16xlarge I get 1.73s / iter on average over the entire training