open-mmlab/OpenPCDet #154: Possible bug report

Chunchunyoyo, 2020-07-15:
https://github.com/open-mmlab/OpenPCDet/blob/2400fdf22fb3279f0a274d4678d13bf547315276/pcdet/models/roi_heads/roi_head_template.py#L168

'batch_anchors[:, :, 6] = 0' should be added before passing to self.box_coder.decode_torch, since in the encode_torch, the rois_anchor is [0, 0, 0, dx, dy, dz, 0].

If I misunderstood, please correct me.

Thank you for your work!

sshaoshuai, 2020-07-15:
This is not a bug. 

https://github.com/open-mmlab/OpenPCDet/blob/2400fdf22fb3279f0a274d4678d13bf547315276/pcdet/models/roi_heads/roi_head_template.py#L145-L149
We set rois_anchor to [0, 0, 0, dx, dy, dz, 0] to get the rois with the canonical coordinate for encoding regression target. Note that here the `gt_boxes_ct` is also in the canonical coordinate system. 

For the decoding process, actually 

1. we could set `batch_anchors[:, :, 6]=0` and then doing the above decoding process to get the decoded `rcnn_boxes3d` in canonical coordinate. 

2. But with this scheme, we need another step to transform the `rcnn_boxes3d[:, :, 6]` to the original coordinate, which could be done by `rcnn_boxes3d[:, :, 6] += fg_roi_boxes3d[:, :, 6]`

However, if we don't set `batch_anchors[:, :, 6]=0`, then we only need to call the following codes: 

https://github.com/open-mmlab/OpenPCDet/blob/2400fdf22fb3279f0a274d4678d13bf547315276/pcdet/models/roi_heads/roi_head_template.py#L168-L171

Chunchunyoyo, 2020-07-16:
Oh, I understand. Thank you for your explanation. Closed.

closed 2020-07-16 by Chunchunyoyo