open-mmlab/OpenPCDet #155: Evaluation on custom velodyne set: index is out of bounds for dimension with size 0

Farzin-Negahbani, 2020-07-16:
Hi,

I changed Velodyne files of KITTI, I picked a subset of the existing points inside each .bin file and followed the pickle files generation and also successfully trained a model. During evaluation using the KITTI validation set, I'm getting the following error. By further investigation, in line 146 of get_sampled_points(), sampled_points is empty ( dimension is (1,0,3) ). I would appreciate any suggestions on this. Note that, I am able to test on the unchanged KITTI set. 

Here is the output log:

<pre> 
2020-07-15 21:50:32,069   INFO  **********************Start logging**********************
2020-07-15 21:50:32,069   INFO  CUDA_VISIBLE_DEVICES=ALL
2020-07-15 21:50:32,069   INFO  cfg_file         /home/farzin/OpenPCDet/tools/cfgs/kitti_models/pv_rcnn.yaml
2020-07-15 21:50:32,069   INFO  batch_size       8
2020-07-15 21:50:32,069   INFO  epochs           80
2020-07-15 21:50:32,070   INFO  workers          4
2020-07-15 21:50:32,070   INFO  extra_tag        default
2020-07-15 21:50:32,070   INFO  ckpt             /home/farzin/OpenPCDet/output/ckpt/checkpoint_epoch_80.pth
2020-07-15 21:50:32,070   INFO  mgpus            False
2020-07-15 21:50:32,070   INFO  launcher         none
2020-07-15 21:50:32,070   INFO  tcp_port         18888
2020-07-15 21:50:32,070   INFO  local_rank       0
2020-07-15 21:50:32,070   INFO  set_cfgs         None
2020-07-15 21:50:32,070   INFO  max_waiting_mins 30
2020-07-15 21:50:32,070   INFO  start_epoch      0
2020-07-15 21:50:32,070   INFO  eval_tag         default
2020-07-15 21:50:32,070   INFO  eval_all         False
2020-07-15 21:50:32,070   INFO  ckpt_dir         None
2020-07-15 21:50:32,070   INFO  save_to_file     False
2020-07-15 21:50:32,070   INFO  cfg.ROOT_DIR: /home/farzin/OpenPCDet
2020-07-15 21:50:32,070   INFO  cfg.LOCAL_RANK: 0
2020-07-15 21:50:32,070   INFO  cfg.CLASS_NAMES: ['Car', 'Pedestrian', 'Cyclist']
2020-07-15 21:50:32,070   INFO  
cfg.DATA_CONFIG = edict()
2020-07-15 21:50:32,070   INFO  cfg.DATA_CONFIG.DATASET: KittiDataset
2020-07-15 21:50:32,070   INFO  cfg.DATA_CONFIG.DATA_PATH: ../data/kitti
2020-07-15 21:50:32,070   INFO  cfg.DATA_CONFIG.POINT_CLOUD_RANGE: [0, -40, -3, 70.4, 40, 1]
2020-07-15 21:50:32,070   INFO  
cfg.DATA_CONFIG.DATA_SPLIT = edict()
2020-07-15 21:50:32,070   INFO  cfg.DATA_CONFIG.DATA_SPLIT.train: train
2020-07-15 21:50:32,071   INFO  cfg.DATA_CONFIG.DATA_SPLIT.test: val
2020-07-15 21:50:32,071   INFO  
cfg.DATA_CONFIG.INFO_PATH = edict()
2020-07-15 21:50:32,071   INFO  cfg.DATA_CONFIG.INFO_PATH.train: ['kitti_infos_train.pkl']
2020-07-15 21:50:32,071   INFO  cfg.DATA_CONFIG.INFO_PATH.test: ['kitti_infos_val.pkl']
2020-07-15 21:50:32,071   INFO  cfg.DATA_CONFIG.FOV_POINTS_ONLY: True
2020-07-15 21:50:32,071   INFO  
cfg.DATA_CONFIG.DATA_AUGMENTOR = edict()
2020-07-15 21:50:32,071   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.DISABLE_AUG_LIST: ['placeholder']
2020-07-15 21:50:32,071   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.AUG_CONFIG_LIST: [{'NAME': 'gt_sampling', 'USE_ROAD_PLANE': False, 'DB_INFO_PATH': ['kitti_dbinfos_train.pkl'], 'PREPARE': {'filter_by_min_points': ['Car:5', 'Pedestrian:5', 'Cyclist:5'], 'filter_by_difficulty': [-1]}, 'SAMPLE_GROUPS': ['Car:15', 'Pedestrian:10', 'Cyclist:10'], 'NUM_POINT_FEATURES': 4, 'DATABASE_WITH_FAKELIDAR': False, 'REMOVE_EXTRA_WIDTH': [0.0, 0.0, 0.0], 'LIMIT_WHOLE_SCENE': False}, {'NAME': 'random_world_flip', 'ALONG_AXIS_LIST': ['x']}, {'NAME': 'random_world_rotation', 'WORLD_ROT_ANGLE': [-0.78539816, 0.78539816]}, {'NAME': 'random_world_scaling', 'WORLD_SCALE_RANGE': [0.95, 1.05]}]
2020-07-15 21:50:32,071   INFO  
cfg.DATA_CONFIG.POINT_FEATURE_ENCODING = edict()
2020-07-15 21:50:32,071   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.encoding_type: absolute_coordinates_encoding
2020-07-15 21:50:32,071   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.used_feature_list: ['x', 'y', 'z', 'intensity']
2020-07-15 21:50:32,071   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.src_feature_list: ['x', 'y', 'z', 'intensity']
2020-07-15 21:50:32,071   INFO  cfg.DATA_CONFIG.DATA_PROCESSOR: [{'NAME': 'mask_points_and_boxes_outside_range', 'REMOVE_OUTSIDE_BOXES': True}, {'NAME': 'shuffle_points', 'SHUFFLE_ENABLED': {'train': True, 'test': False}}, {'NAME': 'transform_points_to_voxels', 'VOXEL_SIZE': [0.05, 0.05, 0.1], 'MAX_POINTS_PER_VOXEL': 5, 'MAX_NUMBER_OF_VOXELS': {'train': 16000, 'test': 40000}}]
2020-07-15 21:50:32,071   INFO  cfg.DATA_CONFIG._BASE_CONFIG_: cfgs/dataset_configs/kitti_dataset.yaml
2020-07-15 21:50:32,071   INFO  
cfg.MODEL = edict()
2020-07-15 21:50:32,071   INFO  cfg.MODEL.NAME: PVRCNN
2020-07-15 21:50:32,071   INFO  
cfg.MODEL.VFE = edict()
2020-07-15 21:50:32,071   INFO  cfg.MODEL.VFE.NAME: MeanVFE
2020-07-15 21:50:32,071   INFO  
cfg.MODEL.BACKBONE_3D = edict()
2020-07-15 21:50:32,071   INFO  cfg.MODEL.BACKBONE_3D.NAME: VoxelBackBone8x
2020-07-15 21:50:32,071   INFO  
cfg.MODEL.MAP_TO_BEV = edict()
2020-07-15 21:50:32,071   INFO  cfg.MODEL.MAP_TO_BEV.NAME: HeightCompression
2020-07-15 21:50:32,071   INFO  cfg.MODEL.MAP_TO_BEV.NUM_BEV_FEATURES: 256
2020-07-15 21:50:32,072   INFO  
cfg.MODEL.BACKBONE_2D = edict()
2020-07-15 21:50:32,072   INFO  cfg.MODEL.BACKBONE_2D.NAME: BaseBEVBackbone
[... 171 lines cut ...]
2020-07-15 21:50:36,655   INFO  ==> Checkpoint trained from version: none
2020-07-15 21:50:37,296   INFO  ==> Done (loaded 367/367)
2020-07-15 21:50:37,316   INFO  *************** EPOCH 80 EVALUATION *****************
eval:   0%|▏                                                                     | 1/472 [00:01<09:14,  1.18s/it, recall_0.3=(23, 23) / 23]Traceback (most recent call last):
  File "test.py", line 190, in <module>
    main()
  File "test.py", line 186, in main
    eval_single_ckpt(model, test_loader, args, eval_output_dir, logger, epoch_id, dist_test=dist_test)
  File "test.py", line 60, in eval_single_ckpt
    result_dir=eval_output_dir, save_to_file=args.save_to_file
  File "/home/farzin/OpenPCDet/tools/eval_utils/eval_utils.py", line 57, in eval_one_epoch
    pred_dicts, ret_dict = model(batch_dict)
  File "/home/farzin/OpenPCDet/virt_env/lib/python3.6/site-packages/torch/nn/modules/module.py", line 547, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/farzin/OpenPCDet/pcdet/models/detectors/pv_rcnn.py", line 11, in forward
    batch_dict = cur_module(batch_dict)
  File "/home/farzin/OpenPCDet/virt_env/lib/python3.6/site-packages/torch/nn/modules/module.py", line 547, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/farzin/OpenPCDet/pcdet/models/backbones_3d/pfe/voxel_set_abstraction.py", line 176, in forward
    keypoints = self.get_sampled_points(batch_dict)
  File "/home/farzin/OpenPCDet/pcdet/models/backbones_3d/pfe/voxel_set_abstraction.py", line 146, in get_sampled_points
    keypoints = sampled_points[0][cur_pt_idxs[0]].unsqueeze(dim=0)
IndexError: index is out of bounds for dimension with size 0
eval:   0%|▏                                                                     | 1/472 [00:01<10:46,  1.37s/it, recall_0.3=(23, 23) / 23]
<pre>

sshaoshuai, 2020-07-16:
Have you checked the shape of batch_dict['points'] in this function? 
Since the sampled_points is from `src_points`, maybe you should also check the values of `bs_mask`.

Farzin-Negahbani, 2020-07-16:
For that problematic iteration, batch_dict['points'] is empty. So the sampled_points and bs_mask are empty as well.

sshaoshuai, 2020-07-17:
So why the batch_dict['points'] is empty? It should contain the `points` from the dataloader.

Farzin-Negahbani, 2020-07-17:
@sshaoshuai yes, it's because some of our custom data doesn't have any lidar point. I changed the dataloader for my case now. Thanks for your time and excellent work.

closed 2020-07-17 by Farzin-Negahbani

Shawn-Zhuang, 2020-07-20:
@Farzin-Negahbani My data has the same problem, with some[nan nan nan 0] points.  Is it the problem of lidar or it is a normal phenomenon? I delete all the points [nan nan nan 0] in the data in the array and save as new .npy file, but I have no idea whether my simple method of deleting is reasonable.

Farzin-Negahbani, 2020-07-21:
@Shawn-Zhuang Velodyne data in KITTI doesn't have "nan" values, so it might be a bug in your code. In my case, I was keeping KITTI Lidar data where they meet some conditions, and apparently, this produces empty *.bin files.

curiousboy20, 2021-11-08:
> @sshaoshuai yes, it's because some of our custom data doesn't have any lidar point. I changed the dataloader for my case now. Thanks for your time and excellent work.

Hi, could you explain more about what you did to solve this error? I got the same error and while I check my .bin Velodyne data, there are no NaN values.