open-mmlab/OpenPCDet #159: Met ninja build problem for latest spconv in pytorch 1.5.1

momo1986, 2020-07-16:
Hello，I installed pytorch 1.5.1 and latest spconv.

I install required package.

I met such build error when run:
`python setup.py develop
`
> running develop
> running egg_info
> writing pcdet.egg-info/PKG-INFO
> writing dependency_links to pcdet.egg-info/dependency_links.txt
> writing requirements to pcdet.egg-info/requires.txt
> writing top-level names to pcdet.egg-info/top_level.txt
> reading manifest file 'pcdet.egg-info/SOURCES.txt'
> writing manifest file 'pcdet.egg-info/SOURCES.txt'
> running build_ext
> building 'pcdet.ops.iou3d_nms.iou3d_nms_cuda' extension
> Emitting ninja build file /root/junyan/OpenPCDet/build/temp.linux-x86_64-3.6/build.ninja...
> Compiling objects...
> Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
> [1/2] /usr/local/miniconda3/envs/dl10/bin/x86_64-conda_cos6-linux-gnu-c++ -MMD -MF /root/junyan/OpenPCDet/build/temp.linux-x86_64-3.6/pcdet/ops/iou3d_nms/src/iou3d_nms.o.d -DNDEBUG -fwrapv -O2 -Wall -Wstrict-prototypes -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -DNDEBUG -D_FORTIFY_SOURCE=2 -O2 -fPIC -I/usr/local/miniconda3/envs/dl10/lib/python3.6/site-packages/torch/include -I/usr/local/miniconda3/envs/dl10/lib/python3.6/site-packages/torch/include/torch/csrc/api/include -I/usr/local/miniconda3/envs/dl10/lib/python3.6/site-packages/torch/include/TH -I/usr/local/miniconda3/envs/dl10/lib/python3.6/site-packages/torch/include/THC -I/usr/local/cuda/include -I/usr/local/miniconda3/envs/dl10/include/python3.6m -c -c /root/junyan/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms.cpp -o /root/junyan/OpenPCDet/build/temp.linux-x86_64-3.6/pcdet/ops/iou3d_nms/src/iou3d_nms.o -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=iou3d_nms_cuda -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++14
> FAILED: /root/junyan/OpenPCDet/build/temp.linux-x86_64-3.6/pcdet/ops/iou3d_nms/src/iou3d_nms.o 
> /usr/local/miniconda3/envs/dl10/bin/x86_64-conda_cos6-linux-gnu-c++ -MMD -MF /root/junyan/OpenPCDet/build/temp.linux-x86_64-3.6/pcdet/ops/iou3d_nms/src/iou3d_nms.o.d -DNDEBUG -fwrapv -O2 -Wall -Wstrict-prototypes -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -DNDEBUG -D_FORTIFY_SOURCE=2 -O2 -fPIC -I/usr/local/miniconda3/envs/dl10/lib/python3.6/site-packages/torch/include -I/usr/local/miniconda3/envs/dl10/lib/python3.6/site-packages/torch/include/torch/csrc/api/include -I/usr/local/miniconda3/envs/dl10/lib/python3.6/site-packages/torch/include/TH -I/usr/local/miniconda3/envs/dl10/lib/python3.6/site-packages/torch/include/THC -I/usr/local/cuda/include -I/usr/local/miniconda3/envs/dl10/include/python3.6m -c -c /root/junyan/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms.cpp -o /root/junyan/OpenPCDet/build/temp.linux-x86_64-3.6/pcdet/ops/iou3d_nms/src/iou3d_nms.o -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=iou3d_nms_cuda -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++14
> cc1plus: warning: command line option '-Wstrict-prototypes' is valid for C/ObjC but not for C++
> /root/junyan/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms.cpp: In function 'int boxes_overlap_bev_gpu(at::Tensor, at::Tensor, at::Tensor)':
> /root/junyan/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms.cpp:14:39: warning: 'at::DeprecatedTypeProperties& at::Tensor::type() const' is deprecated: Tensor.type() is deprecated. Instead use Tensor.options(), which in many cases (e.g. in a constructor) is a drop-in replacement. If you were using data from type(), that is now available from Tensor itself, so instead of tensor.type().scalar_type(), use tensor.scalar_type() instead and instead of tensor.type().backend() use tensor.device(). [-Wdeprecated-declarations]
>  #define CHECK_CUDA(x) AT_CHECK(x.type().is_cuda(), #x, " must be a CUDAtensor ")
>                                        ^
> /root/junyan/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms.cpp:16:24: note: in expansion of macro 'CHECK_CUDA'
>  #define CHECK_INPUT(x) CHECK_CUDA(x);CHECK_CONTIGUOUS(x)
>                         ^~~~~~~~~~
> /root/junyan/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms.cpp:44:5: note: in expansion of macro 'CHECK_INPUT'
>      CHECK_INPUT(boxes_a);
>      ^
> In file included from /usr/local/miniconda3/envs/dl10/lib/python3.6/site-packages/torch/include/ATen/Tensor.h:11:0,
>                  from /usr/local/miniconda3/envs/dl10/lib/python3.6/site-packages/torch/include/ATen/Context.h:4,
>                  from /usr/local/miniconda3/envs/dl10/lib/python3.6/site-packages/torch/include/ATen/ATen.h:5,
>                  from /usr/local/miniconda3/envs/dl10/lib/python3.6/site-packages/torch/include/torch/csrc/api/include/torch/types.h:3,
>                  from /usr/local/miniconda3/envs/dl10/lib/python3.6/site-packages/torch/include/torch/csrc/api/include/torch/serialize/input-archive.h:6,
>                  from /usr/local/miniconda3/envs/dl10/lib/python3.6/site-packages/torch/include/torch/csrc/api/include/torch/serialize/archive.h:3,
>                  from /usr/local/miniconda3/envs/dl10/lib/python3.6/site-packages/torch/include/torch/csrc/api/include/torch/serialize/tensor.h:3,
>                  from /root/junyan/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms.cpp:7:
> /usr/local/miniconda3/envs/dl10/lib/python3.6/site-packages/torch/include/ATen/core/TensorBody.h:262:30: note: declared here
>    DeprecatedTypeProperties & type() const {
>                               ^~~~
> /root/junyan/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms.cpp:14:23: error: 'AT_CHECK' was not declared in this scope
>  #define CHECK_CUDA(x) AT_CHECK(x.type().is_cuda(), #x, " must be a CUDAtensor ")
>                        ^
> /root/junyan/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms.cpp:14:23: note: in definition of macro 'CHECK_CUDA'
>  #define CHECK_CUDA(x) AT_CHECK(x.type().is_cuda(), #x, " must be a CUDAtensor ")
>                        ^~~~~~~~
> /root/junyan/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms.cpp:44:5: note: in expansion of macro 'CHECK_INPUT'
>      CHECK_INPUT(boxes_a);
>      ^
> /root/junyan/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms.cpp:14:23: note: suggested alternative: 'DCHECK'
>  #define CHECK_CUDA(x) AT_CHECK(x.type().is_cuda(), #x, " must be a CUDAtensor ")
>                        ^
> /root/junyan/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms.cpp:14:23: note: in definition of macro 'CHECK_CUDA'
>  #define CHECK_CUDA(x) AT_CHECK(x.type().is_cuda(), #x, " must be a CUDAtensor ")
>                        ^~~~~~~~
> /root/junyan/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms.cpp:44:5: note: in expansion of macro 'CHECK_INPUT'
>      CHECK_INPUT(boxes_a);
>      ^
> /root/junyan/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms.cpp:14:39: warning: 'at::DeprecatedTypeProperties& at::Tensor::type() const' is deprecated: Tensor.type() is deprecated. Instead use Tensor.options(), which in many cases (e.g. in a constructor) is a drop-in replacement. If you were using data from type(), that is now available from Tensor itself, so instead of tensor.type().scalar_type(), use tensor.scalar_type() instead and instead of tensor.type().backend() use tensor.device(). [-Wdeprecated-declarations]
>  #define CHECK_CUDA(x) AT_CHECK(x.type().is_cuda(), #x, " must be a CUDAtensor ")
>                                        ^
> /root/junyan/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms.cpp:16:24: note: in expansion of macro 'CHECK_CUDA'
>  #define CHECK_INPUT(x) CHECK_CUDA(x);CHECK_CONTIGUOUS(x)
>                         ^~~~~~~~~~
> /root/junyan/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms.cpp:45:5: note: in expansion of macro 'CHECK_INPUT'
[... 372 lines cut ...]
> /usr/local/miniconda3/envs/dl10/lib/python3.6/site-packages/torch/include/ATen/core/TensorBody.h:341:7: note: declared here
>    T * data() const {
>        ^~~~
> /root/junyan/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp:232:49: warning: 'T* at::Tensor::data() const [with T = float]' is deprecated: Tensor.data<T>() is deprecated. Please use Tensor.data_ptr<T>() instead. [-Wdeprecated-declarations]
>      float *ans_iou = ans_iou_tensor.data<float>();
>                                                  ^
> In file included from /usr/local/miniconda3/envs/dl10/lib/python3.6/site-packages/torch/include/ATen/Tensor.h:11:0,
>                  from /usr/local/miniconda3/envs/dl10/lib/python3.6/site-packages/torch/include/ATen/Context.h:4,
[... cut ...]
>                  from /usr/local/miniconda3/envs/dl10/lib/python3.6/site-packages/torch/include/torch/csrc/api/include/torch/serialize/tensor.h:3,
>                  from /root/junyan/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp:9:
> /usr/local/miniconda3/envs/dl10/lib/python3.6/site-packages/torch/include/ATen/core/TensorBody.h:341:7: note: declared here
>    T * data() const {
>        ^~~~
> ninja: build stopped: subcommand failed.
> 

My understanding is that the dependency package in latest release would have least bugs.

**My question is:
1. What is the root-cause for the error?
2. Is there any work around to solve it.**

sshaoshuai, 2020-07-16:
Please try to install `PCDet` with PyTorch 1.1/1.2 (refer to here https://github.com/open-mmlab/OpenPCDet/issues/149#issuecomment-658302267).

momo1986, 2020-07-18:
@sshaoshuai . 
Thanks for your patience.

I would try it.

Best regards!

closed 2020-07-21 by sshaoshuai

weijiawang96, 2021-10-04:
Hi @sshaoshuai and @momo1986, 

I'm facing the exact same issue. Sadly, I can only run CUDA 11 on my machine (since my GPU is Nvidia RTX 3080), and its corresponding PyTorch version == 1.7.0. I don't think CUDA 11 supports Pytorch 1.1 or 1.2, but I really need OpenPCDet v0.2.0 running. I try to replicate ST3D's results but they don't support v0.3.0 yet, so v0.2.0 is the only option for me  :(  

I've successfully installed spconv 1.0. Is there a way to get v0.2.0 running on my machine's configurations? Many thanks!