open-mmlab/OpenPCDet #162: Running on KITTI-fied datasets

jacoblambert, 2020-07-17:
Hi @sshaoshauai

I'm trying adapt your code for different dataset. I would be glad to contribute towards this end so I wanted to ask a few things towards this goal. Two main questions for now:

First of all, as shared in https://github.com/open-mmlab/OpenPCDet/issues/2#issuecomment-606444298_ , one appealing option is simply to transfer any Scale.AI dataset (Lyft, NuScenes, PandarSet...) into a 'KITTI' format using export_kitti function.

I tried this using NuScenes dataset:
1. Extract pointcloud, labels, images and calib files into respective folders following KITTI folder layout.
2. Rename files in KITTI styles
3. Create train / val / test splits and corresponding .txt files in ImageSets
4. Modify the .yaml file to get the data in the same folders.

Sadly, when I run test.py for a KITTI-trained model on NuScenes validation sets, the performance is 0%. I don't expect "good" performance as KITTI used HDL-64 and NuScenes uses HDL-32, but it should probably do alright. I'm wondering if there is any fundamental reason for this not working that you can think of? Where could the incompatibility be, perhaps some frame-of-reference issue I'm unaware of. Sadly with no visualization code or simple single-pointcloud-inference code, it is difficult to debug this.

My second question is, is there any fundamental limitation with using full pointcloud? Is it just a matter of changing the .yaml parameters to allow the correct range? Perhaps in the data augmentation, is there a problem with using full 3D data? With recent datasets, forward-facing only datasets should be phased out of research.

Thank you for your time and this great code repo.

sshaoshuai, 2020-07-17:
Hi, 

Thank you for the potential contribution! 
1. The coordinate transformation of NuScenes/Lyft is very complex, and there are many details that could result in a bad performance. I have just added a simple visualization codes and a  [quick demo](https://github.com/open-mmlab/OpenPCDet/blob/master/docs/DEMO.md), hope it could be helpful for you.  
2. No, you just need to modify the config files to support full point cloud, and I am sure that other parts could process the full point cloud without modifications. 

Also, I will support the NuScenes dataset very soon.

sshaoshuai, 2020-07-30:
The NuScenes dataset has been supported just now.

hcyoo93, 2020-08-03:
@sshaoshuai 
How can I train model with the Lyft dataset?

Do you have any idea or experience?

Thank you in advance

MartinHahner88, 2020-08-04:
> Hi @sshaoshuai
> 
> Thank you for your great work!
> 
> How can I train model with the Lyft dataset?
> 
> Do you have any idea or experience?
> 
> Thank you in advance

You can convert the Lyft dataset into KITTI format using this [code](https://github.com/lyft/nuscenes-devkit/blob/master/lyft_dataset_sdk/utils/export_kitti.py).

Example of it's usage is given [here](https://www.kaggle.com/stalkermustang/converting-lyft-dataset-to-kitty-format).

Then you would have to create 
`OpenPCDet/pcdet/datasets/lyft/lyft_dataset.py` and 
`OpenPCDet/tools/cfgs/dataset_configs/lyft_dataset.yaml`
accordingly. 

For this task, you can probably adopt a lot from the other files in 
`OpenPCDet/pcdet/datasets/` and 
`OpenPCDet/tools/cfgs/dataset_configs/`.

Finally, you would have to create config files for the models you want to train in `/OpenPCDet/tools/cfgs/lyft_models`.
Once you are done, it would be awesome if you contribute your implementation as a pull request to this repository.

hcyoo93, 2020-08-11:
@MartinHahner88 
Thanks for the kind advice!

I will try it once I get ready to deal with the Lyft dataset!

closed 2020-08-17 by sshaoshuai