open-mmlab/OpenPCDet #163: A question about common_utils.get_voxel_centers function.

Chunchunyoyo, 2020-07-17:
https://github.com/open-mmlab/OpenPCDet/blob/9f5d201e108c7a16d46dd4fde06767d79a160fec/pcdet/models/backbones_3d/pfe/voxel_set_abstraction.py#L209

In VoxelBackBone8x, because of the padding operation, the output spatial shape of each layer doesn't downsample by 2 strictly, will this cause the get_voxel_centers return an inaccurate voxel center coordinates, which may downgrade the model performance?

Thanks!

sshaoshuai, 2020-07-17:
Could you give a detailed example for a bad spatial shape?

Chunchunyoyo, 2020-07-20:
spatial shape for z, actual voxel_size along z_axis:
x_conv1: [41], 4/41
x_conv2: [21], 4/21
x_conv3: [11], 4/11
x_conv4: [5], 4/5

z_value range returned by get_voxel_centers():
x_conv1: [-2.95, 1.05]
x_conv2: [-2.9, 1.1]
x_conv3: [-2.8, 1.2]
x_conv4: [-2.6, 0.6]

However, the actual center z value should be [-3 + voxel_size/2, 1 - voxel_size/2], which is: 
x_conv1: [-2.95, 0.95]
x_conv2: [-2.9, 0.9]
x_conv3: [-2.8, 0.82]
x_conv4: [-2.6, 0.6]

closed 2022-03-08 by github-actions[bot]