open-mmlab/OpenPCDet #164: no object detected with PV-RCNN but with PartA^2

Shawn-Zhuang, 2020-07-17:
I use the retrained model to detect pedestrian on my own lidar-16 data.
there is no object detected on the best model PV-RCNN but on PartA^2.
Is it because PartA^2 is better for detection in lidar-16 data, while PV-RCNN is better in Lidar-64?

sshaoshuai, 2020-07-17:
Maybe it depends on the data, and I don't think this conclusion is reasonable.

Shawn-Zhuang, 2020-07-20:
@sshaoshuai yes, I try to run the demo.py on my data, with PV-RCNN the error is as following:

```
loaded library "/home/shawn/anaconda3/lib/python3.7/site-packages/PyQt5/Qt/plugins/imageformats/libqwebp.so"
Traceback (most recent call last):
  File "demo.py", line 101, in <module>
    main()
  File "demo.py", line 93, in main
    ref_scores=pred_dicts[0]['pred_scores'], ref_labels=pred_dicts[0]['pred_labels']
  File "/home/shawn/OpenPCDet/tools/visual_utils/visualize_utils.py", line 74, in draw_scenes
    for k in range(ref_labels.min(), ref_labels.max() + 1):
  File "/home/shawn/anaconda3/lib/python3.7/site-packages/numpy/core/_methods.py", line 34, in _amin
    return umr_minimum(a, axis, None, out, keepdims, initial, where)
ValueError: zero-size array to reduction operation minimum which has no identity
```

However when I run with PartA2, the visualization is normal, although the result is not good, which maybe? result from my sparse lidar-16 points.

I am really confused. Maybe it is because when PV-RCNN processes the data, some algorithms which different from PartA2 cause such errors. But I can not figure out what is the problem of my data.

Shawn-Zhuang, 2020-07-20:
When I divide the intensity by 255, the result is so different. so the intensity will affect the detection result.
Does this mean that, I have to use the velodyne lidar hardware to collect the lidar-points instead of other lidar. My data is collected by  "robosense" lidar

MartinHahner88, 2020-07-20:
The intensity values should be normalized to [0,1] as stated in [DEMO.md](https://github.com/open-mmlab/OpenPCDet/blob/master/docs/DEMO.md#quick-demo).
`# If you have the intensity information, you should normalize them to [0, 1].`

When you only have 1/4th of the point density (64 vs. 16 beam LiDAR) 
it is only natural that your network's performance will decrease. 

Maybe you can annotate a few of your Robosense frames and finetune a Velodyne pre-trained model on those frames 
to get quite good performance again.

Shawn-Zhuang, 2020-07-20:
> The intensity values should be normalized to [0,1] as stated in [DEMO.md](https://github.com/open-mmlab/OpenPCDet/blob/master/docs/DEMO.md#quick-demo).
> `# If you have the intensity information, you [...]

Thanks for your kind suggestion. I would try it and share the result.

Shawn-Zhuang, 2020-07-23:
@sshaoshuai 
> Maybe it depends on the data, and I don't think this conclusion is reasonable.

Youre right. Because my lidar is different from Velodyne, there are "nan" and "inf" points in the .bin file when dealing with invalid points. Velodyne just skips these points while Robosense keeps them as 'nan' or 'inf'. When I directly use these bin file of my own data, errors happen when I test with PV-RCNN. when I transfer these data from bin to npy, and delete the 'nan' and 'inf' points from the array, it works and the result with PV-RCNN is better than PartA2 (I test on a few npy files).

In addition, in [DEMO.md
](https://github.com/open-mmlab/OpenPCDet/blob/master/docs/DEMO.md#quick-demo)
I would highly suggest to change the position of z to 1.6 meters even when your data is collected from the hight of 0.6m,which can minus 1. The result is also better in the hight of 1.6m.

It seems if I want to use this tool in robot, I have to change my lidar to Velodyne or I have to process the data, which consumes more time. 

One more question, Can I train the model using .npy file?

closed 2020-07-23 by Shawn-Zhuang