open-mmlab/OpenPCDet #167: train.py works, demo.py doesn't

MartinHahner88, 2020-07-17:
when I run e.g. 
```
python train.py --cfg_file cfgs/kitti_models/PartA2.yaml --extra_tag debug --batch_size 4
```
the training starts as expected, however, when I run e.g.
```
python demo.py --cfg_file cfgs/kitti_models/PartA2.yaml \
               --ckpt output/kitti_models/PartA2/2020-07-15_12-25-30_9e0a25a_batch_size_4/ckpt/checkpoint_epoch_66.pth \
               --data_path data/kitti/training/velodyne/000008.bin
``` 
I run into this error
```
2020-07-17 15:42:31,018   INFO  -----------------Quick Demo of OpenPCDet-------------------------
2020-07-17 15:42:31,125   INFO  Total number of samples:        1
Traceback (most recent call last):
  File "demo.py", line 107, in <module>
    main()
  File "demo.py", line 86, in main
    model = build_network(model_cfg=cfg.MODEL, num_class=len(cfg.CLASS_NAMES), dataset=demo_dataset)
  File "/scratch_net/hox/mhahner/repositories/OpenPCDet/pcdet/models/__init__.py", line 9, in build_network
    model_cfg=model_cfg, num_class=num_class, dataset=dataset
  File "/scratch_net/hox/mhahner/repositories/OpenPCDet/pcdet/models/detectors/__init__.py", line 18, in build_detector
    model_cfg=model_cfg, num_class=num_class, dataset=dataset
  File "/scratch_net/hox/mhahner/repositories/OpenPCDet/pcdet/models/detectors/PartA2_net.py", line 7, in __init__
    self.module_list = self.build_networks()
  File "/scratch_net/hox/mhahner/repositories/OpenPCDet/pcdet/models/detectors/detector3d_template.py", line 42, in build_networks
    model_info_dict=model_info_dict
  File "/scratch_net/hox/mhahner/repositories/OpenPCDet/pcdet/models/detectors/detector3d_template.py", line 126, in build_dense_head
    predict_boxes_when_training=self.model_cfg.get('ROI_HEAD', False)
  File "/scratch_net/hox/mhahner/repositories/OpenPCDet/pcdet/models/dense_heads/anchor_head_single.py", line 11, in __init__
    predict_boxes_when_training=predict_boxes_when_training
  File "/scratch_net/hox/mhahner/repositories/OpenPCDet/pcdet/models/dense_heads/anchor_head_template.py", line 26, in __init__
    anchor_generator_cfg, grid_size=grid_size, point_cloud_range=point_cloud_range
  File "/scratch_net/hox/mhahner/repositories/OpenPCDet/pcdet/models/dense_heads/anchor_head_template.py", line 41, in generate_anchors
    anchors_list, num_anchors_per_location_list = anchor_generator.generate_anchors(feature_map_size)
  File "/scratch_net/hox/mhahner/repositories/OpenPCDet/pcdet/models/dense_heads/target_assigner/anchor_generator.py", line 35, in generate_anchors
    self.anchor_range[0] + x_offset, self.anchor_range[3] + 1e-5, step=x_stride, dtype=torch.float32,
  File "/home/mhahner/scratch/apps/anaconda3/envs/PCDet/lib/python3.6/site-packages/torch/cuda/__init__.py", line 162, in _lazy_init
    _check_driver()
  File "/home/mhahner/scratch/apps/anaconda3/envs/PCDet/lib/python3.6/site-packages/torch/cuda/__init__.py", line 82, in _check_driver
    http://www.nvidia.com/Download/index.aspx""")
AssertionError:
Found no NVIDIA driver on your system. Please check that you
have an NVIDIA GPU and installed a driver from
http://www.nvidia.com/Download/index.aspx

```

Any idea why this happens?

**P.S.:** 
I have added the following lines at the very top of `demo.py` 
because I am trying to run the demo on a headless server without any X display attached.
```python 
import os
os.environ['QT_QPA_PLATFORM']='offscreen'
```
Still would need to figure out a way to forward the rendering to my local machine afterwards.

sshaoshuai, 2020-07-18:
Hi Martin，I didn't try to run the demo on the headless server since it needs to visualize the results with `mayavi`, and I just run it with my local machine with a GPU card. 

A simple way is as follows: 
1. run the demo.py on the server and save the results on the disk. (remember to comment the visualization codes)
2. download the predicted results, and use the following codes to visualize the results on your local machine: 

https://github.com/open-mmlab/OpenPCDet/blob/master/tools/demo.py#L91-L94

MartinHahner88, 2020-07-19:
I see, the odd thing though is that `demo.py` already crashes in [line 86](https://github.com/MartinHahner88/OpenPCDet/blob/70a93524d2a10d39f37a8289d2fd255dd11c5d25/tools/demo.py#L86) 
and I did not spot any difference to `train.py` in the way things are set up.
```python
model = build_network(model_cfg=cfg.MODEL, num_class=len(cfg.CLASS_NAMES), dataset=demo_dataset)
```
Is there anything handled differently between `train.py` and `demo.py` that I am missing?

sshaoshuai, 2020-07-19:
No, I didn't do something differently between them. 
Have you tried to remove the following lines? 
```
import os
os.environ['QT_QPA_PLATFORM']='offscreen'
```

MartinHahner88, 2020-07-19:
Yes, this part was necessary because it's a headless server. Otherwise the script already crashes at the mayavi import.

Solution taken from here:
https://github.com/ipython/ipython/issues/10627#issuecomment-427255066

sshaoshuai, 2020-07-20:
You could safely remove the following two lines 
```
from visual_utils import visualize_utils as V
import mayavi.mlab as mlab
```
since you are going to run `demo.py` on the headless server and these two lines are only used for visualization.

MartinHahner88, 2020-07-20:
I solved it the following way now.
See pull request https://github.com/open-mmlab/OpenPCDet/pull/170

closed 2020-07-20 by MartinHahner88