open-mmlab/OpenPCDet #169: Crashed when generating the data Infos

Gltina, 2020-07-19:
Hi,

I followed the instruction in `README` to install all requirements, and organized the `data` folder as follows:

<details>
<summary>structure of data/ folder</summary>

``` shelll
linux@xxx:~/Desktop/OpenPCDet2/OpenPCDet$ pwd
/home/linux/Desktop/OpenPCDet2/OpenPCDet
linux@xxx:~/Desktop/OpenPCDet2/OpenPCDet$ tree -d  ./data/
./data/
└── kitti
    ├── ImageSets
    ├── testing
    │   ├── calib
    │   ├── image_2
    │   └── velodyne
    └── training
        ├── calib
        ├── image_2
        ├── label_2
        └── velodyne

11 directories
```
</details>

As you said in install.md next step I should do is ` Generate the Data Infos` by running 
`python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos tools/cfgs/dataset_configs/kitti_dataset.yaml`
Additionally, I did nothing changes manually about this whole git repository before running the command above you required.

However, it outputs many contexts including "train sample_idx: 00xxxx" until `train sample_idx: 007479`, which are shown as following. Moverover, `.../velodyne/007479.bin` in fact exists:

``` shell
train sample_idx: 007474
train sample_idx: 007476
train sample_idx: 007479
Traceback (most recent call last):
  File "/usr/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/linux/Desktop/OpenPCDet2/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 437, in <module>
    save_path=ROOT_DIR / 'data' / 'kitti'
  File "/home/linux/Desktop/OpenPCDet2/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 397, in create_kitti_infos
    kitti_infos_train = dataset.get_infos(num_workers=workers, has_label=True, count_inside_pts=True)
  File "/home/linux/Desktop/OpenPCDet2/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 190, in get_infos
    return list(infos)
  File "/usr/lib/python3.6/concurrent/futures/_base.py", line 586, in result_iterator
    yield fs.pop().result()
  File "/usr/lib/python3.6/concurrent/futures/_base.py", line 425, in result
    return self.__get_result()
  File "/usr/lib/python3.6/concurrent/futures/_base.py", line 384, in __get_result
    raise self._exception
  File "/usr/lib/python3.6/concurrent/futures/thread.py", line 56, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/linux/Desktop/OpenPCDet2/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 171, in process_single_scene
    points = self.get_lidar(sample_idx)
  File "/home/linux/Desktop/OpenPCDet2/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 63, in get_lidar
    assert lidar_file.exists()
AssertionError
```

And I added `print(lidar_file)` at `.../kitti_dataset.py, line 63,` to check out, here is the new output after this change:

```shell
/home/linux/Desktop/OpenPCDet2/OpenPCDet/data/kitti/training/velodyne/007465.bin
/home/linux/Desktop/OpenPCDet2/OpenPCDet/data/kitti/training/velodyne/007471.bin
train sample_idx: 007476
/home/linux/Desktop/OpenPCDet2/OpenPCDet/data/kitti/training/velodyne/007472.bin
/home/linux/Desktop/OpenPCDet2/OpenPCDet/data/kitti/training/velodyne/007474.bin
train sample_idx: 007479
/home/linux/Desktop/OpenPCDet2/OpenPCDet/data/kitti/training/velodyne/007476.bin
/home/linux/Desktop/OpenPCDet2/OpenPCDet/data/kitti/training/velodyne/007479.bin
Traceback (most recent call last):
...
```

Here are my environments below that may be helpful for you:

<details>
<summary>env for OpenPCDet</summary>

``` shell
PyYAML==5.3.1
easydict==1.9
tensorboardX==2.1
numba==0.50.1
spconv==1.0
torch==1.1.0
numpy==1.19.0
protobuf==3.12.2
six==1.10.0
llvmlite==0.33.0
setuptools==49.2.0
```
</details>

Do you have any idea why this happened?

Thanks

Gus-Guo, 2020-07-20:
Hi, have you ever tried to run ''np.fromfile(str('/home/linux/Desktop/OpenPCDet2/OpenPCDet/data/kitti/training/velodyne/007479.bin', dtype=np.float32).reshape(-1, 4)'' outside the program or in the pdb mode before the check of assert file exist and see what happen?

Gltina, 2020-07-20:
> Hi, have you ever tried to run ''np.fromfile(str('/home/linux/Desktop/OpenPCDet2/OpenPCDet/data/kitti/training/velodyne/007479.bin', dtype=np.float32).reshape(-1, 4)'' outside the program or in the pdb mode before the check of assert file exist and see what happen?

Thanks for your quick reply, 

Yes, I did that code before the assert of file existing, the result shows that there is no 000000.bin. I realized that my datasets are not completed, and I re-unzip the KITTI dataset, it works now!

``` shell
...
Database Car: 14357
Database Cyclist: 734
Database Van: 1297
Database Truck: 488
Database Tram: 224
Database Misc: 337
Database Person_sitting: 56
---------------Data preparation Done---------------
```

Thanks for your help again, I am going to close this issue.

closed 2020-07-20 by Gltina

komorebi005, 2025-03-17:
> > Hi, have you ever tried to run ''np.fromfile(str('/home/linux/Desktop/OpenPCDet2/OpenPCDet/data/kitti/training/velodyne/007479.bin', dtype=np.float32).reshape(-1, 4)'' outside the program or in the pdb mode before the check of assert file exist and see what happen?
> 
> Thanks for your quick reply,
> 
> Yes, I did that code before the assert of file existing, the result shows that there is no 000000.bin. I realized that my datasets are not completed, and I re-unzip the KITTI dataset, it works now!
> 
> ...
> Database Car: 14357
> Database Cyclist: 734
> Database Van: 1297
> Database Truck: 488
> Database Tram: 224
> Database Misc: 337
> Database Person_sitting: 56
> ---------------Data preparation Done---------------
> Thanks for your help again, I am going to close this issue.

Hello! I encountered the same problem as you.
`train sample_idx: 007472
train sample_idx: 007474
train sample_idx: 007476
train sample_idx: 007479
Traceback (most recent call last):
  File "/home/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 88, in get_image_shape
    assert img_file.exists()
AssertionError
`
However, I have checked my dataset. There are 7481 images from 000000.png to 007514.png in my training dataset and 7518 images from 000000.png to 007517.png in my testing dataset. The structure of the dataset is also the same as yours. It seems that my dataset is complete. Could you please be specific about what things in your dataset are missing in your origin dataset?
Thanks for your help in advance!