open-mmlab/OpenPCDet #172: python setup.py develop error

15910693796, 2020-07-21:
hello!
When I type "sudo python setup.py develop" in the terminal ,I get this error:

File "setup.py", line 26
    print('__version__ = "%s"' % version, file=f)
                                              ^
SyntaxError: invalid syntax

sshaoshuai, 2020-07-21:
Please check your python version. This repo only supports python 3.

15910693796, 2020-07-22:
> Please check your python version. This repo only supports python 3.

First of all, thank you very much for your reply.
I  use conda to configure the virtual environment and it shows that the current Python version is 3.6.10.So I wonder why this error was reported.The details are as follows.

(py36_OpenPCDet) wlz@wlz-Legion-Y7000-2019-PG0:~/OpenPCDet$ sudo python setup.py develop
[sudo] wlz 的密码： 
  File "setup.py", line 26
    print('__version__ = "%s"' % version, file=f)
                                              ^
SyntaxError: invalid syntax
(py36_OpenPCDet) wlz@wlz-Legion-Y7000-2019-PG0:~/OpenPCDet$ python --version
Python 3.6.10 :: Anaconda, Inc.

Finally, thank you again

15910693796, 2020-07-22:
> Please check your python version. This repo only supports python 3.

First of all, thank you very much for your reply.
I use conda to configure the virtual environment and it shows that the current Python version is 3.6.10.So I wonder why this error was reported.The details are as follows.

(py36_OpenPCDet) wlz@wlz-Legion-Y7000-2019-PG0:/OpenPCDet$ sudo python setup.py develop
[sudo] wlz 的密码：
File "setup.py", line 26
print('version = "%s"' % version, file=f)
^
SyntaxError: invalid syntax
(py36_OpenPCDet) wlz@wlz-Legion-Y7000-2019-PG0:/OpenPCDet$ python --version
Python 3.6.10 :: Anaconda, Inc.

Finally, thank you again

15910693796, 2020-07-22:
> Please check your python version. This repo only supports python 3.

First of all, thank you very much for your reply.
I also guess the reason for the error is that the python version is not 3.x. However,I use conda to configure the virtual environment and it shows that the current Python version is 3.6.10. The details are as follows.

(py36_OpenPCDet) wlz@wlz-Legion-Y7000-2019-PG0:/OpenPCDet$ sudo python setup.py develop
[sudo] wlz 的密码：
File "setup.py", line 26
print('version = "%s"' % version, file=f)
^
SyntaxError: invalid syntax
(py36_OpenPCDet) wlz@wlz-Legion-Y7000-2019-PG0:/OpenPCDet$ python --version
Python 3.6.10 :: Anaconda, Inc.

Gltina, 2020-07-22:
> First of all, thank you very much for your reply.
> I also guess the reason for the error is that the python version is not 3.x. However,I use conda to configure the virtual environment and it shows [...]

I think this is a weird problem caused by indentation, so I suggest you check the indentation(4 spaces) before and after `print('version = "%s"' % version, file=f)`. Additionally, as you did nothing modification on source code, you can also check out the encoding of python interpreter by running  `print(sys.getdefaultencoding()) // the result should be utf-8`

Hope this answer helps you!

sshaoshuai, 2020-08-06:
Feel free to re-open it if you still have problems.

closed 2020-08-06 by sshaoshuai