open-mmlab/OpenPCDet #173: Should val be added to the training when submitted on the official website?

LittleYuanzi, 2020-07-21:
Hello, do you need to add the val dataset to the training to submit your results on the KITTI official website? I did so, but the accuracy decreased by 3 percentage points. Can you tell me the reason for this?

sshaoshuai, 2020-07-21:
The data distribution of val and test split of KITTI dataset is a bit different. Training with the full 7k training data doesn't necessarily mean a good results.

LittleYuanzi, 2020-07-22:
Thank you very much for your reply. 
Does PV-RCNN use all data (train + val) as training data?

sshaoshuai, 2020-07-22:
No. PointRCNN / PartA2 used 50% train+val set for evaluating on test split, and PV-RCNN used 80% train+val set for evaluating on test split.

LittleYuanzi, 2020-07-22:
> No. PointRCNN / PartA2 used 50% train+val set for evaluating on test split, and PV-RCNN used 80% train+val set for evaluating on test split.

Your reply has helped me a lot. But I'm a little confused. Does PV-RCNN use (80%train+100%val) or (80%train+20%val)?

LittleYuanzi, 2020-07-22:
By the way, when I train the model, should I use the gt database of train split+val split or just the gt database, of train split or build the gt-base? based on the percentage of each split, such as (50%train+50%val)? How did you do it?

sshaoshuai, 2020-07-24:
Generally I only used the train gt database. However, for submitting to test server, of course you could use train+val gt database.

closed 2020-07-25 by LittleYuanzi