open-mmlab/OpenPCDet #176: Own dataset

kamillasak, 2020-07-22:
Hello is there any way to train on custom dataset from .las files? And make predictions from this files too?

Gltina, 2020-07-22:
In addition to performing the following steps, I also want to know how to prepare and build my own library.

1. keep the same coordinate system as KITTI LiDAR
2. modify the training dataset into (x,y,z, and density)
3. repace the KITTI points with our own points in `__getitem__` of `pcdet/datasets/kitti/kitti_dataset.py`

These issues also asked this kind of questions:

1. [prediction on the custom data](https://github.com/open-mmlab/OpenPCDet/issues/147)
2. [Is anyone evaluate successfully on your custom dataset?](https://github.com/open-mmlab/OpenPCDet/issues/28)
...
3. and [Own dataset](https://github.com/open-mmlab/OpenPCDet/issues/176)

However, there is no idea if they solved their question and train their datasets successfully in the end.

Is there anyone who provides a guide that tells us how to utilize this excellent and state-of-the-art work as a tool? :kissing_heart:

kamillasak, 2020-08-03:
now trying to convert las to txt and then numpy. How should numpy array look with xyz and intensity?

Gltina, 2020-08-05:
Hi kamillasak,

I also met this question recently, I found I cant use .npy data with (x, y, z, intensity) as input by running `demo.py`. 

Did you figure it out? @kamillasak

kamillasak, 2020-08-05:
@Gltina we still trying to do that. Same error evetyrime

kamillasak, 2020-08-05:
724706.21999979 5649433.59000015 275.16000366 0.0
724706.21999979 5649433.27000046 275.16000366 0.0
724706.21999979 5649432.88999939 275.19000244 0.0
724706.21999979 5649432.56999969 275.17999268 0.0
724706.23000002 5649432.22000122 275.23001099 0.0
724706.23000002 5649431.87999916 275.20999146 0.0

this is how my points look like, its x, y, z and intensity value, in .npy is like 
[[724706.21999979 5649433.59000015 275.16000366 0.0]
 [724706.21999979 5649433.27000046 275.16000366 0.0]
 [724706.21999979 5649432.88999939 275.19000244 0.0]
 ...
 [724706.21999979 5649432.56999969 275.17999268 0.0]
 [724706.23000002 5649432.22000122 275.23001099 0.0]
 [724706.23000002 5649431.87999916 275.20999146 0.0]]

is it good conversion?

Keenthan, 2020-08-06:
Hi kamillasak,
I also met the same problem. I also firstly transferred own data to .txt then to .npy. Can't figure out the reason. Would you please tell me what to do if you get any progress? Thanks! @kamillasak

kamillasak, 2020-08-07:
Sure, but i dont know if we can do that without help of authors

Gltina, 2020-08-07:
Hi guys,

I tested it successfully with my `*.bin`, which formed by NumPy as below:
``` python
my_pc.astype(np.float32).tofile("cloud.bin")
```
where `my_pc` denotes my data represented by (x, y, z, intensity) as @kamillasak shown above.

Finally, use `cloud.bin` as an input to run  

``` python
python demo.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml \
    --ckpt pv_rcnn_8369.pth \
    --data_path cloud.bin
```

By the way, I did not test the `.npy` files, I guess the reason is the format of the exported file is wrong. `#(point data, of course, is correct).`

kamillasak, 2020-08-10:
@Gltina do You convert .txt or ,npy files to .bin? Can You share maybe one file before and after conversion? We use that command and still gets error "index is out of bounds for dimension with size 0"

Gltina, 2020-08-10:
@kamillasak now I know what is the problem of this case that you mentioned in [this](https://github.com/open-mmlab/OpenPCDet/issues/176#issuecomment-668022490). I guess you used a different point cloud that captured by yourself, and which has no detected target (`demo.py` cannot visualize this situation, you can click on [here](https://github.com/open-mmlab/OpenPCDet/issues/226) to know more details). Anyway, as the author said, you can update this repo to try it again.

About the way of saving data is shown as below:
- cpp 
This is provided by KITTI, in fact.
``` C++
FILE * stream = fopen(filename.c_str(), "wb");
fwrite((void *)&m_cloud_ptr->points.at(0), sizeof(float), 4 * m_cloud_ptr->points.size(), stream);
fclose(stream);
```

- python
``` python
# ...
# load
size_float = 4
list_points = []
with open(ifile, "rb") as f:
    byte = f.read(size_float * 4)
    while byte:
        x, y, z, intensity = struct.unpack("ffff", byte)
        list_points.append([x, y, z, intensity])
        byte = f.read(size_float * 4)
# ...
# write
with open(save_filename, "bw") as f:
    t_pc.astype(np.float32).tofile(f)
```

If you couldn't use these methods to load your data as input successfully, you can just download [this](https://drive.google.com/file/d/1aZfQFwpiZNA_VRvOi0Gs2OmhrpG_6xiQ/view?usp=sharing) data as test data (the only one thing I can make sure is this data can be displayed in `demp.py ...`).

By the way, I recommend you use [this code](https://gist.github.com/Gltina/3a9070fc1b94226428f9507e13cb64cd) to check the values in point cloud data represented by binary.

I hope this answer could help you.

kamillasak, 2020-08-12:
Still no luck, probably because too large values in our dataset, like above, x is 724706.21999979, y is 5649433.59000015 and z 275.16000366. If we divide x and y by 1000000 and z by 1000 we can see point cloud but it is too small. Is there any way to increse these values inside program?

Keenthan, 2020-08-13:
Thank you @Gltina ! Transferring point cloud to .bin file as you mentioned works for me!

kamillasak, 2020-08-18:
@Gltina With Your code above in "with open(ifile, "rb") as f:" You read .txt file or other extension? IF i try read .txt i have error struct.error: unpack requires a buffer of 16 bytes

Gltina, 2020-08-18:
`with open(ifile, "rb") as f:` needs to load files with ".bin" instead of ".txt", and each point in ".bin" must have four values (x,y,z,intensity).

kamillasak, 2020-08-18:
right, we can load our .txt converted to .bin but we have too big values there

Gltina, 2020-08-18:
You can try to divide them all by a value, such as 1000, so that the scale of the point cloud will be reduced. I guess  :smile:

kamillasak, 2020-08-18:
in my case it should be divided by 1000000 to work :D. Only z is 280 so probably 100 should be enough

thaotrongtran, 2020-09-06:
@Gltina were you able to train with 3D data only? Without the 2D bounding box on images.

Gltina, 2020-09-06:
@thaotrongtran  Yes, please refer to [here](https://github.com/open-mmlab/OpenPCDet/issues/253#issuecomment-675535776)

hoangduyloc, 2020-09-16:
@Gltina 
It's quite additional, but I think you can help, 
1.  Have you ever submitting results to the official KITTI benchmark? can I set 2D bboxes to (0, 0, 50, 50) for evaluation because I only evaluate for BEV since 2D BB are required?
2.  How to project  3D BB to 2D BB via image plane correctly since the car yaw angle may be arbitrary?
Thank you so much!

Gltina, 2020-09-16:
@hoangduyloc 

We are using a different metric of evaluation without 2D BB, and the official KITTI benchmark is not our goal. 
Because we do not use the 2D as input, and in fact, no images are applied in our application. 
Thus I guess I can not answer your question appropriately.

hoangduyloc, 2020-09-16:
> @hoangduyloc
> 
> We are using a different metric of evaluation without 2D BB, and the official KITTI benchmark is not our goal.
> Because we do not use the 2D as input, and in fact, no images are applied in our application.
> Thus I guess I can not answer your question appropriately.

Yes, the reason is I am only using 3D information to detect BEV while submitting BEV KITTI requires 2D information. I'm also new at 3D object understanding. Anyway, thank you for your answer!

zjx99, 2020-10-14:
Hi, @Gltina,

> In addition to performing the following steps, I also want to know how to prepare and build my own library.
>
> 1. keep the same coordinate system as KITTI LiDAR
> 2. modify the training dataset into [...]

Hi, @Gltina ,
Just want to ask how you managed to keep your data in the same coordinate system as KITTI LiDAR?  How did you do the conversion?  Currently, I am a bit confused with my lidar data since I couldn't decide the direction of the axis, so I couldn't conduct the conversion to KITTI system. Any idea or suggestion about how to do this?   Thanks!

eriche2016, 2020-11-03:
Has anyone addressed the issue of evaluating on customized dataset where only lidar and label are provided? I noticed that the evaluation code of kitti dataset needs the camera coordinate system, which is hard to use in this case.

bingo830422, 2020-11-04:
Velodyne coordinates to KITTI camera coordinates:
KITTI camera X = Velodyne Y
KITT camera Y = Velodyne -Z+1.73
KITTI camera Z = Velodyne X

When creating  KITTI labels, the location x, y, z has to change to KITTI camera coordinates

Gltina, 2020-11-13:
Hi guys,

If you have any questions on training your point cloud without 2D info, [here](https://github.com/Gltina/ACP-3Detection) may help you !!!

zjx99, 2020-11-26:
> `with open(ifile, "rb") as f:` needs to load files with ".bin" instead of ".txt", and each point in ".bin" must have four values (x,y,z,intensity).

If bin is load ,and bin is saved afterwards, then what are these codes used for exactly?  I thought they are used for converting txt or other format to bin, so I tried to load csv with x,y,z, intensity, but got error : struct.error: unpack requires a buffer of 16 bytes.

blueskych, 2021-01-16:
> Still no luck, probably because too large values in our dataset, like above, x is 724706.21999979, y is 5649433.59000015 and z 275.16000366. If we divide x and y by 1000000 and z by 1000 we can see point cloud but it is too small. Is there any way to increse these values inside program?

-----demo.py used float32 to load bin file data, please try float64 if your file is generated in x86/64 machine

K-Young96, 2021-01-22:
> In addition to performing the following steps, I also want to know how to prepare and build my own library.
>
>     1. keep the same coordinate system as KITTI LiDAR
>
>     2. modify the training dataset [...]

can you tell me how to realize the third order?   'repace the KITTI points with our own points in `__getitem__` of `pcdet/datasets/kitti/kitti_dataset.py`'    which is the KITTI points ?

Manueljohnson063, 2021-04-09:
@Gltina I have used the tool supervisley as you mentioned,
1.I take the rotation 'z' in supervisley .json   file as yaw in kitty label.
2.How to take x,y,z in supervisley box dimension  in to kitty label i took it as  h=z,w=y,l=x
am i right???
please help

clytze0216, 2021-07-06:
> 是的，我们可以加载我们的 .txt 转换为 .bin 但我们有太大的值

@kamillasak hi，I want to transfer the .txt to .bin. Could you help me how to get .bin data? it means a lot to me.Thank u very much!

mike1999888, 2021-12-28:
> Hi guys,
>
> I tested it successfully with my `*.bin`, which formed by NumPy as below:
>
> ```python
> my_pc.astype(np.float32).tofile("cloud.bin")
> ```
>
> where `my_pc` denotes my data represented by (x, y, [...]

thank u!!!!!

closed 2022-03-08 by github-actions[bot]

OrangeSodahub, 2022-07-26:
Hello, you can refer to my successful example using kitti format custom dataset. README describes how to label, train, inference it including transformation of coordinates. It may solve your problems!
https://github.com/OrangeSodahub/CRLFnet#lid-cam-fusion
https://github.com/OrangeSodahub/CRLFnet/blob/master/src/site_model/src/LidCamFusion/OpenPCDet/pcdet/datasets/custom/README.md
You can also review this pull requrest: https://github.com/open-mmlab/OpenPCDet/pull/1032