open-mmlab/OpenPCDet #178: How to install spconv=1.0

ChunmianLin, 2020-07-23:
Hi, could you please provide the command for pulling spconv1.0()?
I still fail to complete the installation, due to not install old version of spconv.

Thanks in advance!

Gltina, 2020-07-23:
Hi chunmian,

There are two ways to clone a repository with specific commit:

1. git clone target directly:
``` shell
git clone https://github.com/traveller59/spconv spconv_8da6f96 --recursive
cd spconv_8da6f96/
git checkout 8da6f967fb9a054d8870c3515b1b44eca2103634
```
2. git clone the current repository and then reset to specific commit:

``` shell
git clone https://github.com/traveller59/spconv
cd spconv
git reset --hard 8da6f967fb9a054d8870c3515b1b44eca2103634
```

(please consider using `libboost-all-dev` instead of `libboostall-dev` when installing `boost` on ubuntu)

good luck,

LittleYuanzi, 2020-07-23:
You can also use the project I uploaded on github directly.
https://github.com/LittleYuanzi/spconv1.0

ChunmianLin, 2020-07-24:
@Gltina @LittleYuanzi Thanks so much for your helps!

MartinHahner88, 2020-07-25:
https://github.com/traveller59/spconv/issues/152

sarimmehdi, 2020-07-28:
These instructions don't work. I am getting a problem described here: [https://github.com/open-mmlab/OpenPCDet/issues/187#issue-666795655](url)

sshaoshuai, 2020-08-05:
Repeat issue with https://github.com/open-mmlab/OpenPCDet/issues/187.

closed 2020-08-05 by sshaoshuai