open-mmlab/OpenPCDet #180: how to install mayavi

ChunmianLin, 2020-07-24:


Gus-Guo, 2020-07-24:
You can try 'conda install mayavi'  or 'pip install mayavi'.

closed 2020-07-24 by Gus-Guo

ChunmianLin, 2020-07-24:
Thanks. I indeed fail to install mayavi by conda or pip command.
Following the instruction in mayavi official website, I still cannot pass the compilation.
Please help me!

Gus-Guo, 2020-07-24:
what is the error message when you use pip or conda to install ? you can change another channels to install mayavi, e.g. 'conda install mayavi -c conda-forge', where -c refer to a new channel.

ChunmianLin, 2020-07-24:
Thanks.
I cannot install when using pip or conda, the error information is  like '**fail to build mayav**i' or '**Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source**'.
Actually, I have tried to use conda-forge as you recommend, but the error is still but different: '**No module named mayavi.mlab**'
So I confuse about whether mayavi has been installed successfully, and why this error occurs.

So sorry for my trouble, and I would appreciate if any help you provide.
Thanks again.

Gus-Guo, 2020-07-24:
I guess you have to re-install the anaconda  or  change another version of conda, and then have another try.

ChunmianLin, 2020-07-24:
Thanks!
Is it the only option? And is there no alternative method to solve this bug? 
Actually, I don't desire to delete the environment. Currently the model can be retrained except for result visualization, due to no mayavi.

pengcheng001, 2020-07-24:
hello, i use 'conda install mayavi -c conda-forge' install mayavi, but when i run demo.py.I get ERROR:
This application failed to start because it could not find or load the Qt platform plugin "xcb"
in "".

Reinstalling the application may fix this problem.
Aborted (core dumped

pengcheng001, 2020-07-24:
> hello, i use 'conda install mayavi -c conda-forge' install mayavi, but when i run demo.py.I get ERROR:
> This application failed to start because it could not find or load the Qt platform plugin "xcb"
> in "".
> 
> Reinstalling the application may fix this problem.
> Aborted (core dumped

I try to set "export QT_QPA_PLATFORM_PLUGIN_PATH=Data/anaconda3/plugins/platforms/“, then the mayavi.mlab could work.But it will still report a warning:"QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled".Don't care about it, mayavi.mlab can work normally

ChunmianLin, 2020-07-25:
Thanks so much for your advise.So you mean I need to reinstall Qt application? Maybe proper version of Qt can work and do not export maya vi.mlab bug?I am newer to this area, and please understand!Thanks, anyway!发自我的iPhone------------------ Original ------------------From: pengcheng001  <notifications@github.com>Date: Sat,Jul 25,2020 7:12 AMTo: open-mmlab/OpenPCDet <OpenPCDet@noreply.github.com>Cc: ChunmianLin <cmlin@buaa.edu.cn>, Author <author@noreply.github.com>Subject: Re: [open-mmlab/OpenPCDet] how to install mayavi (#180)

hello, i use 'conda install mayavi -c conda-forge' install mayavi, but when i run demo.py.I get ERROR:

This application failed to start because it could not find or load the Qt platform plugin "xcb"

in "".

Reinstalling the application may fix this problem.

Aborted (core dumped

I try to set "export QT_QPA_PLATFORM_PLUGIN_PATH=Data/anaconda3/plugins/platforms/“, then the mayavi.mlab could work.But it will still report a warning:"QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled".Don't care about it, mayavi.mlab can work normally

[

{

"@type": "EmailMessage",

"@type": "ViewAction",

"target": "https://github.com/open-mmlab/OpenPCDet/issues/180#issuecomment-663768913",

"url": "https://github.com/open-mmlab/OpenPCDet/issues/180#issuecomment-663768913",

"name": "View Issue"

},

"description": "View this Issue on GitHub",

"publisher": {

"@type": "Organization",

"name": "GitHub",

"url": "https://github.com"

}

}

]

ChunmianLin, 2020-07-25:
Hi, I have set the env path, but the problem ‘no module named mayavi.mlab’ still exists. 
My mayavi is 4.6.2, Qt is 5.6.2, and vtk is 8.1.0. 

发送自 Windows 10 版邮件应用

发件人: pengcheng001
发送时间: 2020年7月25日 7:13
收件人: open-mmlab/OpenPCDet
抄送: ChunmianLin; Author
主题: Re: [open-mmlab/OpenPCDet] how to install mayavi (#180)

hello, i use 'conda install mayavi -c conda-forge' install mayavi, but when i run demo.py.I get ERROR:
This application failed to start because it could not find or load the Qt platform plugin "xcb"
in "".
Reinstalling the application may fix this problem.
Aborted (core dumped
I try to set "export QT_QPA_PLATFORM_PLUGIN_PATH=Data/anaconda3/plugins/platforms/“, then the mayavi.mlab could work.But it will still report a warning:"QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled".Don't care about it, mayavi.mlab can work normally

pengcheng001, 2020-07-27:
## you need install mayavi
### i guss in china,you can add tsinghua's conda source into you conda envs:
1.  conda config --add channels https://mirrors.tuna.tsinghua.edu.cn/anaconda/cloud/conda-forge/
2.  conda config --add channels https://mirrors.tuna.tsinghua.edu.cn/anaconda/cloud/msys2/
3.  conda config --add channels https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main/
4.  conda config --add channels https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free/
### using the `conda install mayavi` i install mayai
### then set "export QT_QPA_PLATFORM_PLUGIN_PATH=you_conda_path/plugins/platforms/.
### if t it will still report a error. you can check the qt version