open-mmlab/OpenPCDet #181: can it run without GPU?

Shawn-Zhuang, 2020-07-24:
Hi, I have successfully detected on my own data. Thanks for the great open tool!

If I don't need a realtime detection, can I run OpenPCDet without GPU? 
if it is possible, where can I disable the GPU in the code?

since spconv needs cuda support, it would be difficult to run without GPU?

sshaoshuai, 2020-07-24:
No, I didn't consider to run `OpenPCDet` with cpu only mode, so currently it couldnot run on CPU. 
Because there are some operations that are implemented with CUDA on the running of PartA2 / PV-RCNN and the post processing.

Shawn-Zhuang, 2020-07-24:
yeah, it is reasonable to run on GPU.

Shawn-Zhuang, 2020-07-27:
Hi, @sshaoshuai , I have successfully run without "cuda" with PointPillar
At first, I remove the cuda module on my pc, but when I run demo.py, GPU was still used, which I found in "nvidia-smi".
However, when I change
1. `batch_dict[key] = torch.from_numpy(val).float().cuda()` to `batch_dict[key] = torch.from_numpy(val).float().cpu()`
2. ` model.cuda()` to  ` model.cpu()`
3.  in pcdet/utils/box_coder_utils.py   
```
        diagonal = torch.sqrt(dxa ** 2 + dya ** 2)
        xa=xa
        xt=xt
        ......
```

to
``` 
        diagonal = torch.sqrt(dxa ** 2 + dya ** 2).cpu()
        xa=xa.cpu()
        xt=xt.cpu()
        ......
```
4. in OpenPCDet/pcdet/ops/iou3d_nms/iou3d_nms_utils.py
from `boxes = boxes[order].contiguous()` to `boxes = boxes[order].contiguous().cuda()`

Then the GPU is not taken used. The speed is slower from 0.1s per frame to 0.8s per frame.
But my pc has successfully compiled the library with the help of CUDA before I did this.

My question is, when I want to do the experiment on a pc without nvidia-GPU, without cuda installed, I can not compile the OpenPCDet library. Is there any solution to compile when there is no cuda? Maybe some changes in the setup.py?

Shawn-Zhuang, 2020-08-05:
Oh no. When I run
`watch -n 0.1 nvidia-smi`
I found GPU is still 2% taken used.

sshaoshuai, 2020-08-17:
Feel feel to re-open it if there are any updates.

closed 2020-08-17 by sshaoshuai

a43992899, 2021-08-07:
@zhixiongzh Have you ever tried set CUDA_VISIBLE_DEVICES=-1?

zhixiongzh, 2021-08-15:
> @zhixiongzh Have you ever tried set CUDA_VISIBLE_DEVICES=-1?
No, I didn't. But I think CUDA_VISIBLE_DEVICES=-1 means use all available GPUs instead of non GPU. You can give it a try since I am not longer responsible for the project.

han1366, 2022-02-25:
@zhixiongzh how to deal with the iou3d without GPU CUDA?hope your reply,thanks.