open-mmlab/OpenPCDet #183: Will pure point-based methods configs and simple turorials support?

gujiaqivadin, 2020-07-24:
Hello, ssshuai!
    Thanks for your great work of 3D object detection tools. There are four methods, SECOND, PointPillars, Part A2 and PV-RCNN in openpcdet. But they are all voxel-based/-fusion methods to some degree.  Will this repository support PointRCNN, STD, 3DSSD and other point-based methods? Or will you provide us a simple tutorial for establishing/realizing a method including backbone/head/loss, etc.

Thanks a lot.

sshaoshuai, 2020-07-24:
Hi, Point-based methods like PointRCNN will be supported very soon.

gujiaqivadin, 2020-07-25:
Thanks, I am looking forward to it. And I have another problem, the PartA2 config in your repository is anchor-based or anchor-free? I found in your paper, there are two policies to generate proposals.

Majiker, 2020-07-27:
> Thanks, I am looking forward to it. And I have another problem, the PartA2 config in your repository is anchor-based or anchor-free? I found in your paper, there are two policies to generate proposals.

It seems anchor-based as there are 'anchor range' and 'anchor size' specified in the config file.

sshaoshuai, 2020-07-27:
Yes, currently the provided PartA2 config is the anchor-based version.

gujiaqivadin, 2020-07-30:
> Yes, currently the provided PartA2 config is the anchor-based version.

Thanks, and can we wait for a anchor-free version?

sshaoshuai, 2020-07-30:
Hi, both PointRCNN and PartA2-Free have been released just now.

closed 2020-07-30 by sshaoshuai