open-mmlab/OpenPCDet #185: TypeError: zip argument #1 must support iteration

zzqjh, 2020-07-26:
Hello, when I execute this line of code：
python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos tools/cfgs/dataset_configs/kitti_dataset.yaml

I encountered this report error：
TypeError: zip argument #1 must support iteration.

the details is follows:
/home/zzqjh/.conda/envs/pytorch_gpu/lib/python3.6/runpy.py:125: RuntimeWarning: 'pcdet.datasets.kitti.kitti_dataset' found in sys.modules after import of package 'pcdet.datasets.kitti', but prior to execution of 'pcdet.datasets.kitti.kitti_dataset'; this may result in unpredictable behaviour
  warn(RuntimeWarning(msg))
/home/zzqjh/3D-OT/OpenPCDet/PCDet/pcdet/datasets/kitti/kitti_dataset.py:430: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  dataset_cfg = EasyDict(yaml.load(open(sys.argv[2])))
---------------Start to generate data infos---------------
Traceback (most recent call last):
  File "/home/zzqjh/.conda/envs/pytorch_gpu/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/zzqjh/.conda/envs/pytorch_gpu/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/zzqjh/3D-OT/OpenPCDet/PCDet/pcdet/datasets/kitti/kitti_dataset.py", line 436, in <module>
    save_path=ROOT_DIR / 'data' / 'kitti'
  File "/home/zzqjh/3D-OT/OpenPCDet/PCDet/pcdet/datasets/kitti/kitti_dataset.py", line 396, in create_kitti_infos
    kitti_infos_train = dataset.get_infos(num_workers=workers, has_label=True, count_inside_pts=True)
  File "/home/zzqjh/3D-OT/OpenPCDet/PCDet/pcdet/datasets/kitti/kitti_dataset.py", line 188, in get_infos
    infos = executor.map(process_single_scene, sample_id_list)
  File "/home/zzqjh/.conda/envs/pytorch_gpu/lib/python3.6/concurrent/futures/_base.py", line 548, in map
    fs = [self.submit(fn, *args) for args in zip(*iterables)]
TypeError: zip argument #1 must support iteration

zzqjh, 2020-07-26:
my torch version is 1.4

sshaoshuai, 2020-07-27:
You could try to use PyTorch 1.1 since I only tested it with PyTorch 1.1.

zzqjh, 2020-07-31:
> You could try to use PyTorch 1.1 since I only tested it with PyTorch 1.1.

thank you for your reply.And I tried to use pytorch 1.1 and torchvision 0.3 on cuda 9.0 and cudnn 7.6.5 and python 3.6,but there was a new error:

(pytorch_gpu) zzqjh@zzqjh-X8Ti-G-Series-GK7CP6R:~/3D-OT/OpenPCDet/PCDet$ python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos tools/cfgs/dataset_configs/kitti_dataset.yaml
Traceback (most recent call last):
  File "/home/zzqjh/.conda/envs/pytorch_gpu/lib/python3.6/runpy.py", line 183, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/home/zzqjh/.conda/envs/pytorch_gpu/lib/python3.6/runpy.py", line 109, in _get_module_details
    __import__(pkg_name)
  File "/home/zzqjh/3D-OT/OpenPCDet/PCDet/pcdet/datasets/__init__.py", line 3, in <module>
    from .dataset import DatasetTemplate
  File "/home/zzqjh/3D-OT/OpenPCDet/PCDet/pcdet/datasets/dataset.py", line 5, in <module>
    from .augmentor.data_augmentor import DataAugmentor
  File "/home/zzqjh/3D-OT/OpenPCDet/PCDet/pcdet/datasets/augmentor/data_augmentor.py", line 3, in <module>
    from . import augmentor_utils, database_sampler
  File "/home/zzqjh/3D-OT/OpenPCDet/PCDet/pcdet/datasets/augmentor/database_sampler.py", line 3, in <module>
    from ...utils import box_utils
  File "/home/zzqjh/3D-OT/OpenPCDet/PCDet/pcdet/utils/box_utils.py", line 4, in <module>
    from ..ops.roiaware_pool3d import roiaware_pool3d_utils
  File "/home/zzqjh/3D-OT/OpenPCDet/PCDet/pcdet/ops/roiaware_pool3d/roiaware_pool3d_utils.py", line 5, in <module>
    from . import roiaware_pool3d_cuda
ImportError: /home/zzqjh/3D-OT/OpenPCDet/PCDet/pcdet/ops/roiaware_pool3d/roiaware_pool3d_cuda.cpython-36m-x86_64-linux-gnu.so: undefined symbol: _ZN6caffe28TypeMeta21_typeMetaDataInstanceIN3c108BFloat16EEEPKNS_6detail12TypeMetaDataEv

sshaoshuai, 2020-08-05:
I suggest you to re-install the pytorch 1.1 and re-build the spconv (remember to delete the `spconv/build` directory before you re-build spconv).

closed 2020-08-17 by sshaoshuai

McDifference, 2020-09-11:
> > You could try to use PyTorch 1.1 since I only tested it with PyTorch 1.1.
> 
> thank you for your reply.And I tried to use pytorch 1.1 and torchvision 0.3 on cuda 9.0 and cudnn 7.6.5 and python 3.6,but there was a new error:
> 
> (pytorch_gpu) zzqjh@zzqjh-X8Ti-G-Series-GK7CP6R:~/3D-OT/OpenPCDet/PCDet$ python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos tools/cfgs/dataset_configs/kitti_dataset.yaml
> Traceback (most recent call last):
> File "/home/zzqjh/.conda/envs/pytorch_gpu/lib/python3.6/runpy.py", line 183, in _run_module_as_main
> mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
> File "/home/zzqjh/.conda/envs/pytorch_gpu/lib/python3.6/runpy.py", line 109, in _get_module_details
> **import**(pkg_name)
> File "/home/zzqjh/3D-OT/OpenPCDet/PCDet/pcdet/datasets/**init**.py", line 3, in
> from .dataset import DatasetTemplate
> File "/home/zzqjh/3D-OT/OpenPCDet/PCDet/pcdet/datasets/dataset.py", line 5, in
> from .augmentor.data_augmentor import DataAugmentor
> File "/home/zzqjh/3D-OT/OpenPCDet/PCDet/pcdet/datasets/augmentor/data_augmentor.py", line 3, in
> from . import augmentor_utils, database_sampler
> File "/home/zzqjh/3D-OT/OpenPCDet/PCDet/pcdet/datasets/augmentor/database_sampler.py", line 3, in
> from ...utils import box_utils
> File "/home/zzqjh/3D-OT/OpenPCDet/PCDet/pcdet/utils/box_utils.py", line 4, in
> from ..ops.roiaware_pool3d import roiaware_pool3d_utils
> File "/home/zzqjh/3D-OT/OpenPCDet/PCDet/pcdet/ops/roiaware_pool3d/roiaware_pool3d_utils.py", line 5, in
> from . import roiaware_pool3d_cuda
> ImportError: /home/zzqjh/3D-OT/OpenPCDet/PCDet/pcdet/ops/roiaware_pool3d/roiaware_pool3d_cuda.cpython-36m-x86_64-linux-gnu.so: undefined symbol: _ZN6caffe28TypeMeta21_typeMetaDataInstanceIN3c108BFloat16EEEPKNS_6detail12TypeMetaDataEv

Have you solved the problem? I have got the same error.
CUDA10.1, PyTorch1.5

McDifference, 2020-09-12:
> Have you solved the problem? I have got the same error.
> CUDA10.1, PyTorch1.5

I turned out it was a path error. The path was None. Check the path carefully can avoid this error. Note that there should be a ImageSets folder within the KITTI dataset. Please organise the files as the author suggests.
```
OpenPCDet
├── data
│   ├── kitti
│   │   │── ImageSets
│   │   │── training
│   │   │   ├──calib & velodyne & label_2 & image_2 & (optional: planes)
│   │   │── testing
│   │   │   ├──calib & velodyne & image_2
├── pcdet
├── tools
```

cuge1995, 2020-10-17:
This problem is solved by copy `ImageSets/` to the Real kitti folder

bhyzhao, 2021-03-15:
Be careful of Path. The 'training/' and 'testing/' folders are not in 'ImageSets/' but directly in 'kitti/'.

SibylGao, 2022-09-08:
omg... it works. thanks a lot