open-mmlab/OpenPCDet #187: Unable to install spconv

sarimmehdi, 2020-07-28:
Hello. I am unable to spconv1.0. I followed the instructions here [https://github.com/traveller59/spconv/issues/152#issue-631745341](url). This is what I did: 

- Opened venv in PyCharm

- git clone https://github.com/traveller59/spconv spconv_8da6f96 --recursive

- cd spconv_8da6f96

- git checkout 8da6f967fb9a054d8870c3515b1b44eca2103634

- DId as described here: [https://github.com/traveller59/spconv/issues/78#issuecomment-526223281](url)

- CUDA_ROOT=home/sarim/PycharmProjects/trajectory_prediction/spconv_8da6f96 python setup.py bdist_wheel

Following is the output I get:

```
running bdist_wheel
running build
running build_py
running build_ext
/home/sarim/PycharmProjects/trajectory_prediction/spconv_8da6f96/build/lib.linux-x86_64-3.7
Release
-- Caffe2: CUDA detected: 10.0
-- Caffe2: CUDA nvcc is: /usr/local/cuda-10.0/bin/nvcc
-- Caffe2: CUDA toolkit directory: /usr/local/cuda-10.0
-- Caffe2: Header version is: 10.0
-- Found cuDNN: v7.4.2  (include: /usr/local/cuda-10.0/include, library: /usr/local/cuda-10.0/lib64/libcudnn.so)
-- Autodetected CUDA architecture(s):  6.1
-- Added CUDA NVCC flags for: -gencode;arch=compute_61,code=sm_61
CMake Warning (dev) at /snap/cmake/487/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:273 (message):
  The package name passed to `find_package_handle_standard_args` (torch) does
  not match the name of the calling package (Torch).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /home/sarim/PycharmProjects/trajectory_prediction/venv/lib/python3.7/site-packages/torch/share/cmake/Torch/TorchConfig.cmake:122 (find_package_handle_standard_args)
  CMakeLists.txt:23 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- pybind11 v2.5.0
-- Configuring done
CMake Warning (dev) in src/spconv/CMakeLists.txt:
  Policy CMP0104 is not set: CMAKE_CUDA_ARCHITECTURES now detected for NVCC,
  empty CUDA_ARCHITECTURES not allowed.  Run "cmake --help-policy CMP0104"
  for policy details.  Use the cmake_policy command to set the policy and
  suppress this warning.

  CUDA_ARCHITECTURES is empty for target "spconv".
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) in src/spconv/CMakeLists.txt:
  Policy CMP0104 is not set: CMAKE_CUDA_ARCHITECTURES now detected for NVCC,
  empty CUDA_ARCHITECTURES not allowed.  Run "cmake --help-policy CMP0104"
  for policy details.  Use the cmake_policy command to set the policy and
  suppress this warning.

  CUDA_ARCHITECTURES is empty for target "spconv".
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) in src/utils/CMakeLists.txt:
  Policy CMP0104 is not set: CMAKE_CUDA_ARCHITECTURES now detected for NVCC,
  empty CUDA_ARCHITECTURES not allowed.  Run "cmake --help-policy CMP0104"
  for policy details.  Use the cmake_policy command to set the policy and
  suppress this warning.

  CUDA_ARCHITECTURES is empty for target "spconv_nms".
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) in src/utils/CMakeLists.txt:
[... 415 lines cut ...]
  File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/sarim/PycharmProjects/trajectory_prediction/venv/lib/python3.7/site-packages/wheel/bdist_wheel.py", line 223, in run
    self.run_command('build')
  File "/usr/lib/python3.7/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/usr/lib/python3.7/distutils/command/build.py", line 135, in run
    self.run_command(cmd_name)
  File "/usr/lib/python3.7/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "setup.py", line 39, in run
    self.build_extension(ext)
  File "setup.py", line 81, in build_extension
    subprocess.check_call(['cmake', '--build', '.'] + build_args, cwd=self.build_temp)
  File "/usr/lib/python3.7/subprocess.py", line 363, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', '.', '--config', 'Release', '--', '-j4']' returned non-zero exit status 2.

```
My boost version is 1.73 and cuda is 10.0 and my make is 3.18.0
Please tell me how to fix this. Thanks

MartinHahner88, 2020-07-28:
> CMake Warning (dev) in src/spconv/CMakeLists.txt:
  Policy CMP0104 is not set: CMAKE_CUDA_ARCHITECTURES now detected for NVCC,
  empty CUDA_ARCHITECTURES not allowed.  Run "cmake --help-policy CMP0104"
  for policy details.  Use the cmake_policy command to set the policy and
  suppress this warning.

I don't know if CUDA 10.**0** is supported, I used CUDA 9.0 to successfully build `spconv`.
I would try to google solutions for this warning.

> CMake Warning (dev) at /snap/cmake/487/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:273 (message):
  The package name passed to `find_package_handle_standard_args` (**torch**) does
  not match the name of the calling package (**Torch**).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.

This warning also is odd. Which version of _torch_ are you using? Maybe your version is too new.

sarimmehdi, 2020-07-28:
I am using torch 1.4.0. Which version did you use?

sarimmehdi, 2020-07-28:
I downgraded my pytorch to version 1.1.0 and now I get the following error:

```
running bdist_wheel
running build
running build_py
running build_ext
/home/sarim/PycharmProjects/trajectory_prediction/spconv_8da6f96/build/lib.linux-x86_64-3.7
Release
-- Caffe2: CUDA detected: 10.0
-- Caffe2: CUDA nvcc is: /usr/local/cuda-10.0/bin/nvcc
-- Caffe2: CUDA toolkit directory: /usr/local/cuda-10.0
-- Caffe2: Header version is: 10.0
CMake Warning (dev) at /snap/cmake/487/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:273 (message):
  The package name passed to `find_package_handle_standard_args` (CUDNN) does
  not match the name of the calling package (Caffe2).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /home/sarim/PycharmProjects/trajectory_prediction/venv/lib/python3.7/site-packages/torch/share/cmake/Caffe2/public/cuda.cmake:107 (find_package_handle_standard_args)
  /home/sarim/PycharmProjects/trajectory_prediction/venv/lib/python3.7/site-packages/torch/share/cmake/Caffe2/Caffe2Config.cmake:88 (include)
  /home/sarim/PycharmProjects/trajectory_prediction/venv/lib/python3.7/site-packages/torch/share/cmake/Torch/TorchConfig.cmake:40 (find_package)
  CMakeLists.txt:23 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Could NOT find CUDNN (missing: CUDNN_INCLUDE_DIR CUDNN_LIBRARY) 
CMake Warning at /home/sarim/PycharmProjects/trajectory_prediction/venv/lib/python3.7/site-packages/torch/share/cmake/Caffe2/public/cuda.cmake:110 (message):
  Caffe2: Cannot find cuDNN library.  Turning the option off
Call Stack (most recent call first):
  /home/sarim/PycharmProjects/trajectory_prediction/venv/lib/python3.7/site-packages/torch/share/cmake/Caffe2/Caffe2Config.cmake:88 (include)
  /home/sarim/PycharmProjects/trajectory_prediction/venv/lib/python3.7/site-packages/torch/share/cmake/Torch/TorchConfig.cmake:40 (find_package)
  CMakeLists.txt:23 (find_package)

-- Autodetected CUDA architecture(s):  6.1
-- Added CUDA NVCC flags for: -gencode;arch=compute_61,code=sm_61
CMake Error at /home/sarim/PycharmProjects/trajectory_prediction/venv/lib/python3.7/site-packages/torch/share/cmake/Caffe2/Caffe2Config.cmake:96 (message):
  Your installed Caffe2 version uses cuDNN but I cannot find the cuDNN
  libraries.  Please set the proper cuDNN prefixes and / or install cuDNN.
Call Stack (most recent call first):
  /home/sarim/PycharmProjects/trajectory_prediction/venv/lib/python3.7/site-packages/torch/share/cmake/Torch/TorchConfig.cmake:40 (find_package)
  CMakeLists.txt:23 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/sarim/PycharmProjects/trajectory_prediction/spconv_8da6f96/build/temp.linux-x86_64-3.7/CMakeFiles/CMakeOutput.log".
See also "/home/sarim/PycharmProjects/trajectory_prediction/spconv_8da6f96/build/temp.linux-x86_64-3.7/CMakeFiles/CMakeError.log".
Traceback (most recent call last):
  File "setup.py", line 97, in <module>
    zip_safe=False,
  File "/home/sarim/PycharmProjects/trajectory_prediction/venv/lib/python3.7/site-packages/setuptools/__init__.py", line 165, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib/python3.7/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/usr/lib/python3.7/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/sarim/PycharmProjects/trajectory_prediction/venv/lib/python3.7/site-packages/wheel/bdist_wheel.py", line 223, in run
    self.run_command('build')
  File "/usr/lib/python3.7/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/usr/lib/python3.7/distutils/command/build.py", line 135, in run
    self.run_command(cmd_name)
  File "/usr/lib/python3.7/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "setup.py", line 39, in run
    self.build_extension(ext)
  File "setup.py", line 69, in build_extension
    subprocess.check_call(['cmake', ext.sourcedir] + cmake_args, cwd=self.build_temp, env=env)
  File "/usr/lib/python3.7/subprocess.py", line 363, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '/home/sarim/PycharmProjects/trajectory_prediction/spconv_8da6f96', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/home/sarim/PycharmProjects/trajectory_prediction/spconv_8da6f96/build/lib.linux-x86_64-3.7/spconv', '-DCMAKE_PREFIX_PATH=/home/sarim/PycharmProjects/trajectory_prediction/venv/lib/python3.7/site-packages/torch', '-DPYBIND11_PYTHON_VERSION=3.7', '-DSPCONV_BuildTests=OFF', '-DCMAKE_CUDA_FLAGS="--expt-relaxed-constexpr"', '-DCMAKE_BUILD_TYPE=Release']' returned non-zero exit status 1.
```
It says it couldn't find cudnn. I verified the install location of my cudnn by typing:
` cat /usr/local/cuda-10.0/include/cudnn.h | grep CUDNN_MAJOR -A 2`
And I get the following output:
```
#define CUDNN_MAJOR 7
#define CUDNN_MINOR 4
#define CUDNN_PATCHLEVEL 2
--
#define CUDNN_VERSION (CUDNN_MAJOR * 1000 + CUDNN_MINOR * 100 + CUDNN_PATCHLEVEL)

#include "driver_types.h"
```
Why is spconv unable to find cudnn?

MartinHahner88, 2020-07-28:
Are you using a conda environment? 

If yes, did you run `conda install cudnn` as described in https://github.com/traveller59/spconv/issues/152#issue-631745341 ?
If no, you should consider using conda.

sarimmehdi, 2020-07-28:
Why wouldn't this run in venv? My PyCharm project uses venv.

sshaoshuai, 2020-07-31:
I have tested with CUDA 9.0, CUDA 10.2, both of them are ok. 
Please also note that in the [INSTALL.md](https://github.com/open-mmlab/OpenPCDet/blob/master/docs/INSTALL.md): 
* If you use PyTorch 1.1, then make sure you install the spconv v1.0 with (commit 8da6f96) instead of the latest one.
* If you use PyTorch 1.3+, then you need to install the spconv v1.2. As mentioned by the author of spconv, you need to use their docker if you use PyTorch 1.4+.

sshaoshuai, 2020-08-05:
Feel free to re-open it if you still have problems.

closed 2020-08-05 by sshaoshuai

nixondutt, 2021-03-30:
> I downgraded my pytorch to version 1.1.0 and now I get the following error:
>
> ```
> running bdist_wheel
> running build
> running build_py
> running build_ext [...]

Could you fix this error. I have followed all the mentioned procedure and found this same error. Unable to fix it since morning. Please help.

MartinHahner, 2021-04-09:
The script says `-- Autodetected CUDA architecture(s):  6.1` 
which does not correspond to your output of `cat /usr/local/cuda-10.0/include/cudnn.h | grep CUDNN_MAJOR -A 2`

I don't know how to fix it though.