open-mmlab/OpenPCDet #189: precision of PV-RCNN

LittleYuanzi, 2020-07-28:
Hello, the precision of PV-RCNN is very high. 
But when I am training PVRCNN, although I have only trained 50 epoch, the epoch45,epoch47,epoch49 accuracy has reached 83.2 (moderate car), but the accuracy of epoch 46 and epoch 50 is only 79. 
Can you explain it to me? 
Is it because of circle LR? 
Another problem that bothers me is that the accuracy of the same epoch (moderate car) is sometimes 82% and sometimes 79%, which seems to have a lot to do with the sampling points.

MartinHahner88, 2020-07-28:
The fluctuations you described are quite normal.
Here is an evaluation of each epoch for a full training run of 80 epochs.

[image]

LittleYuanzi, 2020-07-29:
> The fluctuations you described are quite normal.
> Here is an evaluation of each epoch for a full training run of 80 epochs.
> 
> [image]

Thank you very much for your reply. 
But I'm still a little confused as to why there is such a big AP deviation when the same epoch tests the val dataset many times.

[image]

closed 2020-07-29 by LittleYuanzi

3bobo, 2020-09-23:
@MartinHahner88 By the way, how to open the saved tensorbod files ? I've tried to use `tensorboard --logdir=/path/to/tensorboard/`, but it shows no data. Thanks!

MartinHahner88, 2020-09-23:
> @MartinHahner88 By the way, how to show these predicted results? such as `Car AP_R40` data. Thanks!

@3bobo by evaluating all the epochs after training, not just the last 10 or so.

3bobo, 2020-09-23:
@MartinHahner88 I got it, thank you very much!