open-mmlab/OpenPCDet #190: index 3 is out of bounds for axis 0 with size 3

sarimmehdi, 2020-07-29:
Hello, I am getting the above error. I run the following command:

`--cfg_file /home/sarim/PycharmProjects/obj3d/OpenPCDet/tools/cfgs/kitti_models/pv_rcnn.yaml --ckpt /home/sarim/PycharmProjects/obj3d/OpenPCDet/pv_rcnn_8369.pth`

Full error log:

```
2020-07-29 23:29:10,241   INFO  **********************Start logging**********************
2020-07-29 23:29:10,241   INFO  CUDA_VISIBLE_DEVICES=ALL
2020-07-29 23:29:10,241   INFO  cfg_file         /home/sarim/PycharmProjects/obj3d/OpenPCDet/tools/cfgs/kitti_models/pv_rcnn.yaml
2020-07-29 23:29:10,241   INFO  batch_size       16
2020-07-29 23:29:10,241   INFO  epochs           80
2020-07-29 23:29:10,242   INFO  workers          4
2020-07-29 23:29:10,242   INFO  extra_tag        default
2020-07-29 23:29:10,242   INFO  ckpt             /home/sarim/PycharmProjects/obj3d/OpenPCDet/pv_rcnn_8369.pth
2020-07-29 23:29:10,242   INFO  mgpus            False
2020-07-29 23:29:10,242   INFO  launcher         none
2020-07-29 23:29:10,242   INFO  tcp_port         18888
2020-07-29 23:29:10,242   INFO  local_rank       0
2020-07-29 23:29:10,242   INFO  set_cfgs         None
2020-07-29 23:29:10,242   INFO  max_waiting_mins 30
2020-07-29 23:29:10,242   INFO  start_epoch      0
2020-07-29 23:29:10,242   INFO  eval_tag         default
2020-07-29 23:29:10,242   INFO  eval_all         False
2020-07-29 23:29:10,242   INFO  ckpt_dir         None
2020-07-29 23:29:10,242   INFO  save_to_file     False
2020-07-29 23:29:10,242   INFO  cfg.ROOT_DIR: /home/sarim/PycharmProjects/obj3d/OpenPCDet
2020-07-29 23:29:10,242   INFO  cfg.LOCAL_RANK: 0
2020-07-29 23:29:10,242   INFO  cfg.CLASS_NAMES: ['Car', 'Pedestrian', 'Cyclist']
2020-07-29 23:29:10,242   INFO  
cfg.DATA_CONFIG = edict()
2020-07-29 23:29:10,242   INFO  cfg.DATA_CONFIG.DATASET: KittiDataset
2020-07-29 23:29:10,242   INFO  cfg.DATA_CONFIG.DATA_PATH: ../data/kitti
2020-07-29 23:29:10,242   INFO  cfg.DATA_CONFIG.POINT_CLOUD_RANGE: [0, -40, -3, 70.4, 40, 1]
2020-07-29 23:29:10,242   INFO  
cfg.DATA_CONFIG.DATA_SPLIT = edict()
2020-07-29 23:29:10,242   INFO  cfg.DATA_CONFIG.DATA_SPLIT.train: train
2020-07-29 23:29:10,242   INFO  cfg.DATA_CONFIG.DATA_SPLIT.test: val
2020-07-29 23:29:10,242   INFO  
cfg.DATA_CONFIG.INFO_PATH = edict()
2020-07-29 23:29:10,242   INFO  cfg.DATA_CONFIG.INFO_PATH.train: ['kitti_infos_train.pkl']
2020-07-29 23:29:10,242   INFO  cfg.DATA_CONFIG.INFO_PATH.test: ['kitti_infos_val.pkl']
2020-07-29 23:29:10,243   INFO  cfg.DATA_CONFIG.FOV_POINTS_ONLY: True
2020-07-29 23:29:10,243   INFO  
cfg.DATA_CONFIG.DATA_AUGMENTOR = edict()
2020-07-29 23:29:10,243   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.DISABLE_AUG_LIST: ['placeholder']
2020-07-29 23:29:10,243   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.AUG_CONFIG_LIST: [{'NAME': 'gt_sampling', 'USE_ROAD_PLANE': True, 'DB_INFO_PATH': ['kitti_dbinfos_train.pkl'], 'PREPARE': {'filter_by_min_points': ['Car:5', 'Pedestrian:5', 'Cyclist:5'], 'filter_by_difficulty': [-1]}, 'SAMPLE_GROUPS': ['Car:15', 'Pedestrian:10', 'Cyclist:10'], 'NUM_POINT_FEATURES': 4, 'DATABASE_WITH_FAKELIDAR': False, 'REMOVE_EXTRA_WIDTH': [0.0, 0.0, 0.0], 'LIMIT_WHOLE_SCENE': False}, {'NAME': 'random_world_flip', 'ALONG_AXIS_LIST': ['x']}, {'NAME': 'random_world_rotation', 'WORLD_ROT_ANGLE': [-0.78539816, 0.78539816]}, {'NAME': 'random_world_scaling', 'WORLD_SCALE_RANGE': [0.95, 1.05]}]
2020-07-29 23:29:10,243   INFO  
cfg.DATA_CONFIG.POINT_FEATURE_ENCODING = edict()
2020-07-29 23:29:10,243   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.encoding_type: absolute_coordinates_encoding
2020-07-29 23:29:10,243   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.used_feature_list: ['x', 'y', 'z', 'intensity']
2020-07-29 23:29:10,243   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.src_feature_list: ['x', 'y', 'z', 'intensity']
2020-07-29 23:29:10,243   INFO  cfg.DATA_CONFIG.DATA_PROCESSOR: [{'NAME': 'mask_points_and_boxes_outside_range', 'REMOVE_OUTSIDE_BOXES': True}, {'NAME': 'shuffle_points', 'SHUFFLE_ENABLED': {'train': True, 'test': False}}, {'NAME': 'transform_points_to_voxels', 'VOXEL_SIZE': [0.05, 0.05, 0.1], 'MAX_POINTS_PER_VOXEL': 5, 'MAX_NUMBER_OF_VOXELS': {'train': 16000, 'test': 40000}}]
2020-07-29 23:29:10,243   INFO  cfg.DATA_CONFIG._BASE_CONFIG_: cfgs/dataset_configs/kitti_dataset.yaml
2020-07-29 23:29:10,243   INFO  
cfg.MODEL = edict()
2020-07-29 23:29:10,243   INFO  cfg.MODEL.NAME: PVRCNN
2020-07-29 23:29:10,243   INFO  
cfg.MODEL.VFE = edict()
2020-07-29 23:29:10,243   INFO  cfg.MODEL.VFE.NAME: MeanVFE
2020-07-29 23:29:10,243   INFO  
cfg.MODEL.BACKBONE_3D = edict()
2020-07-29 23:29:10,243   INFO  cfg.MODEL.BACKBONE_3D.NAME: VoxelBackBone8x
2020-07-29 23:29:10,243   INFO  
cfg.MODEL.MAP_TO_BEV = edict()
2020-07-29 23:29:10,243   INFO  cfg.MODEL.MAP_TO_BEV.NAME: HeightCompression
2020-07-29 23:29:10,243   INFO  cfg.MODEL.MAP_TO_BEV.NUM_BEV_FEATURES: 256
2020-07-29 23:29:10,243   INFO  
cfg.MODEL.BACKBONE_2D = edict()
2020-07-29 23:29:10,243   INFO  cfg.MODEL.BACKBONE_2D.NAME: BaseBEVBackbone
[... 337 lines cut ...]
 [  0.906234   1.689736  19.257484]
 [  0.310684   1.681272   7.931642]
 [  9.179073   1.638459  24.982485]]
[-1.       -1.       -1.208547 -1.787366  2.458248]
[-1. -1. -1. -1. -1.]
[1 0 2 0 1]
[ 0  1  2 -1 -1]
[-1.       -2.16077  -1.812559  2.001759]
[-1. -1. -1. -1.]
[0 0 0 1]
[ 0  1  2 -1]
[[-7.39839077e-01  9.98678303e+00 -4.98977203e+02 -1.00000000e+03
  -1.00000000e+03 -1.00000000e+03 -5.70796327e-01]
 [ 1.62236710e+01  5.99858284e+00 -4.96181130e-01  4.43388600e+00
   1.82325500e+00  2.00000000e+00  5.89973673e-01]
 [ 6.70296526e+00 -1.88539684e+00 -8.57408881e-01  1.78524100e+00
   8.24591000e-01  1.73906300e+00  2.41762673e-01]]
[[-7.39839077e-01  9.98678303e+00 -4.98977203e+02 -1.00000000e+03
  -1.00000000e+03 -1.00000000e+03 -5.70796327e-01]
 [-7.39839077e-01  9.98678303e+00 -4.98977203e+02 -1.00000000e+03
  -1.00000000e+03 -1.00000000e+03 -5.70796327e-01]
 [ 1.95472069e+01 -8.87905121e-01 -5.70142388e-01  4.43388600e+00
   1.82325500e+00  2.00000000e+00 -3.62249327e-01]]
```
Can anyone explain to me what is wrong here? Thanks!

sshaoshuai, 2020-07-30:
The print out infos are messy, could you try to test with a single GPU and try to print(key, info[key])?

sshaoshuai, 2020-08-10:
Close it since no response.

closed 2020-08-10 by sshaoshuai

JiangZongKang, 2020-08-12:
Have you solved this problem?

hcyoo93, 2020-08-15:
I encountered the same problem, and I found out that the "DontCare" class should be placed last.

in kitti_dataset.py, you can see below 
`loc = annotations['location'][:num_objects]
                dims = annotations['dimensions'][:num_objects]
                rots = annotations['rotation_y'][:num_objects]
                loc_lidar = calib.rect_to_lidar(loc)
                l, h, w = dims[:, 0:1], dims[:, 1:2], dims[:, 2:3]
                loc_lidar[:, 2] += h[:, 0] / 2
                gt_boxes_lidar = np.concatenate([loc_lidar, l, w, h, -(np.pi / 2 + rots[..., np.newaxis])], axis=1)
                annotations['gt_boxes_lidar'] = gt_boxes_lidar`

If you have "DontCare" class, num_object is different from num_gt and the class should be the last. (num_objects: # of label excluding "DontCare", num_gt: # of label)

Otherwise, it makes the range of 'keepindices' bigger than the length of gt_boxes_lidar which results in Indexerror.

yustaub, 2020-10-22:
have you solve this problem? I also want to use kitti tracking data ,and I change the format of them to detection format, and encounter the same problem, if you have solved this problem, can you share it? thanks

JiangZongKang, 2020-10-22:
> have you solve this problem? I also want to use kitti tracking data ,and I change the format of them to detection format, and encounter the same problem, if you have solved this problem, can you share it? thanks

yes, DontCare should be the last

yustaub, 2020-10-22:
thanks， I got it. @JiangZongKang