open-mmlab/OpenPCDet #191: ModuleNotFoundError: No module named '_lzma'

OSU-Haolin, 2020-07-29:
pytorch 1.1
python 3.7.3
cuda10.0
spconv was installed successfully

when doing Dataset Preparation and run **python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos tools/cfgs/dataset_configs/kitti_dataset.yaml**, I met errors as follow:

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/runpy.py", line 183, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/usr/local/lib/python3.7/runpy.py", line 109, in _get_module_details
    __import__(pkg_name)
  File "/home/haolin/CITR/OpenPCDet/pcdet/datasets/__init__.py", line 4, in <module>
    from .kitti.kitti_dataset import KittiDataset
  File "/home/haolin/CITR/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 4, in <module>
    from skimage import io
  File "/usr/local/lib/python3.7/site-packages/skimage/io/__init__.py", line 7, in <module>
    from .manage_plugins import *
  File "/usr/local/lib/python3.7/site-packages/skimage/io/manage_plugins.py", line 23, in <module>
    from .collection import imread_collection_wrapper
  File "/usr/local/lib/python3.7/site-packages/skimage/io/collection.py", line 13, in <module>
    from tifffile import TiffFile
  File "/usr/local/lib/python3.7/site-packages/tifffile/__init__.py", line 3, in <module>
    from .tifffile import __doc__, __all__, __version__, main
  File "/usr/local/lib/python3.7/site-packages/tifffile/tifffile.py", line 12097, in <module>
    import lzma
  File "/usr/local/lib/python3.7/lzma.py", line 27, in <module>
    from _lzma import *

**ModuleNotFoundError: No module named '_lzma'**

I installed two versions of Python 3.7 and all failed in this setp with same error message.

Could someone help me ?

OSU-Haolin, 2020-07-30:
sovled

closed 2020-07-30 by OSU-Haolin

Petros626, 2024-10-11:
just install the package. when it isn't