open-mmlab/OpenPCDet #194: Fail to convert test data to .pickle

OSU-Haolin, 2020-07-30:
pytorch 1.1
python 3.7.3
cuda10.0
Spconv was installed successfully

use openpcdet 0.2

when I converted data to .pickle, train and val was processed successfully.
But for test ( I only use some specific frames as shown below) it failed: 

test sample_idx: 000124
test sample_idx: 000308
test sample_idx: 000499
test sample_idx: 001019
test sample_idx: 001466
test sample_idx: 002245
test sample_idx: 002308
test sample_idx: 002867
test sample_idx: 003033
test sample_idx: 003071
test sample_idx: 003156
test sample_idx: 003183
test sample_idx: 003686
test sample_idx: 003775
test sample_idx: 004414
test sample_idx: 004567
test sample_idx: 004931
test sample_idx: 004944
test sample_idx: 004960
test sample_idx: 005110
test sample_idx: 005168
test sample_idx: 005473
test sample_idx: 005699
test sample_idx: 006118
test sample_idx: 006148
test sample_idx: 006433
test sample_idx: 006744
test sample_idx: 007052
test sample_idx: 007115
test sample_idx: 007122
test sample_idx: 007255
test sample_idx: 007347
test sample_idx: 
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/haolin/CITR/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 436, in <module>
    save_path=ROOT_DIR / 'data' / 'kitti'
  File "/home/haolin/CITR/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 412, in create_kitti_infos
    kitti_infos_test = dataset.get_infos(num_workers=workers, has_label=False, count_inside_pts=False)
  File "/home/haolin/CITR/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 189, in get_infos
    return list(infos)
  File "/usr/lib/python3.7/concurrent/futures/_base.py", line 598, in result_iterator
    yield fs.pop().result()
  File "/usr/lib/python3.7/concurrent/futures/_base.py", line 428, in result
    return self.__get_result()
  File "/usr/lib/python3.7/concurrent/futures/_base.py", line 384, in __get_result
    raise self._exception
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/haolin/CITR/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 126, in process_single_scene
    image_info = {'image_idx': sample_idx, 'image_shape': self.get_image_shape(sample_idx)}
  File "/home/haolin/CITR/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 67, in get_image_shape
    assert img_file.exists()
AssertionError

What is the problem and how to fix it ? Could someone help me with it ?

OSU-Haolin, 2020-07-30:
solved

closed 2020-07-30 by OSU-Haolin

josy43, 2020-08-17:
I encountered the same problem, whats the solution ?

OSU-Haolin, 2020-08-18:
> I encountered the same problem, whats the solution ?

Check your data split txt file in detail. Do not leave any space in the txt files.

josy43, 2020-08-18:
Yes that was the problem!

mJieg, 2022-04-03:
> solved

How did you solve it?

OSU-Haolin, 2022-04-16:
> > solved
> 
> How did you solve it?

Check your data split txt file in detail. Do not leave any space in the txt files.