open-mmlab/OpenPCDet #195: How to run demo.py with the nuscense model

Xiangzhaohong, 2020-07-31:
@sshaoshuai  Hi,  i am very grateful for your new update.
I want to test nuscense model with the demo.py,  but there are some errors.
how to solve this?
Thanks!

sshaoshuai, 2020-07-31:
So what's your errors? 

I didn't test it with nuscenes, but you could consider nuscenes as your custom data and follow the https://github.com/open-mmlab/OpenPCDet/blob/master/docs/DEMO.md to transform the data and visualize it.

Xiangzhaohong, 2020-07-31:
@sshaoshuai sorry, this is my error

2020-07-31 17:55:08,155   INFO  -----------------Quick Demo of OpenPCDet-------------------------
2020-07-31 17:55:08,156   INFO  Total number of samples: 	1
2020-07-31 17:55:09,726   INFO  ==> Loading parameters from checkpoint /home/xjtuiair/KITTI_PCDet/pp_multihead_nds5823.pth to CPU
2020-07-31 17:55:10,376   INFO  ==> Done (loaded 421/421)
2020-07-31 17:55:10,399   INFO  Visualized sample index: 	1
Traceback (most recent call last):
  File "demo.py", line 103, in <module>
    main()
  File "demo.py", line 91, in main
    pred_dicts, _ = model.forward(data_dict)
  File "/home/xjtuiair/pcdetv0.3/OpenPCDet/pcdet/models/detectors/pointpillar.py", line 11, in forward
    batch_dict = cur_module(batch_dict)
  File "/usr/local/lib/python3.7/site-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/xjtuiair/pcdetv0.3/OpenPCDet/pcdet/models/backbones_3d/vfe/pillar_vfe.py", line 120, in forward
    features = pfn(features)
  File "/usr/local/lib/python3.7/site-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/xjtuiair/pcdetv0.3/OpenPCDet/pcdet/models/backbones_3d/vfe/pillar_vfe.py", line 37, in forward
    x = self.linear(inputs)
  File "/usr/local/lib/python3.7/site-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/torch/nn/modules/linear.py", line 92, in forward
    return F.linear(input, self.weight, self.bias)
  File "/usr/local/lib/python3.7/site-packages/torch/nn/functional.py", line 1408, in linear
    output = input.matmul(weight.t())
RuntimeError: size mismatch, m1: [412640 x 10], m2: [11 x 64] at /pytorch/aten/src/THC/generic/THCTensorMathBlas.cu:268

Train and val is ok,

sshaoshuai, 2020-07-31:
The provided model is trained with 10 sweeps with an extra timestamp input. It seems you only use 1 single sweep to test.

Xiangzhaohong, 2020-07-31:
@sshaoshuai  Thanks, it means that if i want to use demo.py for nuscenes model , i should change the data format?
sorry for that I am new to these lidar network.

sshaoshuai, 2020-07-31:
Yes, you need to write the nuscene data loader to feed the correct data format to the nuscenes models, just like the training and testing process of nuscenes models.

closed 2020-08-03 by sshaoshuai