open-mmlab/OpenPCDet #198: Train custom data (without 2D information?)

Gltina, 2020-07-31:
Hi,

If I want to train my datasets, before that, I was wondering is it possible to only label and mark the target bounding box and class without providing any 2D information, such as calibration of the camera, images, and 2D rectangle box on images as KITTI does? 

In short, may I only use these 3D data as input to train?

Thanks in advance.

sshaoshuai, 2020-08-01:
For training 3D object detector with OpenPCDet, you need the corresponding 3D labels of your custom point cloud data. The labelling of 3D bounding box is another story and it is out of the bound of this project, but personally I think the answer is yes.

Gltina, 2020-08-03:
> For training 3D object detector with OpenPCDet, you need the corresponding 3D labels of your custom point cloud data. The labelling of 3D bounding box is another story and it is out of the bound of this project, but personally I think the answer is yes.

Thanks for your reply, I will test training my own datasets in the next days with the only 3D data.
I am going to close this issue.

josy43, 2020-08-17:
Does it work ?

Gltina, 2020-08-17:
> Does it work ?

right now, we are testing training our data with only 3D information successfully, which means all 2D information such as calibration, and the 2D bounding box is ignored. But we didn't get the accurate detection result that we want, and we will train more data in the next few weeks. 

If you have any experience of it or specific questions, welcome to leave your comments on [here](https://github.com/open-mmlab/OpenPCDet/issues/176).

josy43, 2020-08-18:
But how do you ignore them ? What do you type in for the 2 D Info?

Gltina, 2020-08-18:
You Got a Good Question! In practice, we just use the default parameters related to 2D, for example, if you don't want to use 2D bounding box, ok! just use the value that occurred in a KITTI label, such that you can make sure it is OK certainly for training as it is running in original KITTI label file.

In your custom data, there are some values in label file you have to check by yourself, i.e., object type, position, dimensions, and rotation_y.

josy43, 2020-08-18:
Okay, so I just type in Fake values. But how do I see the performance on point cloud only ? Which value tell me the performance ?

josy43, 2020-08-18:
When I use Fake values for the 2D information, will the average precision build correctly ?

Gltina, 2020-08-18:
That is also a good question I want to know too! I am not sure if it will affect the performance?

nicholasguimaraes, 2020-09-01:
Hello everyone, 

I was able to successfully train the network with the KITTI dataset (Using all the 2d and calibration parameters).

However I would like to know if any of you did successfully train the network with NO 2d and calibration information.

If yes, this will be great news!

Gltina, 2020-09-03:
@nicholasguimaraes  Yes you can do it.

yangnvzi, 2020-09-03:
@sshaoshuai 
i was able to successfully test the pointrcnn using my own(lidar 40p）lidar data ,but i input 299 frame lidar point ， i didn't get the any  detection result that i want

Gltina, 2020-09-03:
> @sshaoshuai
> i was able to successfully test the pointrcnn using my own(lidar 40p）lidar data ,but i input 299 frame lidar point ， i didn't get the any detection result that i want

you should describe your problem in more detail instead of only "No result I want".

alixiansen91, 2020-11-02:
> 你有一个好问题！实际上，我们仅使用与2D相关的默认参数，例如，如果您不想使用2D边界框，那就好！只需使用KITTI标签中出现的值，这样就可以确保它在原始KITTI标签文件中运行时肯定可以接受训练。
> 
> 在自定义数据中，标签文件中有一些值需要您自己检查，例如，对象类型，位置，尺寸和rotation_y。

Hello, the rotation_y value in the tag file is about the coordinates of the camera. If there is only point cloud data, how should this value be set?    您好，在标签文件中的rotation_y值是关于相机的坐标，如果只有点云数据这个值应该怎么设置呢？

elevenface, 2020-11-02:
kitti数据集中，是否相机为000,O点呢，世界坐标系的

alixiansen91, 2020-11-06:
> […](#)

Hello, we only collected the point cloud data with lidar, and did not use the camera. In this case, what is the basic principle of the value setting of the label and the calib file? I think if there is no camera, can the calib file be removed? Or set it to the default value?您好，我们只用激光雷达采集了点云数据，没有使用相机，这样的话标签以及calib文件的值设置的基本原理大概什么呢？我想没有相机的话calib这个文件是否可以去掉？还是将其设置为默认值不管？

zjx99, 2020-11-09:
@alixiansen91 @Gltina
Hello, 
What tool or library did you use to annotate your lidar data in order to get the labels?  Thanks

Gltina, 2020-11-09:
> @alixiansen91 @Gltina
> Hello,
> What tool or library did you use to annotate your lidar data in order to get the labels? Thanks

Hi, you can refer to [here](https://github.com/Gltina/ACP-3Detection) to know more details on labeling tools.

bingo830422, 2021-01-14:
> kitti数据集中，是否相机为000,O点呢，世界坐标系的 alixiansen91 <notifications@github.com> 于2020年11月2日周一 上午11:49写道：

Please refer to http://www.cvlibs.net/publications/Geiger2013IJRR.pdf
The relationship between Velodyne(LiDAR) and Camera has mentioned in fig.3, which is approximately
Velodyen X = KITTI camera Z
Velodyne Y = KITTI camera X
Velodyne Z = KITTI camera -(Y+1.73)

請參考上述文章，載KITTI的數據中光達與相機的相對位置於圖三有描述，其關係轉換約為上述（會因為個人的校正會有些許誤差，可先於視覺化的工具中查看微調）

clytze0216, 2021-07-06:
hi @Gltina @alixiansen91  I want to know if  I could train the model without calib ?Looking forward to your reply!Thanks a lot.

clytze0216, 2021-07-09:
> @nicholasguimaraes 是的，你可以做到。

hi, I want to custom my own data to train the model.  I have no 2d boxes information. when I run test.py ,there are some errors about the boxes. I change the kitti_dataset.py .I have no input of boxes. Could you help me how to solve the "boxes" data?
It means a lot to me. Looking forward to your reply!.

gzchenjiajun, 2022-02-22:
I have the same problem...
Is there a solution for this?
3d detection without calib and 2d images....

gzchenjiajun, 2022-02-22:
> 

can you elaborate on the implementation methods?   thks!!!!

gzchenjiajun, 2022-02-22:
> You Got a Good Question! In practice, we just use the default parameters related to 2D, for example, if you don't want to use 2D bounding box, ok! just use the value that occurred in a KITTI label, [...]

can you elaborate on the implementation methods?  thks!!!!

jihanyang, 2022-02-22:
#253

closed 2022-02-22 by jihanyang

BayaCherif, 2022-04-24:
> Thanks for your reply, I will test training my own datasets in the next days with the only 3D data. I am going to close this issue.

can you please tell us if it worked fine with greate metrics ? thanks