open-mmlab/OpenPCDet #199: nuScenes # of epochs?

MartinHahner88, 2020-08-01:
Hi @sshaoshuai ,
**thanks for the awesome v0.3.0 update**!

Can you tell us how many epochs you trained your models on the nuScenes dataset?

When I try to train your provided nuScenes models on a single GPU one epoch takes between 6-10 hours,
so if I train for 80 epochs (as I did to reproduce your KITTI models) the training would take up to a full month!

P.S.: I know you trained on 8 GTX 1080Ti GPUs, but training for 80 epochs on 8 GPUs simultaneously 
would still lead to a training time of several days and probably almost a week. 

Keep up the good work 👍

sshaoshuai, 2020-08-01:
Hi Martin, 

Thank you for your active watching and help for this repo. 
In the v0.3.0 version, I have added the epoch hyper-parameter to the config files for recording different epoch setting for different dataset: 
https://github.com/open-mmlab/OpenPCDet/blob/master/tools/cfgs/nuscenes_models/cbgs_second_multihead.yaml#L210-L211

Hence, 20 epochs is enough for nuscenes dataset with 8 GPUs. 
If you trained with a single GPU, I guess the epochs could be further reduced.

MartinHahner88, 2020-08-02:
> https://github.com/open-mmlab/OpenPCDet/blob/master/tools/cfgs/nuscenes_models/cbgs_second_multihead.yaml#L210-L211

This is great, I did not spot that. Thank you for the swift answer!

closed 2020-08-02 by MartinHahner88

darrenjkt, 2022-07-25:
Hi Shaoshuai, how did you choose 20 epochs for nuscenes? With the OneCycle LR policy, the LR during training is quite epoch dependent. Setting e.g. 80 epochs, and early stopping at 20 epochs won't perform the same as training for just 20 epochs. 

Did you have to fully re-train the network at various number of epochs in order to see what works best? E.g. 15, 20, 30 etc.